webdino/gecko-embedded #4: i.MX6: H.264ハードウェアデコードのサポート

ashie, 2016-07-26:
i.MX6でH.264のハードウェアデコードに対応させたい。
i.MX6でハードウェアデコードに対応させる方法としてベンダ独自のライブラリ（libfslvpuwrap）を使うか、GStreamerを使う方法があると理解している。
一方でFirefox側の事情として最新版ではGStreamerサポートがドロップされているので、将来的なことを考えるとGStreamerを使うのは得策ではなさそう。Firefoxに直接VPU対応コードを入れることを検討する。

ashie, 2016-09-26:
ちなみに、GStreamerでの動作は以前に確認している。再生はできるのだが、CPU負荷がFFmpegでのソフトウェア再生より高いという結果になっている。ハードウェアデコーダが使われていることは #3 と同じ方法で確認しているが、CPU負荷が高い原因はまだ調査していない。#3 の RZ/G1 の事例で映像が表示できていない状態でもCPU負荷が高い現象を確認しているので、i.MX6でも一度調査して比較してみた方が良さそう。

ashie, 2016-09-26:
GStreamerを有効化するオプションはFirefoxのレシピに入れてある（B2Gはまだ）:

https://github.com/mozilla-japan/meta-browser/blob/firefox-45.0esr/recipes-mozilla/firefox/firefox_45.3.0esr.bb#L41

有効化するには、local.confに以下を追加:

```
PACKAGECONFIG_append_pn-firefox = " gstreamer1.0 "
```

ただし依存パッケージの指定が間違っているような気がする（RDEPENDSで指定しているパッケージもDEPENDSな気がする）ので要チェック。

dynamis, 2016-09-29:
@sotaroikeda 昨夜のミーティングでも少し話を振りましたが、このあたりの実装していく上でヒントとか助言とか頂ければ幸いです。

ashie, 2016-10-04:
Buzillaにもbugを立てた: Bug 1306529

ashie, 2016-10-31:
#3 の対応でPDM(PalatformDecoderModule)が少し分かってきたので、こちらも折を見て対応を進める。

> i.MX6でハードウェアデコードに対応させる方法としてベンダ独自のライブラリ（libfslvpuwrap）を使うか、GStreamerを使う方法があると理解している。
> 一方でGecko側の事情として最新版ではGStreamerサポートがドロップされているので、将来的なことを考えるとGStreamerを使うのは得策ではなさそう。Geckoに直接VPU対応コードを入れることを検討する。

いずれにしてもポーティングは必要だが、将来的に別のボードに対応することも考えると、PDM版のGStreamer対応もあった方が良いと考えている。両方やってみたいが、まずどちらを優先すべきか迷ううところ。