webdino/gecko-embedded #9: Node.js と組み合わせて使えるようにする

dynamis, 2016-08-08:
Web ブラウザからデバイス操作をする場合は
* 各デバイス操作専用の WebAPI を WebIDL 書いて Gecko に実装してそれを操作する
* 各デバイスを操作するローカルのアプリやサーバと Web 技術で通信して操作する
という解があるが、後者の解として Node.js と組み合わせることは多いので対応したい。

具体的には meta-nodejs レイヤーがあるのでそれと組み合わせて使えるようにする。ハード制御などは Node 側で行ってブラウザの Web コンテンツ側からはローカルの Node と通信させる。

https://github.com/imyller/meta-nodejs
https://layers.openembedded.org/layerindex/branch/master/layer/meta-nodejs/
* 現時点で用意されているバージョンは 4.4.7 と 6.3.1
  * v4 が現在の Active LTS なのでまずはそこから
  * v6 は 2016/10 以降に Active LTS に切り変わったら
でしょうか。

note: 少なくとも Qt などと組み合わせて使ってる例はある様子
https://blog.qt.io/blog/2016/06/23/enabling-connectivity-embedded-product/

dynamis, 2016-08-08:
具体的な利用例としては Serial や BLE などを使っての各デバイスとの通信用途。
* USB Serial
  * モジュールは node-serialport とか
  * 例えば ECHONET Lite のスマートメーターなどは Wi-SUN 経由
* BLE = Bluetooth Low Energy
  * モジュールは noble とか
  * BLE (GATT) は色々あるが
  * [Project Linking](https://linkingiot.com/) デバイスなんかも結構お手軽か

dynamis, 2016-08-18:
どうせなら同じ JS エンジンが使える SpiderNode も使えると良いと思ってる:
* https://github.com/mozilla/spidernode

まだ実験的なプロジェクトという位置づけになってるので採用は早いかも知れないが、本格的に使われる見込みも出てきてるので少し先ならありか。

dynamis, 2016-09-27:
外部デバイスとして Serial / BLE の前にセンサー系の I/O にも対応する
* GPIO
  * RasPi で動いている [gpio モジュール](https://www.npmjs.com/package/gpio) がそのまま or 少し弄って使える想定
  * [別の実装](https://www.npmjs.com/package/gpio-js) とか [高速動作用](https://www.npmjs.com/search?q=gpio)とか[他にもとてもいっぱいある](https://www.npmjs.com/search?q=gpio)
* I2C
  * RasPi や BeagleBone などで動いている [i2c モジュール](https://www.npmjs.com/package/i2c
) がそのまま or 少し弄って使える想定
  * [ほかにもいろいろあるっぽい](https://www.npmjs.com/search?q=i2c)

一杯モジュールがあるので最新の Node.js (安定ブランチ) などでちゃんと動くモノを確認していく

naobsd, 2016-10-17:
https://github.com/imyller/meta-nodejs は v4.6.0/6.8.0 になっていますね。 (最初のメッセージを修正しました)

96boards krogoth 上で v4.6.0 で [gpio](https://www.npmjs.com/package/gpio)/[i2c](https://www.npmjs.com/package/i2c) を使ってみましたが、特に問題ありませんでした。
しかし、時期的にはもう v6 の方がよいですかね...

dynamis, 2016-10-26:
v6.9.x もリリースされたので v6 に移行して行きたいとは思いますが、ひとまずは v4.6.0 でも全然大丈夫です。Firefox 側も何か別のビルド試すときにでもまた。

note: v6 が嬉しい理由は 4 系列は ES6 の対応が弱く Firefox 側と同じコードが動かないところが増えるから。デバイス使うだけであれば取り敢えずそんなにややこしいコードは書かなければ大丈夫。
http://kangax.github.io/compat-table/es6/#node4

dynamis, 2016-11-22:
Node と組み合わせて GPIO/I2C デバイスを使うという Issue は解決済みなのでこちらは close します。
Node v6 へのアップデートについては #36 で (他と一緒に) 続きという事で。

現状としては https://github.com/imyller/meta-nodejs が v4.6.2/6.9.1 まで更新されたようで、6.9.x は LTS なので切り替えても大丈夫な感じにはなったかと思います。

closed 2016-11-22 by dynamis