webdino/gecko-embedded #12: 9 月末あたりにミートアップイベント開催

dynamis, 2016-09-13:
7 月末に初回のミートアップをしてから少し空いてしまいましたが、2 ヶ月ぶりくらいでコミュニティミートアップをしましょう！

比較的集まりやすい人が多そうな、平日水曜日夜に 1.5 ~ 2 時間程度を想定しています。例えばこんな感じですが如何でしょう？

* 日時: 9/28 (水) 19:00 ~ 21:00
* 場所: Mozilla Japan オフィス
* 内容:
  * プロジェクトアップデート by dynamis
  * Wayland サポートの進捗 by 足永さん
  * 96boards にも移植してみた by 深海さん
  * ディスカッションタイム (今度こそ QA だけじゃなくて...)

dynamis, 2016-09-13:
TODO:
- [ ] 日時と講師を確定
- [ ] Doorkeeper にイベントページを作成
- [ ] Doorkeeper (mozilla/firefox) などで告知
- [ ] 一部企業の方に個別にご案内

naobsd, 2016-09-13:
@dynamis ひとまず9/28 19:00～は大丈夫です。

内容は、96Boards HiKeyで動かしてみた、くらいで... (移植らしいことは現状していないので...)

naobsd, 2016-09-26:
DragonBoard 410cも入手しましたので、これについても少しお話ができればと思います。

dynamis, 2016-09-27:
こちらに書いてませんでしたが @Hidetaro7 @tommmmmmmmy と #10 についてもお話ししたりコミュニティ的にやりたいこととかフィードバックとかもらうのも含めています

dynamis, 2016-09-29:
DONE. Thanks!

closed 2016-09-29 by dynamis