webdino/gecko-embedded #13: 最新版Firefoxのビルド

ashie, 2016-09-16:
現在のポーティング対象バージョンは45ESRとしているが、最新版も常にビルドは通るようにしておきたい。48については途中まで作業した版を以下に置いてある。

https://github.com/mozilla-japan/meta-browser/tree/firefox-48-wip

libxulのリンク時にツールチェーンへのパスが通らなくなっており、まだビルドは成功していない。
49.0b1でも同様の状況。

```
    /bin/sh: 1: arm-linux-gnueabi-readelf: not found
    test "$(arm-linux-gnueabi-nm -g libxul.so | grep _NSModule$ | grep -vw refptr | sort | sed -n 's/^.* _*\([^ ]*\)$/\1/;1
    /bin/sh: 1: arm-linux-gnueabi-nm: not found
    NSModules are not ordered appropriately
```

ashie, 2016-10-05:
> https://github.com/mozilla-japan/meta-browser/tree/firefox-48-wip

49に更新して以下に移動した。

https://github.com/mozilla-japan/meta-browser/tree/firefox-49-wip

エラー内容は変わらず。
どこかで環境変数渡せば良いはずだが、どこなのか分かっていない。

ashie, 2016-10-05:
>   /bin/sh: 1: arm-linux-gnueabi-nm: not found

よく見たらtoolchainのprefixが実際と異なる。
実際は`arm-poky-linux-gnueabi`。
`--with-toolchain-prefix="arm-poky-linux-gnueabi-"`を与えるとビルドが通る。

ashie, 2016-11-17:
50に更新した。

https://github.com/mozilla-japan/meta-browser/tree/firefox-50-wip

RZ/G1E上のX11で動作を確認した。アクセラレーションは効かないが、特に問題は見られない。
WaylandパッチとOpenMAXパッチは移植が必要で、まだ対応していない。

ashie, 2016-11-17:
> https://github.com/mozilla-japan/meta-browser/tree/firefox-49-wip

こっちは消した。

ashie, 2016-11-30:
> WaylandパッチとOpenMAXパッチは移植が必要で、まだ対応していない。

OpenMAXの方は僅かな修正でビルドが通り、X11上ではあるが動作も確認できた。
（現在のtrunk向けパッチをそのまま当てられる。）
後でレシピに投入しておく。

WaylandとCompositorのGPUアクセラレーションは少してこずりそう。

ashie, 2016-11-30:
45ESRではAACデコーダが無いRZ/G1用にBlankデコーダにfallbackするコードを入れたが、50ではそんな小細工は必要ないようだ（ボリュームボタンに☓が付き映像だけ再生できる）。

ashie, 2016-11-30:
iWave RZ/G1E（X11）だとなぜかabout:supportでクラッシュする。
他のボードで確認したとき（i.MX6だったかRZ/G1E StarterKitだったかは記憶が定かでは無い）は問題無かった。

ashie, 2016-12-14:
Firefox 51betaも試しているが、ビルド周りが壊れていて、まだビルドできていない。

https://github.com/mozilla-japan/meta-browser/tree/firefox-51-wip

ashie, 2016-12-20:
> 50に更新した。
> 
> https://github.com/mozilla-japan/meta-browser/tree/firefox-50-wip
> 
> RZ/G1E上のX11で動作を確認した。アクセラレーションは効かないが、特に問題は見られない。

`layers.acceleration.force-enabled`が`true`でもBasic Compositorになってしまう。
gfx/thebe/gfxPlatform.cppで、force-enabledなのにドライバのブラックリストを見に行ってしまうようだ。
（バグなのか仕様なのかは現時点ではよくわからない。）
このチェックを飛ばせば、これまでどおりアクセラレーションが効くようになった。

ashie, 2016-12-26:
> Firefox 51betaも試しているが、ビルド周りが壊れていて、まだビルドできていない。
> 
> https://github.com/mozilla-japan/meta-browser/tree/firefox-51-wip

なんとかビルドは通した。
mozilla-centralでは直っているところもあるが、直っていないところもあるので、
後者については報告しておく。

ashie, 2016-12-26:
Wayland版について

All HandsでStranskyから状況を教えてもらったところ、最近のバージョンでは少なくとも以下のCompositorWidgetの修正を入れないと動作しないようだ:

https://bugzilla.mozilla.org/show_bug.cgi?id=1281425

まずは修正を入れずに動かしてみたところ、確かにCompositorWidgetを作成しようとするところでクラッシュする。

今のmozilla-centralにはパッチが当たらなくなっているので少し修正して適用してみたところ、クラッシュはせずWindowは出るようになったが、コンテンツが描画されない。

ashie, 2016-12-26:
> Wayland版について
> 
> All HandsでStranskyから状況を教えてもらったところ、最近のバージョンでは少なくとも以下の> CompositorWidgetの修正を入れないと動作しないようだ:

Firefox 45とかで当てていたWayland関連のバッチは既にほぼ全部取り込まれている。

ashie, 2016-12-26:
ちなみに開発リソースが足りないので通常版を本プロジェクトとしてリリースすることは無いが、次のESR版（52）についてはリリースからなるべく間を置かずに対応したいというのがこのissueの趣旨（なのでタイトル変えた）。

Firefox 52のリリース予定日は2017-03-07:

https://wiki.mozilla.org/RapidRelease/Calendar

ashie, 2016-12-27:
>     Firefox 51betaも試しているが、ビルド周りが壊れていて、まだビルドできていない。

> なんとかビルドは通した。

krogoth (Yocto 2.1)だとまだ通らないようだ。

ashie, 2016-12-27:
> krogoth (Yocto 2.1)だとまだ通らないようだ。

パッチがミスってただけなので修正した

ashie, 2016-12-27:
>    50に更新した。

> layers.acceleration.force-enabledがtrueでもBasic Compositorになってしまう。

51では修正されているようだ。

> このチェックを飛ばせば、これまでどおりアクセラレーションが効くようになった。

このパッチを当てなくてもアクセラレーションが効いている。

ashie, 2016-12-28:
ビルドが壊れたらなるべく早く対応しておいてレシピで当てるパッチは減らしたいので、mozilla-centralを追いかけるレシピも作成しておきたい。YoctoにはMercurialのfetcherもあるのでこれを使うつもりだが、使っている事例が見つからなくて使い方がまだよくわかっていない（gecko-devの方でgit使うのでもいいけど）。

ashie, 2017-01-05:
> YoctoにはMercurialのfetcherもあるのでこれを使うつもりだが、使っている事例が見つからなくて使い方がまだよくわかっていない

一つだけ使用例を見つけた:

http://cgit.openembedded.org/meta-openembedded/plain/meta-multimedia/recipes-dvb/dvb-apps/dvb-apps_1.1.1.bb

ashie, 2017-01-25:
firefox-51-wipブランチを51.0b9から51.0リリース版に更新した

https://github.com/mozilla-japan/meta-browser/tree/firefox-51-wip

51.0b9では一応動作してアクセラレーションも効いていたが、ポップアップウィンドウが正しく動作していなかった。
リリース版では、そもそもクラッシュして起動できなくなっている。

cosmo0920, 2017-04-12:
https://github.com/meta-rust/meta-rust/pull/148 Rust 1.15.0が来てました。
mozilla-centralで1.16.0が必要になっていたらこれと同じように更新するといいんじゃないでしょうか。

ashie, 2017-04-24:
@cosmo0920 さんが進めているFirefox 52ESRのブランチ:

* https://github.com/mozilla-japan/meta-browser/tree/firefox-52.1esr-wip

以下は確認できている

* Waylandでの動作
* CompositorのGPUアクセラレーション対応
* OpenMAX IL対応

ただし、45ESRでも問題があったポップアップウィンドウの挙動については、52ESRでも引き続き問題がある。

ashie, 2017-05-26:
> @cosmo0920 さんが進めているFirefox 52ESRのブランチ:
> 
>    * https://github.com/mozilla-japan/meta-browser/tree/firefox-52.1esr-wip

現在はこちらに移行している:

* https://github.com/mozilla-japan/meta-browser/tree/firefox-52.1esr

cosmo0920, 2017-05-26:
ポップアンウィンドウの挙動については改善のめどが付きました。
ロケーションバーの挙動はEGLを有効化すると悪化するようです。

ashie, 2017-05-30:
Firefox 52のビルドは通るようになって、Firefox 45のときのパッチも一通り入ってるのでクローズします。

upstream側にFirefox 52のレシピが入ったらそれをベースにしたいとか、その他細かい問題はまだあるが、
必要に応じて別issueを立てるなどして対応する。

closed 2017-05-30 by ashie