webdino/gecko-embedded #14: 96Boards で動かす

naobsd, 2016-09-27:
(とりあえずメモ。後で書き直す)

# Status

## hikey

 | EGL | firefox
---- | ---- | ----
X11 | no | crash
Wayland | yes | crash

## hikey-32

 | EGL | firefox
---- | ---- | ----
X11 | no | slow
Wayland | yes | working

## dragonboard-410c

 | EGL | firefox
---- | ---- | ----
X11 | yes | crash
Wayland | (TBD) | (TBD)

# Todo

- [ ] wiki を書く (build, install)
- [ ] aarch64 で crash する問題を調査する

dynamis, 2016-10-26:
howto 少し書き始めました https://github.com/mozilla-japan/gecko-embedded/wiki/Dragon-Board-410c
頂いた sample コードをベースにまたいろいろ追記していきます。

dynamis, 2016-11-22:
Firefox/Gecko を動かすようにするという本題以外の Node やコマンド類の更新については #36 に分離して Issue を立てました。

dynamis, 2019-05-28:
動かすのは完了していたし、その後も普通にビルドできていたのでこちらはクローズ。
aarch 64 問題は Gecko 68 でやっと公式対応するから多分解決するんじゃないかなと思ってる (未確認)。

closed 2019-05-28 by dynamis