webdino/gecko-embedded #15: Firefox 45ESR が aarch64 で crash する

naobsd, 2016-09-27:
96Boards の aarch64 userland 環境では、 Firefox の起動後、ページがレンダリングされた直後に SEGV で落ちる。ネットワークに未接続などコンテンツが取得できない状態だと、少し経ってから落ちる。
(一度落ちると、 ~/.mozilla を消さない限り、ウィンドウが出る前に落ちる)

https://bugzilla.mozilla.org/show_bug.cgi?id=1304962 と同じ問題のように思えるが、パッチを適用しても直らなかった。 (要再確認)

naobsd, 2016-09-27:
https://bugs.launchpad.net/ubuntu/+source/firefox/+bug/1598605 同じ問題?

naobsd, 2016-09-27:
[96Boards 公式 Debian Jessie 環境](https://github.com/Linaro/documentation/blob/master/Reference-Platform/RPTest/ConsumerEdition/DragonBoard-410c/InstallDebianRPB-16.06.md#fastboot-method) の firefox-esr (45.4.0) も crash した。

dynamis, 2016-09-27:
note: ESR45 ではなく mozilla-central の場合は現状 aarch64 で動かすにはこのあたりがブロッカーになっている:
https://bugzilla.mozilla.org/show_bug.cgi?id=1142056
https://bugzilla.mozilla.org/show_bug.cgi?id=1303952

makotokato, 2016-10-07:
たぶんこれと同じ話なんだけど、96BoardsにDebianで再現できるなら入れてちょっと試してみます。
https://bugzilla.mozilla.org/show_bug.cgi?id=1302554

naobsd, 2016-10-07:
ああ、48bit VA (bug 1143022)ではない事は確認していましたが、nsIOService::NewChannelFromURIWithProxyFlagsInternal()の最後のchannel.forget()の話は、見落としていました。

ここかもしれませんね... (ちゃんとdump取らないとです...)

makotokato, 2016-10-07:
Bug 1302554の話だとXPCOM/JS Binding (ex. Bug 1304962) かコンパイラのバグ以外ありえないので、いろいろテストしてみないとなんとも言えないです (手元のPINE64だとgcc 5.3.1なので)

ashie, 2017-05-31:
#13 のFirefox 52で再度試しみたいところ。

cosmo0920, 2017-06-19:
土日にJetson TX2上のUbuntu 16.04でセルフビルドしたところ、以下のパッチを投入するとビルド&起動できました。

```diff
diff -r 95543bdc59bd media/libcubeb/cubeb-pulse-rs/src/backend/context.rs
--- a/media/libcubeb/cubeb-pulse-rs/src/backend/context.rs      Sun Jun 18 20:35:11 2017 +0200
+++ b/media/libcubeb/cubeb-pulse-rs/src/backend/context.rs      Mon Jun 19 05:11:34 2017 +0000
@@ -82,7 +82,7 @@
 
 impl Context {
     #[cfg(feature = "pulse-dlopen")]
-    fn _new(name: *const i8) -> Result<Box<Self>> {
+    fn _new(name: *const u8) -> Result<Box<Self>> {
         let libpulse = unsafe { open() };
         if libpulse.is_none() {
             return Err(cubeb::ERROR);
```

cosmo0920, 2017-06-19:
Jetson TX2でのgccのバージョンは 5.4.0です。

```log
$ gcc --version
gcc-5.real (Ubuntu/Linaro 5.4.0-6ubuntu1~16.04.4) 5.4.0 20160609
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

```

cosmo0920, 2017-06-19:
リビジョンは以下。

```log
$ hg tip
changeset:   364563:95543bdc59bd
tag:         tip
parent:      364552:ae20db43e815
parent:      364562:8c7dd6a9e935
user:        Sebastian Hengst <archaeopteryx@coole-files.de>
date:        Sun Jun 18 20:35:11 2017 +0200
summary:     merge mozilla-inbound to mozilla-central. r=merge a=merge

```

56.0a1相当。

cosmo0920, 2017-06-23:
以下のリビジョンでは上記パッチを当てなくてもビルドが通るようになりました。

```log
$ hg tip
changeset:   365771:594cc32b6323
tag:         tip
user:        Carsten "Tomcat" Book <cbook@mozilla.com>
date:        Fri Jun 23 13:45:55 2017 +0200
summary:     Backed out changeset a7a5b2fc4f0a (bug 1357715) for causing perma failures after merged to m-c
```

コンパイルエラーの該当箇所は `Option<CString>` に。C FFIが関係してくるコードなら妥当な変更かと思います。

```rust
impl Context {
    #[cfg(feature = "pulse-dlopen")]
    fn _new(name: Option<CString>) -> Result<Box<Self>> {
        let libpulse = unsafe { open() };
        if libpulse.is_none() {
            return Err(cubeb::ERROR);
        }
  // <snip>
```

ashie, 2017-10-17:
#21 でFirefox 52 + R-Car gen3（aarch64）であれば動作することを確認できた。
Firefox 45は今後はメンテナンスしないので、本issueはクローズします。

closed 2017-10-17 by ashie