webdino/gecko-embedded #19: WebRTCの有効化

ashie, 2016-10-04:
現在のレシピではWebRTCを有効化してビルドしている。

以前WandboardでUVCデバイスを接続して試したところ、デバイスは認識してリストに表示されたが、映像を表示しようとするとクラッシュした。原因は調査していない。
ほぼ同一の設定でビルドしたPC上のLinuxでは動作した。

ashie, 2016-10-04:
Bugzillaにもbugを立てた

https://bugzilla.mozilla.org/show_bug.cgi?id=1307329

cosmo0920, 2016-10-04:
mozconfigの情報をbugzillaに添付しなくても良いのでしょうか…？

ashie, 2016-10-04:
今のところ、ちょっと試してみたけどダメだったという備忘録でしかないので、mozconfigも含めて詳細情報はもう少し情報を収集できてから出します。どのカメラ使ったかも、今手元に無くて情報出せてないですし...

ashie, 2016-10-05:
使用したカメラはLogicool C910

ashie, 2016-10-14:
メモ: Wandboardのcore-image-satoではデフォルトだとuvcvideoが組み込まれてなかった。

```
# /sbin/depmod -a
# /sbin/modprobe usbvideo
```

ashie, 2016-10-18:
デバッグログを取るために新規でイメージを作成して試したところ、クラッシュせずにカメラ映像が表示されてしまった。

* クラッシュする環境: Firefox 45.2: GTK2
* クラッシュしない環境: Firefox 45.4: GTK3

ashie, 2016-10-18:
> デバッグログを取るために新規でイメージを作成してgdbにかけて試したところ、クラッシュせずにカメラ映像が表示されてしまった。以後、gdbにかけずに試してもクラッシュしない。

と思ったが、OSを再起動して試したらクラッシュした。
core吐かせないとスタックトレースとれないか。

ashie, 2017-05-11:
現時点の以下のブランチを使って、PC上のWayland版FirefoxでWebRTCを試してみた

https://github.com/mozilla-japan/gecko-dev/tree/esr52-wayland-with-egl
https://github.com/mozilla-japan/gecko-dev/commit/b8e9a71d2ff106132a7b9da3b3b84103ad81fad0

* Wayland版Firefoxの問題で、WebRTC以前にカメラへのアクセス許可のポップアップウィンドウが出ない
* 予め許可しておいても、カメラへのアクセス時にクラッシュする
* 同じビルドのX版では動作する

Wayland版については、実機以前にPC上で動くようにしないと話が始まらなそう。

ashie, 2017-10-26:
iwg20mで試した結果:

```
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x48f6294c in raise (sig=11) at ../sysdeps/unix/sysv/linux/pt-raise.c:36
36	  return INLINE_SYSCALL (tgkill, 3, pid, THREAD_GETMEM (THREAD_SELF, tid),
(gdb) where
#0  0x48f6294c in raise (sig=11) at ../sysdeps/unix/sysv/linux/pt-raise.c:36
#1  0xb554fbfc in nsProfileLock::FatalSignalHandler(int, siginfo_t*, void*) () from /usr/lib/firefox-52.4.1/libxul.so
#2  <signal handler called>
#3  0x45544e48 in ?? ()
#4  0xb69f1760 in XSync (dpy=0xb6da6be0, discard=0) at /usr/src/debug/libx11/1_1.6.3-r0/libX11-1.6.3/src/Sync.c:42
#5  0xb4e827f8 in nsWindow::HideWindowChrome(bool) () from /usr/lib/firefox-52.4.1/libxul.so
#6  0xb5364f48 in nsXULWindow::SyncAttributesToWidget() () from /usr/lib/firefox-52.4.1/libxul.so
#7  0xb536a158 in nsXULWindow::OnChromeLoaded() () from /usr/lib/firefox-52.4.1/libxul.so
#8  0xb536aeb0 in nsWebShellWindow::OnStateChange(nsIWebProgress*, nsIRequest*, unsigned int, nsresult) () from /usr/lib/firefox-52.4.1/libxul.so
#9  0xb40a7f10 in nsDocLoader::DoFireOnStateChange(nsIWebProgress*, nsIRequest*, int&, nsresult) () from /usr/lib/firefox-52.4.1/libxul.so
#10 0xb40aa420 in nsDocLoader::doStopDocumentLoad(nsIRequest*, nsresult) () from /usr/lib/firefox-52.4.1/libxul.so
#11 0xb40aabec in nsDocLoader::DocLoaderIsEmpty(bool) () from /usr/lib/firefox-52.4.1/libxul.so
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

該当コード: widget/gtk/nsWindow.cpp:

```cpp
5264 NS_IMETHODIMP
5265 nsWindow::HideWindowChrome(bool aShouldHide)
5266 {
... 
5306 #ifdef MOZ_X11
5307     XSync(GDK_DISPLAY_XDISPLAY(gdk_display_get_default()) , False);
5308 #else
5309     gdk_flush ();
5310 #endif /* MOZ_X11 */
```

ランタイムのX11/Waylandの切り分けが入っていない。

ashie, 2017-10-26:
そこを突破しても別の箇所で落ちる

```
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x48f6294c in raise (sig=11) at ../sysdeps/unix/sysv/linux/pt-raise.c:36
36	  return INLINE_SYSCALL (tgkill, 3, pid, THREAD_GETMEM (THREAD_SELF, tid),
(gdb) where
#0  0x48f6294c in raise (sig=11) at ../sysdeps/unix/sysv/linux/pt-raise.c:36
#1  0xb5580bfc in nsProfileLock::FatalSignalHandler(int, siginfo_t*, void*) () from /usr/lib/firefox-52.4.1/libxul.so
#2  <signal handler called>
#3  0xb4eb0b28 in moz_container_get_wl_surface () from /usr/lib/firefox-52.4.1/libxul.so
#4  0xb4ec6074 in mozilla::widget::WindowSurfaceWayland::Commit(mozilla::gfx::IntRegionTyped<mozilla::LayoutDevicePixel> const&) () from /usr/lib/firefox-52.4.1/libxul.so
#5  0xb4ebab5c in nsWindow::OnExposeEvent(_cairo*) [clone .part.162] [clone .constprop.194] () from /usr/lib/firefox-52.4.1/libxul.so
#6  0xbec3e1e4 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

ashie, 2017-10-26:
こんな感じでとりあえずは動くようになった。

```cpp
diff --git a/widget/gtk/nsWindow.cpp b/widget/gtk/nsWindow.cpp
index 6f1215cbdcfa..186e98a02162 100644
--- a/widget/gtk/nsWindow.cpp
+++ b/widget/gtk/nsWindow.cpp
@@ -5304,7 +5304,10 @@ nsWindow::HideWindowChrome(bool aShouldHide)
     // error later when this happens (when the persistence timer fires
     // and GetWindowPos is called)
 #ifdef MOZ_X11
-    XSync(GDK_DISPLAY_XDISPLAY(gdk_display_get_default()) , False);
+    if (mIsX11Display)
+        XSync(GDK_DISPLAY_XDISPLAY(gdk_display_get_default()) , False);
+    else
+        gdk_flush ();
 #else
     gdk_flush ();
 #endif /* MOZ_X11 */
@@ -7194,6 +7197,8 @@ nsWindow::GetWaylandDisplay()
 wl_surface*
 nsWindow::GetWaylandSurface()
 {
-  return moz_container_get_wl_surface(MOZ_CONTAINER(mContainer));
+  if (IS_MOZ_CONTAINER(mContainer))
+    return moz_container_get_wl_surface(MOZ_CONTAINER(mContainer));
+  return nullptr;
 }
 #endif
```

ashie, 2017-10-27:
RZ/G1Mで動作を確認できたのでクローズします。
修正は以下のブランチに入れてあります。

https://github.com/webdino/meta-browser/tree/firefox-52.4esr

closed 2017-10-27 by ashie

ashie, 2017-10-30:
Stranskyにも報告しておいた

https://bugzilla.redhat.com/show_bug.cgi?id=1507369