webdino/gecko-embedded #20: GTK+更新レシピの整理

ashie, 2016-10-04:
#1 でRZ/G1E用にGTK+その他を更新するレシピを実験的に作成したが、試行錯誤しながらだったので整理されていない。またmozilla-japan以下に移動した方が良いので、そのタイミングで整理し直す。

* リポジトリ名は meta-gecko-embedded とする
    * meta-gemにしたいが、Rubyのgemと誤解されそうなので...
* ドキュメントを用意する
* ライセンスを明記する
* https://github.com/ashie/meta-mozwayland のレシピはmeta-gecko-embedded以下にサブディレクトリを作ってそこに移動する。
    * pokyから拝借したレシピはmeta-pokyに移動。
    * RZ/G1依存のレシピはmeta-rz-g1に移動。
    * 他のボード向けにも追加レシピが必要な場合は、mera-rz-g1と同様のレイヤーを用意する
* pokyから拝借したレシピは、Yocto-1.6でビルドできるように少し修正しているが、修正した版しかコミットしていなかったため、修正内容がわかりづらい。変更履歴がわかるように以下のようにコミットし直す。
    * まず無修正のものをコミットする。コミットメッセージに拝借元のレポジトリとリビジョンを明記する。
    * 必要な修正をコミットする

ashie, 2016-10-04:
B2Gのレシピはこれまでmeta-browserのブランチで作成していたが、B2G自体の行方がよくわからず、いまさらmeta-browserにupstreamするのも微妙なので、こちらに移動した方が良いかも...

ashie, 2016-10-04:
@dynamis さん、`meta-gecko-embedded`という名前で空のリポジトリを作って頂けますか?

dynamis, 2016-10-05:
ええっと、こんな感じの理解で宜しいでしょうか？

* meta-mozwayland などあるのを整理してまとめるリポジトリを mozilla-japan organization 配下に作成
   * https://github.com/mozilla-japan/meta-gecko-embedded
* B2G 関連のレシピの upstream は当面考えないことにする
* Firefox ESR のレシピを meta-browser に upstream することは継続で考えている

ashie, 2016-10-05:
はい、そんな感じで考えています。

> Firefox ESR のレシピを meta-browser に upstream することは継続で考えている

全然安定はしてないですが、最低限のドキュメントを追加した後、いったん[Do not merge]でpull request投げて存在を知らせて、マージまでの残りの作業として何が必要か・どうマージしてもらうかを相談しようと思っています。

その際、B2Gのレシピの存在も伝えて、meta-browserに入れてもらうのか、meta-gecko-embeddedかどこかに入れるのかを決定するのかなと思ってます。

ashie, 2016-10-05:
meta-gecko-embeddedが出来たのを確認しました。
ありがとうございます！

ashie, 2016-12-28:
GTK+のバージョンアップは3.18まではなんとかできているが、3.20以降に上げるにはWayland-1.10以降が必要で、Waylandを上げるにはGPUドライバの更新が必要。RZ/GのBSPがR-Car Gen3ベース(Wayland 1.9)になれば上げられる可能性は高そうだが、現状のGen2ベース(Wayland 1.5)の場合はちょっと辛そう。

Waylandのフルサポートを謳っているGTKは3.20からであり、

* https://wiki.gnome.org/Projects/GTK+/Roadmap

実際PC上では3.20に上げることでいくつかの不具合が解消することが分かっている(#2)ので少し無理をしてでも上げたいところ。

ashie, 2017-01-18:
GTK+の更新部分についてまとめ直したものを [meta-gecko-embedded](https://github.com/mozilla-japan/meta-gecko-embedded) にpushした。
あとは

> RZ/G1依存のレシピはmeta-rz-g1に移動。

これを追加したらWikiのインストール手順をこちらに切り替える。

ashie, 2017-01-18:
>    RZ/G1依存のレシピはmeta-rz-g1に移動。
>
> これを追加したらWikiのインストール手順をこちらに切り替える。

これは[meta-rzg-demos](https://github.com/renesas-rz/meta-rzg-demos)から持ってきたものだが、本当に必要なものは現在は[meta-renesas](https://github.com/renesas-rz/meta-renesas)の方に取り込まれているようだ。
最新版でビルドを検証してみて、問題がなければWikiを更新して本issueをクローズする。

ashie, 2017-01-18:
上記はYocto 1.6用だが、Yocto 2.0ベースのBSPが昨年末に公開されていたようだ:

http://elinux.org/RZ-G/Boards/Yocto_2.0

ashie, 2017-01-19:
> 最新版でビルドを検証してみて、問題がなければWikiを更新して本issueをクローズする。

[Yocto-1.6の方も少しビルド手順が更新されていた](http://elinux.org/RZ-G/Boards/Yocto)ので、改めてクリーンビルドして確認してみた。
ビルドまでは問題無し。
ついでに #39 の作業の結果 meta-openembedded/meta-gnome  レイヤが必要なくなったことも確認できた。
あとは動作を確認してWikiを更新したら、本issueと #39 はクローズ。

ashie, 2017-01-19:
〉 上記はYocto 1.6用だが、Yocto 2.0ベースのBSPが昨年末に公開されていたようだ:

2.0の方は1.6が一段落したら別issueを立てて進める。

ashie, 2017-01-20:
Yocto-1.6系の最新版で動作を確認したところ、以前のmeta-mozwaylandでも、今回更新したmeta-gecko-embeddedでもWayland-1.6がクラッシュする。Waylandが1.5のままであれば問題ない。
プロプライエタリドライバに変更は無いようなので直せると思うが、思っていたよりも手間がかかりそうだ。

以前の動いていたリビジョンでどうなるかも試してみる。

ashie, 2017-01-20:
meta-renesas:f4e910fcb0b931ee39529cc77d608f2127c6f579 であれば動く。
ドキュメントの方をいったんこのリビジョンで止めて、最新版についてはおいおい調査する。

ashie, 2017-01-20:
Wikiのビルド手順を更新したのでクローズする。
初期状態でよりデモしやすくする改善は #40 で行う。

closed 2017-01-20 by ashie