webdino/gecko-embedded #23: Graphene のビルド

dynamis, 2016-10-05:
Firefox ブラウザの UI などなしで Gecko エンジン上で直接 HTML コンテンツをロードするデスクトップランタイム Graphene をビルドして利用できるようにする。Firefox/Gecko の移植としては Firefox ブラウザで良いが、実際の組み込み製品では Graphene のようなビルドの方が理論上は望ましくなるため。

see: https://mykzilla.org/2016/08/30/graphene-on-gecko-without-b2g/

dynamis, 2016-10-05:
Building Graphene (Gecko flavor)
https://github.com/browserhtml/browserhtml/wiki/Building-Graphene-%28Gecko-flavor%29
graphene-gecko repo
https://github.com/mykmelez/graphene-gecko

dynamis, 2017-01-12:
Graphene プロジェクトの開発は 2017 年の開発プライオリティ対象にならなかったようなので GEM としては忘れる。状況が変われば reopen するが一旦 close します。

closed 2017-01-12 by dynamis

ashie, 2017-06-26:
GrapheneをESR52に移植してみたので、以下に置いておく。

https://github.com/mozilla-japan/gecko-dev/tree/esr52-graphene