webdino/gecko-embedded #29: Wandboard: メニュー類が表示されない

ashie, 2016-10-14:
現在のレシピをWandboard向けにビルドしてみたところ、メニュー類のボタンを押してもメニューが表示されなくなっていた。

ashie, 2016-10-14:
Waylandパッチを全部外しても同様の問題が発生する。
Firefox自体の問題?

（X11の場合、デフォルトでは少しでもフットプリントを小さくするためにgtk2でビルドするようにしている。しかしWandboardのcore-image-satoでは`DISTRO_FEATURES`に`x11`と`wayland`の両方が入るので、waylandパッチを当ててgtk3でビルドされる。waylandパッチの影響かと思って外してみたが、同様の問題が発生する。）

ashie, 2016-10-14:
過去にビルドしたイメージを確認すると、少なくとも45.2.0のレシピでは動作していた。

naobsd, 2016-10-17:
historyとかpreferencesとかがある右上のボタンでしょうか？
96boards (krogoth) 上では 45.4 で特に問題ない、ように見えます

ashie, 2016-10-18:
> historyとかpreferencesとかがある右上のボタンでしょうか？

はい、主にそれです。

> 96boards (krogoth) 上では 45.4 で特に問題ない、ように見えます

確認ありがとうございます！助かります！
ちなみに確認されたのはX11でしょうか? Waylandでしょうか?
また、X11の場合、GTK2でしょうか? GTK3でしょうか?
デフォルトでは以下のようにビルドされるので、`bitbake -e | grep "DISTRO_FEATURES="`を教えて頂けると助かります。

1. DISTRO_FEATUERS="x11": gtk2
2. DISTRO_FEATURES="x11 wayland": gtk3 + waylandパッチ
3. DISTRO_FEATURES="wayland": gtk3 + waylandパッチ

手元では、

1. 未確認
2. （X11上で）問題発生
3. （Wayland上で）問題なし

という状況です。

naobsd, 2016-10-18:
ドロップダウン(?)メニュー自体は表示されるが、表示されたメニューアイテムをクリックしても画面が遷移しない、という問題でしょうか？

1(x11)では問題なかったですが、2(x11 wayland)では発生しています。

ashie, 2016-10-18:
> ドロップダウン(?)メニュー自体は表示されるが、表示されたメニューアイテムをクリックしても画面が遷移しない、という現象でしょうか？

いえ、このissueで対象としている問題はドロップダウンメニューも表示されないという現象です。

> 1(x11)では問題なかったですが、2(x11 wayland)では発生しています。

2.の場合はX11でもWaylandでも起動できるバイナリができるのですが、これはWaylandで起動した場合でしょうか? Waylandでメニューアイテムをクリックしても画面が遷移しないのは既知の問題です（#2）。X11で発生する場合は、新たな問題です。

ちなみに3.の場合はビルドできないので、ここであげるのは不適切でした...

naobsd, 2016-10-18:
なるほど、となると2(wayland/westonで起動)の場合もこのissueは発生していないですね。

naobsd, 2016-10-18:
2でXで起動、はまだ試していません

ashie, 2016-10-18:
本現象を確認したボードで再度試したところ、現象を再現できなくなりました。
なんだったんだろう...?

いったんクローズします。再度発生するようであればReopenします。

closed 2016-10-18 by ashie