webdino/gecko-embedded #40: gem-tanzaniteデモ用レシピ作成

ashie, 2016-12-28:
ルネサスRZ/Gシリーズで[gem-tanzanite](https://github.com/gem-tanzanite/gem-tanzanite.github.io) を用いたデモを行うために手元でSDブートイメージを作成して使用しているが、現状ではレシピだけでは同環境を再現することができず、手動での設定も必要となっている。

[meta-rzg-demos](https://github.com/renesas-rz/meta-rzg-demos)と同様に、bitbake一発でデモ環境を構築できるようにレシピを整備する。

ashie, 2016-12-28:
#20 で作成するレシピ (https://github.com/mozilla-japan/meta-gecko-embedded) 内にデモ用のレイヤを入れる予定。

ashie, 2017-01-20:
TODO

* [ ] westonに`--idle-time=0`オプションを付ける
* [ ] 動画関連カーネルモジュールのロード（最新のmeta-renesasに入っているmm-init）
* [ ] gem-tanzaniteのレシピ追加
* [ ] Firefoxの起動アイコン or 自動起動
* [ ] 小さい画面の場合にフルスクリーンで起動できないことがある問題の対応

ashie, 2017-06-02:
> westonに--idle-time=0オプションを付ける（最新のmeta-renesasには入っている）

現在のmeta-gecko-embeddedと最新のmeta-renesasの組み合わせで動くようなので、特に対応の必要なし。

ashie, 2017-06-02:
> 動画関連カーネルモジュールのロード（最新のmeta-renesasに入っているmm-init）

Yocto-1.6、2.0のどちらも、最新版ではデフォルトで組み込まれているので特に対応の必要は無し。

ashie, 2017-06-13:
> 小さい画面の場合にフルスクリーンで起動できないことがある問題の対応

に関しては

> 専用プロファイル

xulstore.jsonに`"sizemode": "fullscreen"`を入れておくことで対処する。

ashie, 2017-06-13:
> Firefoxでタッチイベントのデフォルト有効化 (firefoxへのパッチ)

ad-hoc過ぎて扱いに困っていたのでどこにも入れてなかったが、

> 専用プロファイル

これと組み合わせて有効化できるようにして各ブランチに入れた。

APZを有効化できるようになればデフォルトでタッチイベントは有効化されるので、
いずれこんなパッチは不要になるはず。

ashie, 2017-06-14:
> 専用プロファイル
> gem-tanzaniteのレシピ追加

レシピを投入した:

https://github.com/mozilla-japan/meta-gecko-embedded/commit/9448703a6a71906e20db9ab061bdf83a0a280e0b

cosmo0920, 2017-06-14:
ESR52でe10sを有効化してもクラッシュしなくなるパッチ https://bugzilla.redhat.com/attachment.cgi?id=1287203&action=diff を投入してもabout:xxx以外は開けない、という結果になったので

>> Firefoxでタッチイベントのデフォルト有効化 (firefoxへのパッチ)

> ad-hoc過ぎて扱いに困っていたのでどこにも入れてなかった

このパッチはESR52では必要そうです。

ashie, 2017-06-14:
> Firefoxの起動アイコン or 自動起動

initスクリプトのレシピを追加（gem-tanzanite-init）
あとはcore-image-gem-tanzaniteとかも追加しておいた方が親切か?

ashie, 2017-06-14:
> あとはcore-image-gem-tanzaniteとかも追加しておいた方が親切か?

gem-tanzaniteに限らず色々なデモを入れたイメージにしたい。
ということで別に考えるとして、本issueはこれにて終了。

ashie, 2017-06-14:
ドキュメントが無いので、ビルド手順におまけとして記載しておく。

closed 2017-06-14 by ashie

ashie, 2017-06-14:
> ドキュメントが無いので、ビルド手順におまけとして記載しておく。

以下に記載した: 

https://github.com/mozilla-japan/meta-browser/wiki/Build-RainboW-G20D-Q7