webdino/gecko-embedded #41: AM572x Evaluation Module向けビルド

ashie, 2017-01-05:
[AM572x Evaluation Module](http://www.ti.com/tool/tmdsevm572x)が入手できそうなので、ビルドを試してみる。

SDKビルド手順: http://processors.wiki.ti.com/index.php/Processor_SDK_Building_The_SDK

ashie, 2017-01-10:
以下の問題が発生

* http://e2e.ti.com/support/arm/automotive_processors/f/1021/t/535904
* https://e2e.ti.com/support/arm/sitara_arm/f/791/t/503804

/bin/shをdashからbashに変えないとビルドできないらしい:

```
$ sudo dpkg-reconfigure dash

(Select "no" when prompted)
```

シェルを変更させるのではなくてコードの方を直してほしいのだが、上記が公式見解になってしまっている?

ashie, 2017-01-10:
以前ルネサスの方でも発生していた、Linaroのパスが変更されてダウンロードに失敗する問題がこちらでも発生した（ルネサスの方は既に直っている）。

```
ERROR: gdb-linaro-7.8-r2014.09.arago0 do_fetch: Fetcher failure: Fetch command failed with exit code 8, output:
https://releases.linaro.org/14.09/components/toolchain/gdb-linaro/gdb-linaro-7.8-2014.09.tar.xz:
2017-01-10 17:54:49 ERROR 404: NOT FOUND.
```

今は https://releases.linaro.org/archives/ 以下に移動されている。

ashie, 2017-01-11:
SDKは上記2点を回避すればビルドできた。

ashie, 2017-01-11:
Firefoxをビルドするには現状では`DISTRO_FEATURES`に`x11`が必要。
他のボードのBSPだとlocal.confに以下を追加するだけでビルドできたが

```
DISTRO_FEATURES_append = " x11 "
```

このBSPだと上記を加えても` DISTRO_FEATURES` に `x11` が追加されなかった。

sources/meta-arago/meta-arago-distro/conf/distro/arago.conf に

```
DISTRO_FEATURES_remove = "x11"
```

があるのでひとまずこれをコメントアウトしてみたところ、x11が追加された状態でビルドが
走るようになったものの、いくつかのパッケージでビルドエラーが発生する。

ashie, 2017-01-11:
meta-arago-distro/conf/distro/include/arago-prefs.inc で以下のようになっているが

```
PREFERRED_PROVIDER_virtual/egl = "ti-sgx-ddk-um"
PREFERRED_PROVIDER_virtual/libgles1 = "ti-sgx-ddk-um"
PREFERRED_PROVIDER_virtual/libgles2 = "ti-sgx-ddk-um"
PREFERRED_PROVIDER_libgbm = "libgbm"
PREFERRED_PROVIDER_libgbm-dev = "libgbm-dev"
PREFERRED_PROVIDER_virtual/libgl = "mesa-gl"
PREFERRED_PROVIDER_virtual/mesa = "mesa-gl"
```

libgbm と mesa-gl を同時にインストールするケースが発生すると、両者が競合するようだ。
(waylandだけだとmesa-glはインストールされない)

これを修正あるいは回避するのは簡単にはいかなそう。

ashie, 2017-01-11:
このボードに対応するためには、

* BSP側を修正して`DISTRO_FEATURES`に`x11`を追加できるようにする
* Firefox側でxlib依存を完全に排除する対応を完了させる

のいずれか（あるいは両方）の選択肢があるが、いずれにしてもそこそこ工数がかかりそう。
本プロジェクトの趣旨としては後者に力を入れるのがスジか。

ashie, 2017-01-11:
mesa-glのbitbake時のエラー内容を貼り付けておく

```
ERROR: mesa-gl-2_11.1.1-r0 do_packagedata: The recipe mesa-gl is trying to install files into a shared area when those file
s already exist. Those files and their manifest location are:
   /home/aho/Projects/tisdk/build/arago-tmp-external-linaro-toolchain/sysroots/am57xx-evm/pkgdata/runtime/libgbm-dev
 Matched in manifest-am57xx-evm-libgbm.packagedata
 /home/aho/Projects/tisdk/build/arago-tmp-external-linaro-toolchain/sysroots/am57xx-evm/pkgdata/runtime/libgbm
 Matched in manifest-am57xx-evm-libgbm.packagedata
Please verify which recipe should provide the above files.
```

```
ERROR: The recipe mesa-gl is trying to install files into a shared area when those files already exist. Those files and the
ir manifest location are:
   /home/aho/Projects/tisdk/build/arago-tmp-external-linaro-toolchain/sysroots/am57xx-evm/usr/include/GL/glext.h
 Matched in manifest-am57xx-evm-ocl-gl-headers.populate_sysroot
 /home/aho/Projects/tisdk/build/arago-tmp-external-linaro-toolchain/sysroots/am57xx-evm/usr/include/GL/wglext.h
 Matched in manifest-am57xx-evm-ocl-gl-headers.populate_sysroot
 /home/aho/Projects/tisdk/build/arago-tmp-external-linaro-toolchain/sysroots/am57xx-evm/usr/include/GL/glxext.h
 Matched in manifest-am57xx-evm-ocl-gl-headers.populate_sysroot
 /home/aho/Projects/tisdk/build/arago-tmp-external-linaro-toolchain/sysroots/am57xx-evm/usr/include/GL/osmesa.h
 Matched in manifest-am57xx-evm-ocl-gl-headers.populate_sysroot
 /home/aho/Projects/tisdk/build/arago-tmp-external-linaro-toolchain/sysroots/am57xx-evm/usr/include/GL/gl_mangle.h
 Matched in manifest-am57xx-evm-ocl-gl-headers.populate_sysroot
 /home/aho/Projects/tisdk/build/arago-tmp-external-linaro-toolchain/sysroots/am57xx-evm/usr/include/GL/gl.h
 Matched in manifest-am57xx-evm-ocl-gl-headers.populate_sysroot
 /home/aho/Projects/tisdk/build/arago-tmp-external-linaro-toolchain/sysroots/am57xx-evm/usr/include/GL/glx.h
 Matched in manifest-am57xx-evm-ocl-gl-headers.populate_sysroot
 /home/aho/Projects/tisdk/build/arago-tmp-external-linaro-toolchain/sysroots/am57xx-evm/usr/include/GL/glx_mangle.h
 Matched in manifest-am57xx-evm-ocl-gl-headers.populate_sysroot
 /home/aho/Projects/tisdk/build/arago-tmp-external-linaro-toolchain/sysroots/am57xx-evm/usr/include/GL/internal/dri_interface.h
 Matched in manifest-am57xx-evm-ocl-gl-headers.populate_sysroot
Please verify which recipe should provide the above files.
```

ashie, 2017-01-11:
Waylandは1.9、GTK+は3.18なので、x11対応さえできればFirefox on Waylandはすぐにそこそこは動きそうではあるが...

ashie, 2017-01-11:
ブート用SDカードの作成は以下辺りを参照したが、手元のSDKだとドキュメント通りになっていない部分があってよくわからなかった。

http://processors.wiki.ti.com/index.php/Processor_SDK_RTOS_create_SD_card_script
http://processors.wiki.ti.com/index.php/Processor_SDK_RTOS_BOOT_AM57x

とりあえず添付のSDカードを参考に

* 第一パーティション: vfat, u-boot.imgを配置
* 第二パーティション: ext3, rootfs(tisdk-rootfs-image-am57xx-evm.tar.xz)を配置

でブートした。

dynamis, 2017-01-11:
> このボードに対応するためには、
> 
>     BSP側を修正してDISTRO_FEATURESにx11を追加できるようにする
>     Firefox側でxlib依存を完全に排除する対応を完了させる
> 
> のいずれか（あるいは両方）の選択肢があるが、いずれにしてもそこそこ工数がかかりそう。
> 本プロジェクトの趣旨としては後者に力を入れるのがスジか。

そうですね。BSP 側の修正で x11 も対応できるようにと言うのはちょっとやりにくそうとか、x11 なしでビルドされた方が結果としても当然良いので、Firefox on Wayland が X11 (xlib) 依存の無い形に持って行くのが本来あるべき方向ですね。今すぐこの作業に Priority を付けて時間は取れないが、何とかして誰かがこれを解決できる体制にしたいですね。

ashie, 2017-05-29:
ボードを手放していて当面対応予定は無いのでいったんクローズする。
再開するようなリオープンする。

closed 2017-05-29 by ashie