webdino/gecko-embedded #45: meta-gecko-embeddedのYocto2.0対応

ashie, 2017-04-12:
#20 でRZ/G向けにGTK+を更新するために meta-gecko-embedded というレイヤを作成したが、
これは renesas-rzg/meta-renesas の Yocto-1.6 向けだった。
meta-renesas に yocto_2.0 ブランチが出来ているので、meta-gecko-embededdedをこちらに
対応させる。
#2 に書いた通りWayland関係の問題を修正するには少しでも新しいWaylandおよびGTK+を使用
した方が良く、Yocto-1.6のままではこれらをアップデートすることが難しいため、Yocto-1.6対応
を維持させることは必ずしも考えなくて良い。

ashie, 2017-04-19:
meta-gecko-embededded/meta-poky に含まれるレシピのうち、最低限以下だけを適用すれば、
Yocto-2.0でも、Yocto-1.6の時と同様にgtk+-3.18へアップグレードできる。
（ベースが上がっているので、他のパッケージは必ずしもアップグレードしなくてもビルドは通るようになった）
この構成で、FirefoxもYocto-1.6の時とほぼ同様に動作することを確認できた。

* glib-2.0
* pango
* gtk+3: (downgrade-xdg-shell.patchは不要、と言うより有害)
* adwaita-icon-thema

RZ/GのGPUドライバが更新されているので、WaylandおよびGTK+をもう少し新しくできる可能性がある。

cosmo0920, 2017-04-19:
ESR52向けのWaylandのパッチは少なくともgtk+3 3.20以降が前提になっているので Yocto-2.0以降 + gtk+3の3.20へのバージョンアップ を行わないと動作させるのが難しいと思います。

ashie, 2017-04-19:
了解です。ただ、Wayland + GTKを上げられるかどうかはGPUドライバが動作するかどうかにかかっているので、
上げられない場合はGTK+-3.18のままで対応するというオプションも考えておいて方が良いかも。
（その場合は、既知のバグは別の方法で回避する、GdkSeatは使わないように修正するなど）

cosmo0920, 2017-04-19:
なるほど。GdkSeatは今現在使っているパッチになっていないので3.18以上を要求するようにして試してみます。

ashie, 2017-04-24:
RZ/GのYocto-2.0ベースのBSPについて、まだどこにも情報をまとめていなかったと思うのでメモ:

meta-renesas/meta-rzg1のYocto-2.0ベースのレシピは以下のブランチにある。

* https://github.com/renesas-rz/meta-renesas/tree/yocto_2.0/meta-rzg1

ビルド方法は以下に記載されている。

* http://elinux.org/RZ-G/Boards/Yocto_2.0

上記の手順で問題なくビルド出来ることは確認できている。
ブートについても基本的に以下の手順で大丈夫だが

* [TFTPブート](http://elinux.org/RZ-G/Boards/Yocto#Loading_kernel_via_TFTP_and_rootfs_via_NFS)
* [microSDブート](http://elinux.org/RZ-G/Boards/SK-RZG1E#Configure_U-Boot_to_boot_over_TFTP_.2B_NFS_or_from_a_micro_SD_card)

microSDブートについてはデバイス番号がmmcblk0からmmcblk1に変更されているので、読み替える必要がある。

```diff
-setenv bootargs 'console=ttySC10,38400 ignore_loglevel rw rootfstype=ext3 root=/dev/mmcblk0p1 rootwait vmalloc=384M'
+setenv bootargs 'console=ttySC10,38400 ignore_loglevel rw rootfstype=ext3 root=/dev/mmcblk1p1 rootwait vmalloc=384M'

```

ashie, 2017-04-24:
Yocto 2.0に対応したmeta-gecko-embeddedのとりあえずのものを以下のブランチに置いておく。

https://github.com/mozilla-japan/meta-gecko-embedded/tree/yocto_2.0

masterでもいいかもしれないが、レシピ名をどうするか等をまだちょっと悩んでいるので。

ashie, 2017-04-27:
> Yocto 2.0に対応したmeta-gecko-embeddedのとりあえずのものを以下のブランチに置いておく。

Yocto-2.0用のレシピはこれまでのmeta-pokyではなくてmeta-poky-jethroというディレクトリに分けたので

* https://github.com/mozilla-japan/meta-browser/wiki/Build-RZ-G1E

の手順は以下の部分を変更する必要がある（将来的にまた変更するかもしれないが）。

```
-BBLAYERS += " ${TOPDIR}/../meta-gecko-embedded/meta-poky "
+BBLAYERS += " ${TOPDIR}/../meta-gecko-embedded/meta-poky-jethro "
```

ashie, 2017-04-28:
メモ: meta-rcar-gen3には既にWayland 1.11が投入されている

https://github.com/renesas-rcar/meta-renesas

ashie, 2017-04-28:
> メモ: meta-rcar-gen3には既にWayland 1.11が投入されている

というかkrogothベースになっていた。

ashie, 2017-04-28:
Wayland + GTKをとこまで上げられるか確認してみた。

* Wayland-1.10 + GTK+3.20: Firefoxが動作することを確認
* Wayland-1.11: westonが起動せず
    * failed to choose EGL config for output

最低目標のGTK+3.20は行けそうだが、それ以上はやはりドライバも含めて上げないとダメか?

ashie, 2017-05-01:
ひとまず以下のブランチにwayland-1.10に上げた版を置いておく。

https://github.com/mozilla-japan/meta-gecko-embedded/tree/yocto_2.0-upgrade-gtk

ashie, 2017-05-01:
> masterでもいいかもしれないが、レイヤ名をどうするか等をまだちょっと悩んでいるので。

汎用的にできるかと思ったが、結局RZ/G1専用なのでmeta-rzg1にした方が良さそう。

ashie, 2017-05-01:
> * https://github.com/mozilla-japan/meta-browser/wiki/Build-RZ-G1E
> 
> の手順は以下の部分を変更する必要がある（将来的にまた変更するかもしれないが）。

もう一点忘れていた。
yocto_2.0ブランチの場合はlocal.confの以下の行があるが、コメントアウトする必要がある。

```
DISTRO_FEATURES_remove = "x11"
```

ashie, 2017-05-26:
ビルド・動作は確認できているので、ビルド手順をWikiにまとめた後、本issueはクローズする。

ashie, 2017-05-29:
meta-renesasは１つ問題があって、環境によってはビルドが失敗して先に進めないことがある。
以下に修正版をpushしてある。

https://github.com/ashie/meta-renesas/tree/yocto_2.0-fix-gst-plugins-bad-build-issue

upstreamには報告済みだが、

https://github.com/renesas-rz/meta-renesas/pull/5

開発中の版を公開してくれる気配が無いので、修正版がupstream側でリリースされるのは当分先かもしれない。

ashie, 2017-05-29:
ビルド方法を以下にまとめたので、本issueはクローズする。

https://github.com/mozilla-japan/meta-browser/wiki/Build-RZ-G1E-Yocto2.0

closed 2017-05-29 by ashie