webdino/gecko-embedded #46: 終了時にクラッシュすることがある

ashie, 2017-05-26:
https://github.com/mozilla-japan/gecko-embedded/issues/2#issuecomment-304246542 より

wayland版で終了時にクラッシュすることがある

cosmo0920, 2017-05-26:
補足です。このクラッシュはCtrl+Qでは起こらず、タイトルバーの終了ボタンを押すことにより発生します。
EGLウィンドウでは現在タイトルバーを消しているのでメニューバーの ファイル(F)＞終了(Q) ではCtrl+Qと同様に処理されるため、この問題は発生しません。

ashie, 2017-06-15:
```
Program received signal SIGSEGV, Segmentation fault.
0x4613ffbc in gtk_widget_get_window (widget=0xa96994b0) at /usr/src/debug/gtk+3/3.20.9-r0/gtk+-3.20.9/gtk/gtkwidget.c:15947
15947	  g_return_val_if_fail (GTK_IS_WIDGET (widget), NULL);
(gdb) where
#0  0x4613ffbc in gtk_widget_get_window (widget=0xa96994b0) at /usr/src/debug/gtk+3/3.20.9-r0/gtk+-3.20.9/gtk/gtkwidget.c:15947
#1  0x460f0798 in _gtk_widget_find_at_coords (window=window@entry=0xa27c6e40, window_x=<optimized out>, window_y=<optimized out>, widget_x=widget_x@entry=0xbeffda9c, 
    widget_y=widget_y@entry=0xbeffdaa0) at /usr/src/debug/gtk+3/3.20.9-r0/gtk+-3.20.9/gtk/gtktooltip.c:637
#2  0x460f14fc in gtk_tooltip_show_tooltip (display=display@entry=0xb6d241a0) at /usr/src/debug/gtk+3/3.20.9-r0/gtk+-3.20.9/gtk/gtktooltip.c:1117
#3  0x460f19e0 in tooltip_popup_timeout (data=0xb6d241a0) at /usr/src/debug/gtk+3/3.20.9-r0/gtk+-3.20.9/gtk/gtktooltip.c:1228
#4  0x45c1956c in gdk_threads_dispatch (data=<optimized out>) at /usr/src/debug/gtk+3/3.20.9-r0/gtk+-3.20.9/gdk/gdk.c:720
#5  0xb696887c in g_timeout_dispatch (source=0x973d1540, callback=0x45c19540 <gdk_threads_dispatch>, user_data=<optimized out>)
    at /usr/src/debug/glib-2.0/1_2.46.2-r0/glib-2.46.2/glib/gmain.c:4577
#6  0xb6967c1c in g_main_dispatch (context=0xb6d206a0) at /usr/src/debug/glib-2.0/1_2.46.2-r0/glib-2.46.2/glib/gmain.c:3154
#7  g_main_context_dispatch (context=context@entry=0xb6d206a0) at /usr/src/debug/glib-2.0/1_2.46.2-r0/glib-2.46.2/glib/gmain.c:3769
#8  0xb6967fb0 in g_main_context_iterate (context=context@entry=0xb6d206a0, block=block@entry=0, dispatch=dispatch@entry=1, self=<optimized out>)
    at /usr/src/debug/glib-2.0/1_2.46.2-r0/glib-2.46.2/glib/gmain.c:3840
#9  0xb696805c in g_main_context_iteration (context=0xb6d206a0, may_block=0) at /usr/src/debug/glib-2.0/1_2.46.2-r0/glib-2.46.2/glib/gmain.c:3901
#10 0xb4fcdb90 in nsAppShell::ProcessNextNativeEvent(bool) () from /usr/lib/firefox-52.1.0/libxul.so
#11 0xb4fa4960 in nsBaseAppShell::DoProcessNextNativeEvent(bool) () from /usr/lib/firefox-52.1.0/libxul.so
#12 0xb4fa6ee0 in nsBaseAppShell::OnProcessNextEvent(nsIThreadInternal*, bool) () from /usr/lib/firefox-52.1.0/libxul.so
#13 0xb3ab181c in nsThread::ProcessNextEvent(bool, bool*) () from /usr/lib/firefox-52.1.0/libxul.so
#14 0xb3abb4b8 in NS_InvokeByIndex () from /usr/lib/firefox-52.1.0/libxul.so
#15 0xbeffddc8 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

ashie, 2017-06-15:
> #2  0x460f14fc in gtk_tooltip_show_tooltip (display=display@entry=0xb6d241a0) at /usr/src/debug/gtk+3/3.20.9-r0/gtk+-3.20.9/gtk/gtktooltip.c:1117

クローズボタンにマウスカーソルを重ねると「Close」というツールチップが出るが、これが出る前に押下するとクラッシュするようだ。
問題はGTK+側か?

ashie, 2017-06-15:
> 問題はGTK+側か?

いや、gtk3-demoではそんなのは出ないので、やはりFirefox側?

ashie, 2017-06-16:
ボタンの在り処はFirefoxの base/content/browser.xul:

```xml
        <toolbarbutton id="close-button"
                       tooltiptext="&fullScreenClose.tooltip;"
                       oncommand="BrowserTryToCloseWindow();"/>
 ```

ashie, 2017-06-16:
> クローズボタンにマウスカーソルを重ねると「Close」というツールチップが出るが、これが出る前に押下するとクラッシュするようだ。

この再現条件嘘だ。出る前に押してもクラッシュするときとしないときがある。

ashie, 2017-06-16:
> #3  0x460f19e0 in tooltip_popup_timeout (data=0xb6d241a0) at /usr/src/debug/gtk+3/3.20.9-r0/gtk+-3.20.9/gtk/gtktooltip.c:1228

Firefox側ではこのルートでツールチップをセットする箇所が無いように見えるので、
Firefox側のツールチップは関係なさそう。

ashie, 2017-06-16:
> クローズボタンにマウスカーソルを重ねると「Close」というツールチップが出るが、これが出る前に押下するとクラッシュするようだ。

Firefoxのツールチップに関係なく、しばらくマウスカーソルを止めてから押下するとクラッシュしない、
というのは恐らく正しい。ツールチップのタイムアウトの実行が終わっているため。
ツールチップのタイムアウトはgtk/gtkmain.cの以下でセットされている:

```
  if (event->type == GDK_ENTER_NOTIFY
      || event->type == GDK_LEAVE_NOTIFY
      || event->type == GDK_BUTTON_PRESS
      || event->type == GDK_2BUTTON_PRESS
      || event->type == GDK_3BUTTON_PRESS
      || event->type == GDK_KEY_PRESS
      || event->type == GDK_DRAG_ENTER
      || event->type == GDK_GRAB_BROKEN
      || event->type == GDK_MOTION_NOTIFY
      || event->type == GDK_TOUCH_UPDATE
      || event->type == GDK_SCROLL)
    {
      _gtk_tooltip_handle_event (event);
    }
```

キーボード・ショートカットで落ちないのは、マウスを動かしていないのでタイムアウトがセットされないからだろう。

ashie, 2017-06-16:
>       _gtk_tooltip_handle_event (event);

これを殺すとクラッシュしなくなる。
また、Firefox側のツールチップは出るので、

> Firefox側ではこのルートでツールチップをセットする箇所が無いように見えるので、
> Firefox側のツールチップは関係なさそう。

これは正解。

gtk_tooltip.cのコードが、破壊済みwidgetにアクセスし得るのでスジが悪いように見えるが...?

ashie, 2017-06-19:
> キーボード・ショートカットで落ちないのは、マウスを動かしていないのでタイムアウトがセットされないからだろう。

マウスを動かしながらだと、やはり再現する。

ashie, 2017-06-19:
> gtk_tooltip.cのコードが、破壊済みwidgetにアクセスし得るのでスジが悪いように見えるが...?

GdkWindowが存在して対応するwidgetが破壊済みというのはあるべきではない（先にGdkWindowを破壊すべき）ので、その意味ではFirefox側の問題かもしれないが、切り分け切れていない。

ashie, 2017-06-28:
```diff
diff -Nur gtk+-3.20.9-bak/gtk/gtktooltip.c gtk+-3.20.9/gtk/gtktooltip.c
--- gtk+-3.20.9-bak/gtk/gtktooltip.c    2016-08-04 23:34:04.000000000 +0900
+++ gtk+-3.20.9/gtk/gtktooltip.c        2017-06-28 17:58:13.618923333 +0900
@@ -1105,6 +1105,8 @@
 
       if (!GDK_IS_WINDOW (window))
         return;
+      if (gdk_window_is_destroyed (window))
+        return;
 
       device = gdk_seat_get_pointer (gdk_display_get_default_seat (display));
```

こんな感じでとりあえずクラッシュは避けられるようだ。
やはり、timeoutをセットしながらdestroyを監視していないGTK+側の方が行儀が悪い感じがする。
もうちょっとちゃんとやるなら、destroyシグナルを監視すべきか。

ashie, 2017-06-28:
上記パッチを投入したので、クローズしておく。

https://github.com/mozilla-japan/meta-gecko-embedded/commit/1d80b365282371ce54062aff890ea3a3c07cc496

あくまでも対症療法ではあるけど、upstreamにも投げてみるか...

closed 2017-06-28 by ashie

ashie, 2017-06-29:
https://bugzilla.gnome.org/show_bug.cgi?id=784319

ashie, 2017-07-14:
FirefoxとGTK+3にアップストリームの修正を入れたところ、当初のクラッシュは修正できたが、
稀に以下のようなクラッシュが発生するようになった。

```
#0  raise (sig=11) at ../sysdeps/unix/sysv/linux/pt-raise.c:38
#1  0xb4f67124 in nsProfileLock::FatalSignalHandler(int, siginfo_t*, void*) () from /usr/lib/firefox-52.2.0/libxul.so
#2  <signal handler called>
#3  wl_proxy_marshal (proxy=proxy@entry=0x0, opcode=1) at /usr/src/debug/wayland/1.10.0-r0/wayland-1.10.0/src/wayland-client.c:679
#4  0xb6421ef0 in wl_surface_attach (y=<optimized out>, x=<optimized out>, buffer=<optimized out>, wl_surface=0x0)
    at /build/rz-g-bsp-yocto-2.0/build/tmp/sysroots/skrzg1e/usr/include/wayland-client-protocol.h:1857
#5  gdk_wayland_window_attach_image (window=window@entry=0x94615240) at /usr/src/debug/gtk+3/3.20.9-r0/gtk+-3.20.9/gdk/wayland/gdkwindow-wayland.c:671
#6  0xb6421fc8 in gdk_window_impl_wayland_end_paint (window=0x94615240) at /usr/src/debug/gtk+3/3.20.9-r0/gtk+-3.20.9/gdk/wayland/gdkwindow-wayland.c:838
#7  0xb63e7e50 in gdk_window_end_paint (window=0x94615240) at /usr/src/debug/gtk+3/3.20.9-r0/gtk+-3.20.9/gdk/gdkwindow.c:3124
#8  0xb66707d0 in gtk_main_do_event (event=0xbe880528) at /usr/src/debug/gtk+3/3.20.9-r0/gtk+-3.20.9/gtk/gtkmain.c:1799
#9  0xb63ce578 in _gdk_event_emit (event=event@entry=0xbe880528) at /usr/src/debug/gtk+3/3.20.9-r0/gtk+-3.20.9/gdk/gdkevents.c:73
#10 0xb63deda4 in _gdk_window_process_updates_recurse_helper (window=0x94615240, expose_region=<optimized out>)
    at /usr/src/debug/gtk+3/3.20.9-r0/gtk+-3.20.9/gdk/gdkwindow.c:3609
#11 0xb63e007c in gdk_window_process_updates_internal (window=0x94615240) at /usr/src/debug/gtk+3/3.20.9-r0/gtk+-3.20.9/gdk/gdkwindow.c:3755
#12 0xb63e024c in gdk_window_process_updates_with_mode (window=<optimized out>, recurse_mode=<optimized out>)
    at /usr/src/debug/gtk+3/3.20.9-r0/gtk+-3.20.9/gdk/gdkwindow.c:3956
#13 0x47bded70 in _g_closure_invoke_va (closure=closure@entry=0x9a9f9fb0, return_value=return_value@entry=0x0, instance=0x0, instance@entry=0x98e3a598, 
    args=args@entry=..., n_params=0, param_types=0x0) at /usr/src/debug/glib-2.0/1_2.46.2-r0/glib-2.46.2/gobject/gclosure.c:864
#14 0x47bfa9f4 in g_signal_emit_valist (instance=instance@entry=0x98e3a598, signal_id=<optimized out>, detail=detail@entry=0, var_args=..., var_args@entry=...)
    at /usr/src/debug/glib-2.0/1_2.46.2-r0/glib-2.46.2/gobject/gsignal.c:3292
#15 0x47bfafa8 in g_signal_emit (instance=instance@entry=0x98e3a598, signal_id=<optimized out>, detail=0)
    at /usr/src/debug/glib-2.0/1_2.46.2-r0/glib-2.46.2/gobject/gsignal.c:3439
#16 0xb63d6e50 in _gdk_frame_clock_emit_paint (frame_clock=frame_clock@entry=0x98e3a598) at /usr/src/debug/gtk+3/3.20.9-r0/gtk+-3.20.9/gdk/gdkframeclock.c:640
#17 0xb63d7704 in gdk_frame_clock_paint_idle (data=0x98e3a598) at /usr/src/debug/gtk+3/3.20.9-r0/gtk+-3.20.9/gdk/gdkframeclockidle.c:430
#18 0xb63c356c in gdk_threads_dispatch (data=<optimized out>) at /usr/src/debug/gtk+3/3.20.9-r0/gtk+-3.20.9/gdk/gdk.c:720
#19 0x4760887c in g_timeout_dispatch (source=0x9505ca40, callback=0xb63c3540 <gdk_threads_dispatch>, user_data=<optimized out>)
    at /usr/src/debug/glib-2.0/1_2.46.2-r0/glib-2.46.2/glib/gmain.c:4577
#20 0x47607c1c in g_main_dispatch (context=0xb6d206a0) at /usr/src/debug/glib-2.0/1_2.46.2-r0/glib-2.46.2/glib/gmain.c:3154
#21 g_main_context_dispatch (context=context@entry=0xb6d206a0) at /usr/src/debug/glib-2.0/1_2.46.2-r0/glib-2.46.2/glib/gmain.c:3769
#22 0x47607fb0 in g_main_context_iterate (context=context@entry=0xb6d206a0, block=block@entry=0, dispatch=dispatch@entry=1, self=<optimized out>)
    at /usr/src/debug/glib-2.0/1_2.46.2-r0/glib-2.46.2/glib/gmain.c:3840
#23 0x4760805c in g_main_context_iteration (context=0xb6d206a0, may_block=0) at /usr/src/debug/glib-2.0/1_2.46.2-r0/glib-2.46.2/glib/gmain.c:3901
#24 0xb48b3150 in nsAppShell::ProcessNextNativeEvent(bool) () from /usr/lib/firefox-52.2.0/libxul.so
#25 0xb488a018 in nsBaseAppShell::DoProcessNextNativeEvent(bool) () from /usr/lib/firefox-52.2.0/libxul.so
#26 0xb488c534 in nsBaseAppShell::OnProcessNextEvent(nsIThreadInternal*, bool) () from /usr/lib/firefox-52.2.0/libxul.so
#27 0xb3399774 in nsThread::ProcessNextEvent(bool, bool*) () from /usr/lib/firefox-52.2.0/libxul.so
#28 0xb33b7ab8 in NS_ProcessNextEvent(nsIThread*, bool) () from /usr/lib/firefox-52.2.0/libxul.so
#29 0xb371f468 in mozilla::ipc::MessagePump::Run(base::MessagePump::Delegate*) () from /usr/lib/firefox-52.2.0/libxul.so
#30 0xb3708240 in MessageLoop::Run() () from /usr/lib/firefox-52.2.0/libxul.so
#31 0xb488832c in nsBaseAppShell::Run() () from /usr/lib/firefox-52.2.0/libxul.so
#32 0xb4f19c74 in nsAppStartup::Run() () from /usr/lib/firefox-52.2.0/libxul.so
#33 0xb4f6e5f0 in XREMain::XRE_mainRun() () from /usr/lib/firefox-52.2.0/libxul.so
#34 0xb4f6e8d4 in XREMain::XRE_main(int, char**, nsXREAppData const*) () from /usr/lib/firefox-52.2.0/libxul.so
#35 0xb4f6eb14 in XRE_main () from /usr/lib/firefox-52.2.0/libxul.so
#36 0x00014a98 in do_main(int, char**, char**, nsIFile*) [clone .constprop.4] ()
---Type <return> to continue, or q <return> to quit---
#37 0x0001410c in main ()
```

Oliverは https://bugzilla.gnome.org/show_bug.cgi?id=773274 の問題では?と言っているが、パッチを当てても発生する。が、Fedora 26では発生しないので、まだGTK+3のパッチが足りていない可能性もある。