webdino/gecko-embedded #47: ナビゲーションのショートカットキー（ALT + ← or →）が効いていない

ashie, 2017-05-26:
https://github.com/mozilla-japan/gecko-embedded/issues/2#issuecomment-304246542 より

Wayland版で ALT + ← or → が効いていない

ashie, 2017-05-29:
GTK+側の問題の気がしているが、まだちゃんとは見ていない。

cosmo0920, 2017-06-28:
Firefox側ではXULの中で定義されている。
具体的にはこの辺り https://dxr.mozilla.org/mozilla-central/source/browser/base/content/browser-sets.inc#270

ashie, 2017-07-12:
キー修飾のハンドリングはGDKに頼らないで、Gecko側で独自に行っている（widget/gtk/nsGtkKeyUtils.cpp)。
これがXに依存しており、Xが無い場合は`mModifierMasks`の中身がまるっと全部`0`になってしまう。

https://dxr.mozilla.org/mozilla-central/source/widget/gtk/nsGtkKeyUtils.cpp#197
https://dxr.mozilla.org/mozilla-central/source/widget/gtk/nsGtkKeyUtils.cpp#433

試しに固定値で

```
mModifierMasks[INDEX_ALT] = 8;
```

とか入れてみたら動いたので、Wayland用の実装を追加してやれば良さそう。

cosmo0920, 2017-07-12:
https://bugzilla.redhat.com/show_bug.cgi?id=1470031 現状分かったことをRed HatのBugzillaにfileしました :yum:

ashie, 2017-07-13:
ちょっと実装方法を検討してみたが、実装コストの割に得られるものが少ない。
ひとまずALTキーだけ使えればおおよそのニーズはカバーできるので、まずは以下のようなパッチだけ
入れてしまって、残りは後でゆっくり考えよう。

```diff
diff --git a/widget/gtk/nsGtkKeyUtils.cpp b/widget/gtk/nsGtkKeyUtils.cpp
index ce55cf18ea2f..55e5600e6c0a 100644
--- a/widget/gtk/nsGtkKeyUtils.cpp
+++ b/widget/gtk/nsGtkKeyUtils.cpp
@@ -196,6 +196,8 @@ KeymapWrapper::Init()
 
     if (GDK_IS_X11_DISPLAY(gdk_display_get_default()))
         InitBySystemSettings();
+    else
+        mModifierMasks[INDEX_ALT] = 1 << 3;
 
     gdk_window_add_filter(nullptr, FilterEvents, this);
```

ashie, 2017-07-13:
meta-browserに上記パッチを投入した。
残りは https://bugzilla.redhat.com/show_bug.cgi?id=1470031 にて、ということしていったんクローズします。

closed 2017-07-13 by ashie