webdino/gecko-embedded #51: ブックマークウィンドウを開くとクラッシュする

ashie, 2017-07-03:
Wayland版で「履歴とブックマークの管理」ウィンドウを開くとクラッシュする。
実際にはクリップボード回りの問題のようだ。

```
#0  raise (sig=11) at ../sysdeps/unix/sysv/linux/pt-raise.c:38
#1  0xb500c024 in nsProfileLock::FatalSignalHandler(int, siginfo_t*, void*) () from /usr/lib/firefox-52.2.0/libxul.so
#2  <signal handler called>
#3  wl_proxy_marshal (proxy=0x0, opcode=1) at /usr/src/debug/wayland/1.10.0-r0/wayland-1.10.0/src/wayland-client.c:679
#4  0xb4959c0c in nsRetrievalContextWayland::GetClipboardContent(char const*, int, nsIInputStream**, unsigned int*) () from /usr/lib/firefox-52.2.0/libxul.so
#5  0xb495ac4c in nsClipboard::GetData(nsITransferable*, int) [clone .part.127] [clone .constprop.244] () from /usr/lib/firefox-52.2.0/libxul.so
#6  0xb34483f8 in NS_InvokeByIndex () from /usr/lib/firefox-52.2.0/libxul.so
#7  0x00000000 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

cosmo0920, 2017-07-03:
Show All HistoriesとShow All Bookmarks、としてShow Downloadsのメニューはまた別のウィンドウを作成する。ここら辺の動きが怪しいのかもしれない。
ちゃんと調べていないですが、これらのウィンドウはXULにて作成されているような気がします。

ashie, 2017-07-06:
> #3  wl_proxy_marshal (proxy=0x0, opcode=1) at /usr/src/debug/wayland/1.10.0-r0/wayland-1.10.0/src/wayland-client.c:679

```c
        va_start(ap, opcode);
        wl_argument_from_va_list(proxy->object.interface->methods[opcode].signature,
                                 args, WL_CLOSURE_MAX_ARGS, ap);
        va_end(ap);
```

例によって想定してるプロトコルバージョンが一致してないのかも。

ashie, 2017-07-06:
単にこれが適用漏れっぽい

https://github.com/mozilla-japan/gecko-dev/commit/7b8cf7f93879cf2a1f20332a4f57fa9b6a306121

ashie, 2017-07-06:
投入した。

個々の変更は https://github.com/mozilla-japan/gecko-dev/tree/esr52-wayland-with-egl で管理してるので、レシピに投入するパッチはsquashして入れたほうがスッキリするかな? まあ、その辺は後で考える。

closed 2017-07-06 by ashie