webdino/gecko-embedded #53: RZ/G1: Firefox起動時にwestonがクラッシュすることがある

ashie, 2017-07-10:
再現条件が不明だが、RZ/G1でFirefoxを起動したときに、westonごとクラッシュすることがある。

[Yocto1.6](https://github.com/mozilla-japan/meta-browser/wiki/Build-RZ-G1E)、[Yocto 2.0](https://github.com/mozilla-japan/meta-browser/wiki/Build-RZ-G1E-Yocto2.0)どちらでも発生したことがある。
思うように再現することができていないので、ログ等は取得できていない。

ashie, 2017-07-14:
Westonのバックトレースが取れた:

```gdb
#0  raise (sig=sig@entry=5) at ../sysdeps/unix/sysv/linux/pt-raise.c:38
#1  0x00029a68 in on_caught_signal (s=<optimized out>, siginfo=<optimized out>, context=<optimized out>)
    at /usr/src/debug/weston/1.10.0-r0/weston-1.10.0/src/main.c:372
#2  <signal handler called>
#3  wl_resource_post_event_array (resource=resource@entry=0xfd8b0, opcode=opcode@entry=0, 
    args=args@entry=0xbee38668) at /usr/src/debug/wayland/1.10.0-r0/wayland-1.10.0/src/wayland-server.c:141
#4  0x48124284 in wl_resource_post_event (resource=0xfd8b0, opcode=0)
    at /usr/src/debug/wayland/1.10.0-r0/wayland-1.10.0/src/wayland-server.c:162
#5  0x0001dad0 in wl_callback_send_done (callback_data=<optimized out>, resource_=<optimized out>)
    at /build/rz-g-bsp-yocto-2.0/build/tmp/sysroots/skrzg1e/usr/include/wayland-server-protocol.h:217
#6  weston_output_repaint (output=<optimized out>)
    at /usr/src/debug/weston/1.10.0-r0/weston-1.10.0/src/compositor.c:2389
#7  output_repaint_timer_handler (data=<optimized out>)
    at /usr/src/debug/weston/1.10.0-r0/weston-1.10.0/src/compositor.c:2442
#8  0x481260ac in wl_event_source_timer_dispatch (source=0x3eb80, ep=<optimized out>)
    at /usr/src/debug/wayland/1.10.0-r0/wayland-1.10.0/src/event-loop.c:181
#9  0x48126608 in wl_event_loop_dispatch (loop=0x344f0, timeout=timeout@entry=-1)
    at /usr/src/debug/wayland/1.10.0-r0/wayland-1.10.0/src/event-loop.c:421
#10 0x48124e48 in wl_display_run (display=display@entry=0x34478)
    at /usr/src/debug/wayland/1.10.0-r0/wayland-1.10.0/src/wayland-server.c:1051
#11 0x000178f0 in main (argc=0, argv=0xbee38da4) at /usr/src/debug/weston/1.10.0-r0/weston-1.10.0/src/main.c:859
```

ashie, 2017-07-14:
```gdb
((gdb) up
#3  wl_resource_post_event_array (resource=resource@entry=0xfd8b0, opcode=opcode@entry=0, 
    args=args@entry=0xbee38668) at /usr/src/debug/wayland/1.10.0-r0/wayland-1.10.0/src/wayland-server.c:141
141		if (wl_closure_send(closure, resource->client->connection))
(gdb) print closure
$2 = (struct wl_closure *) 0xfced8
(gdb) print resource->client
$3 = (struct wl_client *) 0x0
```

ashie, 2017-07-14:
protocol/wayland.xml:

```xml
  <interface name="wl_callback" version="1">
    <description summary="callback object">
      Clients can handle the 'done' event to get notified when
      the related request is done.
    </description>
    <event name="done">
      <description summary="done event">
       Notify the client when the related request is done.
      </description>
      <arg name="callback_data" type="uint" summary="request-specific data for the wl_callback"/>
    </event>
  </interface>
```

ashie, 2017-07-14:
westonが死んだ後もFirefoxはしばらく生きていてサーバ側の反応を待っている（先に死んでいるわけではない）。

ashie, 2017-08-25:
/etc/xdg/weston.iniで以下（HDMIの解像度を下げてVGAをoff）のようにしているとwestonが落ちるのを見なくなった...ような気がする

```
[output]
name=HDMI-A-1
mode=1280x720

[output]
name=VGA-1
mode=off
```

RZ/G1E StarterKitでは、何も設定していないとVGAも出力されていて右側にデスクトップが拡張されているようだ。

cosmo0920, 2017-08-25:
もう一台のRZ/G1E StarterKitでも2回目の起動時にWestonが巻き込まれて落ちていた問題が発生しなくなりました。

ashie, 2017-10-26:
起動時の意図しないウィンドウリサイズでクラッシュしているようだったが、これを抑制するようにしてクラッシュしなくなった。

closed 2017-10-26 by ashie