webdino/gecko-embedded #60: ESR60対応

ashie, 2018-01-26:
ESR60に向けた調査を以下のブランチで開始している。

https://github.com/webdino/meta-browser/tree/firefox-60-wip

ざっとビルドは通るようにしたが、まだ動作は確認できていない（起動直後にクラッシュ）

ashie, 2018-01-26:
ビルドには、これまでの手順に加えて、rustupで最新のRustをインストールする必要がある

```
$ curl https://sh.rustup.rs -sSf | sh
```

あとgcc-4.9以上が必要。Ubuntu-14.04だとつらいかも。Ubuntu-16.04ではビルド可。

ashie, 2018-01-26:
パッチの移植はまだできていない。

ただしWaylandパッチはmozilla-centralにマージされており、追加パッチ無しでビルドできるようになっている。
gecko-embeddedで当てていたWayland版の追加修正も、bugzila.redhat.comからbugzilla.mozilla.orgの方に移行してmozilla-centralに入り始めているので、Wayland対応に関してはほぼパッチ無しで対応できる見込み。

ashie, 2018-01-26:
> ビルドには、これまでの手順に加えて、rustupで最新のRustをインストールする必要がある

meta-rustで対応する方法も検討中だが、入ってるRustのバージョンが追いつかなくなったので保留中。
いずれ追いついたら再開したい。

ashie, 2018-01-26:
現在遭遇している問題

* configure時のRustのtargetの検出でサポートしていないtargetが検出される
    * `rustc --print target-list`と`rustup target list`の不一致が関係していそう
    * https://github.com/rust-lang-nursery/rustup.rs/issues/326
    * ひとまずクソパッチでtargetを固定化して先を急いだ。一通りビルドは通ったのでそろそろちゃんと調査・報告したい。
* `--enable-release`が無いと、`--disable-gold`としていてもgoldリンカーが使われる
* Styloのビルドでx86のヘッダファイルが参照される
    * `--disable-stylo`で保留中
* WebRTCで存在しないヘッダファイル<gnu/stubs-soft.h>が参照される
    * `--disable-webrtc`で保留中
* AV1のビルドでビルドすべきソースが見つからない
    * `--disable-av1`で保留中
* 起動直後にクラッシュする
```
#0  0x4ae1294c in raise (sig=11) at ../sysdeps/unix/sysv/linux/pt-raise.c:36
36	  return INLINE_SYSCALL (tgkill, 3, pid, THREAD_GETMEM (THREAD_SELF, tid),
(gdb) where
#0  0x4ae1294c in raise (sig=11) at ../sysdeps/unix/sysv/linux/pt-raise.c:36
#1  0xb58dd78c in nsProfileLock::FatalSignalHandler(int, siginfo_t*, void*) () from /usr/lib/firefox/libxul.so
#2  0xb58dd78c in nsProfileLock::FatalSignalHandler(int, siginfo_t*, void*) () from /usr/lib/firefox/libxul.so
#3  0xb58dd78c in nsProfileLock::FatalSignalHandler(int, siginfo_t*, void*) () from /usr/lib/firefox/libxul.so
#4  0xb58dd78c in nsProfileLock::FatalSignalHandler(int, siginfo_t*, void*) () from /usr/lib/firefox/libxul.so
#5  0xb58dd78c in nsProfileLock::FatalSignalHandler(int, siginfo_t*, void*) () from /usr/lib/firefox/libxul.so
...（以下延々同じ）
```

makotokato, 2018-01-29:
知ってる話書いておきます

60の段階だと--disable-styloは有効ですが、将来的にはstyloオフにするオプション無くします。もしStyloをビルドするのであれば、こんな感じ(gcc-4.9.3を使ってるの例) で.mozconfigにBINDGEN_CFLAGSを指定してください (クロスビルドは必須)
`export BINDGEN_CFLAGS="--target=arm-linux-gnueabihf -I/usr/arm-linux-gnueabihf/include/c++/4.9.3 -I/usr/arm-linux-gnueabihf/include/c++/4.9.3/arm-linux-gnueabihf -I/usr/lib/llvm-3.9/lib/clang/3.9.1/include"`

gcc-5.xだと多分armの生成コードの問題で色々問題あるので、4.9か6.4以降くらいじゃないと多分動かないです

av1はnightlyビルドだけなので、stableチャンネルだとデフォルトでビルドされないはず。armだとビルドする設定にはなってるけど、あれビルド通らないです (Makefile.inでアセンブラコード生成するコード書き忘れてる)

webrtcのコードはGYPからGNに移行した段階で壊れてるのですが、というかそもそも59くらいからLinux/armだとビルド通らないコードになってる

ashie, 2018-01-29:
情報ありがとうございます！試してみます！

makotokato, 2018-01-29:
WebRTC: https://bugzilla.mozilla.org/show_bug.cgi?id=1433725

ashie, 2018-01-30:
> gcc-5.xだと多分armの生成コードの問題で色々問題あるので、4.9か6.4以降くらいじゃないと多分動かないです

確かに、gcc-4.9で動くバリナリが作成できることを確認できました。
ありがとうござます！

Yoctoの場合はYoctoで導入するgccも入れ替える必要があるので、手順をまとめる必要がある。

ashie, 2018-01-30:
魚拓

[image]

ashie, 2018-01-30:
> Yoctoの場合はYoctoで導入するgccも入れ替える必要があるので、手順をまとめる必要がある。

現状は以下の手順（レシピを直接変更しなくて済むようにしたい）:

* 以下でシステムのgccを4.9に変更
```
(for Ubuntu-16.04)
$ sudo apt-get install g++-4.9
$ sudo update-alternatives --install /usr/bin/gcc gcc /usr/bin/gcc-5 10
$ sudo update-alternatives --install /usr/bin/gcc gcc /usr/bin/gcc-4.9 20
$ sudo update-alternatives --install /usr/bin/g++ g++ /usr/bin/g++-5 10
$ sudo update-alternatives --install /usr/bin/g++ g++ /usr/bin/g++-4.9 20
$ sudo rm /usr/bin/cpp
$ sudo update-alternatives --install /usr/bin/cpp cpp /usr/bin/cpp-5 10
$ sudo update-alternatives --install /usr/bin/cpp cpp /usr/bin/cpp-4.9 20
$ sudo update-alternatives --set cc /usr/bin/gcc
$ sudo update-alternatives --set c++ /usr/bin/g++
```
* meta-renesasを以下のように変更
```diff
diff --git a/meta-rzg1/conf/machine/include/rzg-providers.inc b/meta-rzg1/conf/machine/include/rzg-providers.inc
index b5d6993..83f0e40 100644
--- a/meta-rzg1/conf/machine/include/rzg-providers.inc
+++ b/meta-rzg1/conf/machine/include/rzg-providers.inc
@@ -6,4 +6,4 @@
 #PREFERRED_VERSION_gcc-crosssdk-initial = "linaro-5.2"
 #PREFERRED_VERSION_gcc-crosssdk-intermediate = "linaro-5.2"
 #PREFERRED_VERSION_gcc-cross-canadian-${TRANSLATED_TARGET_ARCH} = "linaro-5.2"
-GCCVERSION = "linaro-5.2"
+GCCVERSION = "linaro-4.9"
```
* meta-linaroを以下のように変更
```diff
diff --git a/meta-linaro-toolchain/recipes-devtools/gcc/gcc-linaro-4.9.inc b/meta-linaro-toolchain/recipes-devtools/gcc/gcc-linaro-4.9.inc
index 852a316..d3b30bb 100644
--- a/meta-linaro-toolchain/recipes-devtools/gcc/gcc-linaro-4.9.inc
+++ b/meta-linaro-toolchain/recipes-devtools/gcc/gcc-linaro-4.9.inc
@@ -8,7 +8,7 @@ BINV = "4.9.4"
 
 FILESPATH = "${@base_set_filespath([ '${FILE_DIRNAME}/gcc-${PV}' ], d)}"
 
-SRC_URI = "https://releases.linaro.org/${MMYY}/components/toolchain/gcc-linaro/${BASEPV}/gcc-${PV}-${RELEASE}.tar.xz \
+SRC_URI = "https://releases.linaro.org/archive/${MMYY}/components/toolchain/gcc-linaro/${BASEPV}/gcc-${PV}-${RELEASE}.tar.xz \
            file://0001-gcc-4.3.1-ARCH_FLAGS_FOR_TARGET.patch \
            file://0002-uclibc-conf.patch \
            file://0003-gcc-uclibc-locale-ctype_touplow_t.patch \
```

ashie, 2018-02-07:
> 60の段階だと--disable-styloは有効ですが、将来的にはstyloオフにするオプション無くします。もしStyloをビルドするのであれば、こんな感じ(gcc-4.9.3を使ってるの例) で.mozconfigにBINDGEN_CFLAGSを指定してください (クロスビルドは必須)
> export BINDGEN_CFLAGS="--target=arm-linux-gnueabihf [...]

教えてもらった方針でStyloをビルドできることは確認できたが、Yoctoでどう汎化させてレシピに落とし込むかというところが悩みどころ。

以下、関連箇所のメモ:

https://dxr.mozilla.org/mozilla-central/rev/0d806b3230fe4767fa70cdee57db87d1e9a5ba49/old-configure.in#4141:

```
BINDGEN_SYSTEM_FLAGS="$_BINDGEN_CFLAGS $NSPR_CFLAGS $NSS_CFLAGS $MOZ_PIXMAN_CFLAGS $MOZ_CAIRO_CFLAGS"
AC_SUBST_TOML_LIST(BINDGEN_SYSTEM_FLAGS)
```

https://dxr.mozilla.org/mozilla-central/rev/0d806b3230fe4767fa70cdee57db87d1e9a5ba49/build/moz.configure/toolchain.configure#1070

```
option(env='BINDGEN_CFLAGS',
       nargs=1,
       default=bindgen_cflags_defaults,
       help='Options bindgen should pass to the C/C++ parser')

@depends('BINDGEN_CFLAGS')
@checking('bindgen cflags', lambda s: s if s else 'no')
def bindgen_cflags(value):
    if value and len(value):
        return value[0].split()

add_old_configure_assignment('_BINDGEN_CFLAGS', bindgen_cflags)
```

https://dxr.mozilla.org/mozilla-central/rev/0d806b3230fe4767fa70cdee57db87d1e9a5ba49/layout/style/bindgen.toml.in#3

```
[build]
args = [
    @BINDGEN_SYSTEM_FLAGS@
]
```

ashie, 2018-02-08:
> gcc-5.xだと多分armの生成コードの問題で色々問題あるので、4.9か6.4以降くらいじゃないと多分動かないです

Yoctoのバージョン毎に使えるgccのバージョンを調査してみた:

* poky
    * 2.0 (jethro)
        * gcc-4.8
        * gcc-4.9
        * gcc-5.2
    * 2.1 (krogoth)
        * gcc-4.9
        * gcc-5.3
    * 2.2 (morty)
        * gcc-5.4
        * gcc-6.2
    * 2.3 (pyro)
        * gcc-5.4
        * gcc-6.3
    * 2.4
        * gcc-6.3
        * gcc-7.3
    * master (2.5 sumo)
        * gcc-7.3
* meta-linaro
    * 2.0 (jethro)
        * gcc-linaro-4.8
        * gcc-linaro-4.9
        * gcc-linaro-5.2
    * 2.1 (krogoth)
        * gcc-linaro-4.9
        * gcc-linaro-5.2
    * 2.2 (morty)
        * gcc-linaro-4.9
        * gcc-linaro-5.2
        * gcc-linaro-5.3
        * gcc-linaro-6.2
    * 2.3 (pyryo): mortyブランチから変更無し
        * gcc-linaro-4.9
        * gcc-linaro-5.2
        * gcc-linaro-5.3
        * gcc-linaro-6.2
    * 2.4 (rocko)
        * gcc-linaro-4.9
        * gcc-linaro-5.2
        * gcc-linaro-5.3
        * gcc-linaro-6.3
        * gcc-linaro-7.1
    * master (2.5 sumo)
        * gcc-linaro-4.9
        * gcc-linaro-5.2
        * gcc-linaro-5.3
        * gcc-linaro-6.3
        * gcc-linaro-7.1
        * gcc-linaro-7.2

ashie, 2018-02-08:
> * configure時のRustのtargetの検出でサポートしていないtargetが検出される
>     * `rustc --print target-list`と`rustup target list`の不一致が関係していそう

状況としては以下のような感じ

* 指定ターゲット: `arm-poky-linux-gnueabi`
* 検出されたターゲット:  `armv5te-unknown-linux-gnueabi`
* 候補（rustc --print target-list）： 検出コード的には下のほうが優先度が高い
    * arm-linux-androideabi
    * arm-unknown-linux-gnueabi
    * arm-unknown-linux-gnueabihf
    * arm-unknown-linux-musleabi
    * arm-unknown-linux-musleabihf
    * armv5te-unknown-linux-gnueabi
    * armv7-linux-androideabi
    * armv7-unknown-linux-gnueabihf
    * armv7-unknown-linux-musleabihf

RZ/G1Mの場合`armv7-unknown-linux-gnueabihf`が引っかかって欲しいところだけど、検出コードを見ると`raw_os`（`linux-gnueabi`の部分）の`hf`の有無でヒットしていないようだ。ビルドオプションには`-mfloat=hard`が付いているので、指定ターゲットも `arm-poky-linux-gnueabihf`になっていて欲しいところだが...

関連コード

* https://dxr.mozilla.org/mozilla-central/rev/8cc2427a322caa1e2c09ca3957335f88e573dc7a/build/moz.configure/rust.configure#185
```python
        if rustc_target is None:
            rustc_target = rust_supported_targets.per_raw_os.get(
                (host_or_target.cpu, host_or_target.endianness,
                 host_or_target_raw_os))
```
* https://dxr.mozilla.org/mozilla-central/rev/8cc2427a322caa1e2c09ca3957335f88e573dc7a/build/moz.configure/rust.configure#114
* https://dxr.mozilla.org/mozilla-central/rev/8cc2427a322caa1e2c09ca3957335f88e573dc7a/build/moz.configure/init.configure#574

ashie, 2018-02-08:
`armv5te-unknown-linux-gnueabi`は割と最近、以下のPRで追加されたようだ。

* https://github.com/rust-lang/rust/pull/46498

これがなければ、`arm-unknown-linux-gnueabi`がヒットしてconfigureは通る。
とはいえ上記のような状況なので、これのせいとは言い難い。

ashie, 2018-02-15:
> webrtcのコードはGYPからGNに移行した段階で壊れてるのですが、というかそもそも59くらいからLinux/armだとビルド通らないコードになってる

GNの対応は以下で行われたようだ。

* https://bugzilla.mozilla.org/show_bug.cgi?id=1393119

media/webrtc/gn-configs/に各ターゲット向けの設定があるが、ここにLinux/ARMのターゲットがないためにビルドできないようだ。ここのファイルの生成は

```
$ ./mach build-backend -b GnConfigGen
```

で行い、media/webrtc/trunk/gn-output/に生成されたjsonファイルをmedia/webrtc/gn-configs/に置いて

```
$ ./mach build-backend -b GnMozbuildWriter
```

でmoz.buildを生成し直すと、Linux/ARM向けにもビルドできるようになる。

ただし、GnConfigGenの際にgnコマンドが必要になる。
gnコマンドは以下のスクリプトでビルドできる

* https://gist.github.com/KSR-Yasuda/ce8a67fa944f6e82fc397b914b5bc8a6

ビルドの際には新しめのninjaが必要になる。以下から最新版をダウンロードするのが無難

* https://github.com/ninja-build/ninja/releases

ashie, 2018-02-15:
生成済みのjsonをfirefox-60-wipブランチに入れた（gnやninja不要）

* https://github.com/webdino/meta-browser/tree/firefox-60-wip/recipes-mozilla/firefox/firefox/gn-configs

ashie, 2018-02-15:
関連Bug

* https://bugzilla.mozilla.org/show_bug.cgi?id=1433725
* https://bugzilla.mozilla.org/show_bug.cgi?id=1434589

dynamis, 2018-02-26:
GCC を 4.9 に落とすのではなく逆に最新の GCC にしていく方向を検討する場合のためのメモ:

* [Use GCC 6 to build what we ship on Linux](https://bugzilla.mozilla.org/show_bug.cgi?id=gcc6)
  * Firefox 57 からは GCC 6 が Linux 用のリリース版に使われているっぽい

* [meta: Build with gcc 7](https://bugzilla.mozilla.org/show_bug.cgi?id=1320654)
  * tree: https://bugzilla.mozilla.org/showdependencytree.cgi?id=1320654&hide_resolved=1
* [meta: Build with gcc 8](https://bugzilla.mozilla.org/show_bug.cgi?id=1409283)
  * tree: https://bugzilla.mozilla.org/showdependencytree.cgi?id=1409283&hide_resolved=1

dynamis, 2018-02-26:
GCC 4.9 は捨てて GCC 6 移行にしたいという bug は立っているが Hazard builds が GCC 6 に移行できるまで無理といって今のところはなしは止まっている。GCC 4.9 のサポートは ESR 的には ESR60 が最後になるのかも？バージョンアップ毎に移行をせずに済むと良いのだが。。。

* [Increase minimum gcc version to gcc 6](https://bugzilla.mozilla.org/show_bug.cgi?id=1404040)

ashie, 2018-02-27:
OpenMAXパッチが動作するようになったのでレシピに投入した

* https://github.com/webdino/meta-browser/tree/firefox-60-wip
    * https://github.com/webdino/meta-browser/commit/59b1517fb43b0fe7efa869e9ecf7c926c863e263

せっかくmozilla-centralに追いついたのでパッチの整理はもう少し進めたいが、そのへんはBuzgillaの方で

* https://bugzilla.mozilla.org/show_bug.cgi?id=1306529

ashie, 2018-02-27:
> * https://github.com/webdino/meta-browser/tree/firefox-60-wip
>    * https://github.com/webdino/meta-browser/commit/59b1517fb43b0fe7efa869e9ecf7c926c863e263

* 間違ってたのでさらに修正

https://github.com/webdino/meta-browser/commit/be2dacda5c7abcf7ba5b67ae68b3ec34c13fb321

ashie, 2018-03-06:
> * 間違ってたのでさらに修正
> 
> https://github.com/webdino/meta-browser/commit/be2dacda5c7abcf7ba5b67ae68b3ec34c13fb321

これである程度は動くのだが、しばらく使ってみると、動画をいくつも再生しているとやがて再生が止まるとまるという問題があることが発覚。まだ何かおかしそう。

ashie, 2018-03-06:
ビルド手順の検証がてらに、Vagrant Boxを作ってみている

* https://github.com/ashie/renesas-gecko60-devenv

VagrantとVirtual Boxが入っている環境なら、以下でビルドまでいけると思う。

ホスト側:

```
$ vagrant init ashie/renesas-gecko-dev
$ vagrant up
$ vagrant ssh
```

ゲスト側:

```
$ cd rzg1-bsp
（プロプライエタリドライバをコピー）
$ source poke/oe-init-buildenv
$ bitbake core-image-weston
```

ただし、取得するFirefoxのリポジトリがgecko-devのmasterに向いているので、その時の状況次第ではビルドに失敗することもあると思う。実際、これを書いている時点ではANGLEのビルドで引っかかる（[誠さんの修正](https://reviewboard.mozilla.org/r/225206/diff/1#index_header)が入れば解消する）。

プロプライエタリドライバのダウンロードは自動化するわけにはいかないので、ちょっと面倒だけどここは手動でやってもらわないといけない。

ashie, 2018-03-06:
> 教えてもらった方針でStyloをビルドできることは確認できたが、Yoctoでどう汎化させてレシピに落とし込むかというところが悩みどころ。

一応、以下でStyloのビルドも通るようにしてある（まだやり方が雑だが）

```
PACKAGECONFIG_append_pn-firefox = " stylo "
```

ashie, 2018-03-06:
* https://github.com/ashie/renesas-gecko60-devenv

名前の付け方しくじったな...対象Geckoバージョンはbranchかtagで切り分けておくべきだった。
後でリポジトリ作り直す。

ashie, 2018-03-09:
>> https://github.com/webdino/meta-browser/commit/be2dacda5c7abcf7ba5b67ae68b3ec34c13fb321
>
> これである程度は動くのだが、しばらく使ってみると、動画をいくつも再生しているとやがて再生が止まるとまるという問題があることが発覚。まだ何かおかしそう。

修正した

https://github.com/webdino/meta-browser/commit/b8de3b0344567a8750214aa323b24f1c0b5f40b0

ashie, 2018-03-09:
> * https://github.com/ashie/renesas-gecko60-devenv
>
> 名前の付け方しくじったな...対象Geckoバージョンはbranchかtagで切り分けておくべきだった。
> 後でリポジトリ作り直す。

名前変えた

* https://github.com/ashie/renesas-gecko-devenv

GitHubって割とカジュアルにリポジトリのリネームできるのね。
単にリネームすると、以前の名前でアクセスしてもリダイレクトされる。

ashie, 2018-03-16:
> meta-rustで対応する方法も検討中だが、入ってるRustのバージョンが追いつかなくなったので保留中。
> いずれ追いついたら再開したい。

追いついたようだけど、そのままではビルドが通らなかったので試行錯誤中。
Upstreamとのやり取りは以下:

https://github.com/meta-rust/meta-rust/pull/191

ashie, 2018-03-19:
EGLパッチもFirefox 60向けに更新したものをレシピに投入しておいた。

https://github.com/webdino/meta-browser/blob/firefox-60-wip/recipes-mozilla/firefox/firefox/wayland/egl/0003-Use-wl_egl_window-as-a-native-EGL-window-on-Wayland.patch

もう少し改善したいところはあるが、その辺はBugzillaの方で。

ちなみにFirefox 52の頃はGDK側と描画処理が取り合っていてワークアラウンドを入れないとまともに描画ができていなかったので、GdkGLContextに対応したGLContextProviderを新規に作る形で根本的に書き直さないといけないと考えていた。が、Firefox 60ではGecko側で独自のwl_surfaceを作るようになってGDK側と描画を取り合うことがなくなったので、本パッチのようにGLContextProviderEGLを実装をほぼそのまま使う形で行けるようになった。

ashie, 2018-03-20:
>> 教えてもらった方針でStyloをビルドできることは確認できたが、Yoctoでどう汎化させてレシピに落とし込むかというところが悩みどころ。
> 
> 一応、以下でStyloのビルドも通るようにしてある（まだやり方が雑だが）
> 
> ```
> PACKAGECONFIG_append_pn-firefox = " stylo "
> ```

core-image-westonをスクラッチからビルドするときにエラーになる問題が発覚したので、コメントアウトした。Styloをビルドしたい場合は該当箇所をアンコメントし、一度Firefox抜きでcore-image-westonをビルドして、Firefoxをビルドする必要がある。

meta-rustを導入すれば`BINDGEN_CFLAGS`を指定せずに済む算段はついているので、meta-rustでのビルド手順を整備できたら該当箇所は完全に削除する。

cosmo0920, 2018-03-29:
> meta-rustを導入すればBINDGEN_CFLAGSを指定せずに済む算段はついているので、meta-rustでのビルド手順を整備できたら該当箇所は完全に削除する。

と思ったら、そんなことはなかったのでそのことについて調査中です。

cosmo0920, 2018-03-29:
こっちにも記しておきます。
https://github.com/webdino/meta-browser/commit/8487ebca6b7ee74b71f768b621c869563e09d377 にてスクラッチからの `bitbake core-image-weston` を動作させることが可能になりました。Rustについてはmeta-rustがきちんと使えているかどうかは調査中です。

cosmo0920, 2018-04-03:
meta-rustについても https://github.com/webdino/meta-browser/tree/firefox-60-wip-with-meta-rust ブランチにてビルドに使用できることを確認済み。このブランチではrustupのインストールが不要になった。

ashie, 2018-04-05:
別件でlibrsvgのクロスビルドの方法を調べていたら、Rustのtarget指定に関するドキュメントが見つかった。

* https://github.com/GNOME/librsvg/blob/master/COMPILING.md#cross-compilation
* https://github.com/japaric/rust-cross#target-specification-files
* https://github.com/rust-lang/rust/blob/master/src/librustc_back/target/mod.rs

cosmo0920, 2018-05-09:
60ESRのソースが公開されたので、作業ブランチを `firefox-60esr` に移動しました。

ashie, 2018-05-29:
ESR60対応のブランチとビルド手順は公開済み:

https://github.com/webdino/meta-browser/wiki/Firefox-ESR60%E3%83%93%E3%83%AB%E3%83%89%E6%89%8B%E9%A0%86%EF%BC%88iWave-RainboW-G20D-Q7-Yocto-2.0%EF%BC%89

課題は残っているが、ESR60のビルド・動作は一応確認できているので、本issueはそろそろ閉じて個別のissueに分けた方が良さそう。

cosmo0920, 2018-05-29:
こちらのissueは閉じます。
他の問題は個別にissueに切り出します。

closed 2018-05-29 by cosmo0920