webdino/gecko-embedded #62: 同期描画でのフリーズ問題

cosmo0920, 2018-05-29:
EGL有効化時にOpenGLコンポジターを有効にするとfirefoxが立ち上がらない。
同期描画ではなく、非同期描画にするとsubsurfaceへの描画が開始され、firefoxが立ち上がる。

see also: https://bugzilla.mozilla.org/show_bug.cgi?id=1464823

cosmo0920, 2018-05-31:
上記のBugではMesaのEGLドライバにバグがあるようだが、RZ/G1M・R-Car M3のEGLドライバにも同様のバグがある模様。

Mesaについては以下にバグ報告されている。
https://bugs.freedesktop.org/show_bug.cgi?id=106753

cosmo0920, 2018-06-01:
こちらは本当の原因はFirerfoxのWaylandの実装にあるようだ。
https://bugs.freedesktop.org/show_bug.cgi?id=106753#c6

cosmo0920, 2018-06-11:
一旦 @ashie さんのパッチをfdirefox-60esrに投入してみる。
https://github.com/webdino/meta-browser/commit/a3cb615cc166b48dec10e51956e9755fe9f3bb29

ashie, 2018-06-19:
PC版では

* https://bugzilla.mozilla.org/show_bug.cgi?id=1464823

でいったん解決しているが、RZ/G に同じパッチを投入するとクラッシュする。
この件については以下にバグを立ててある。

* https://bugzilla.mozilla.org/show_bug.cgi?id=1468907

本プロジェクトとしては

> 一旦 @ashie さんのパッチをfirefox-60esrに投入してみる。
> webdino/meta-browser@a3cb615

で動作しているので、こちらもクローズし、続きは上記Bugにて。

closed 2018-06-19 by ashie