webdino/gecko-embedded #66: meta-rustのUpstreamに追従する

cosmo0920, 2018-07-08:
https://github.com/meta-rust/meta-rust/pull/199 によると、rust-llvmのllvmのインストールパスがmeta-clangのそれと被らなくなった。
またx86_64以外のハッシュ値の追加も容易となったため、この変更を取り込みたい。

cosmo0920, 2018-07-09:
その前にまずはrust-llvmのインストール先を/usr/libではなく、/usr/lib/rust-llvmへインストールする変更だけを取り込んでみる。

cosmo0920, 2018-07-09:
まずはllvm-rustのインストール先の分離だけ先に取り込んだ。
https://github.com/webdino/meta-rust/commit/1cda352362eaac184c19b3e8cdbf35c7b1fa9213

ashie, 2018-08-29:
そもそものmeta-rustの問題は、rust-cross-${TARGET_ARCH}でターゲット用のstd crateが提供されていないこと。なぜかホスト環境用のstd crateしか存在しない。

* https://github.com/meta-rust/meta-rust/pull/191

上記のプルリクエストは、汚くても良いのでとにかくGecko60をビルドできるようにするために、stage0でRustコンパイラをセルフビルドする用にダウンロードしているビルド済みstd createをそのままインストールしてユーザーに使用させるという修正。ただしオリジナルのコードでは（セルフビルド用なので）前バージョンのstd createをダウンロードしているが、これだとバージョンチェックに引っかかってしまうので、今ビルドしようとしているRustと同じバージョンのstd crateをダウンロードするように修正している。また、バージョンは同じでも厳密にはビルドが異なるので、やはり使用時にバージョンチェックで引っかかってしまう。これを回避（ビルドNoは見ないでバージョンだけチェック）するためのパッチをRustに当てている。

```diff
--- a/src/librustc_metadata/locator.rs  2018-03-14 10:50:49.537437539 +0900
+++ b/src/librustc_metadata/locator.rs  2018-03-14 11:10:14.041437539 +0900
@@ -673,7 +673,7 @@
     fn crate_matches(&mut self, metadata: &MetadataBlob, libpath: &Path) -> Option<Svh> {
         let rustc_version = rustc_version();
         let found_version = metadata.get_rustc_version();
-        if found_version != rustc_version {
+        if !&found_version.starts_with(&rustc_version) {
             info!("Rejecting via version: expected {} got {}",
                   rustc_version,
                   found_version);
```

参考: https://github.com/rust-lang/rust/issues/40991

```
The Rust compiler is written in Rust, and therefore needs a rust compiler to compile itself. This is not unusual, some other languages do this as well.
```

ちゃんとやるためにはどうすれば良いのかを相談したくて上記PRを投げたわけだが、今のところ良い解法が見つかっていない。

ashie, 2018-08-29:
たぶん、rust-cross-${TARGET_ARCH}がターゲット用のstd createもビルドしてくれるのがあるべき姿のような気がする。meta-rustにlibstd-rs_1.27.1.bb というレシピがあるので、これをビルドしてパスを通せば良いのか?とも思ったが、うまくいっていない。

meta-browserに60ESRのPRは投げてあるので

https://github.com/OSSystems/meta-browser/pull/160

meta-rustの方に「これビルドしたいんだけど、どうすりゃいい？」というissueを再度投げる予定。

ashie, 2018-10-16:
> meta-rustの方に「これビルドしたいんだけど、どうすりゃいい？」というissueを再度投げる予定。

以下に投げてある

* https://github.com/meta-rust/meta-rust/pull/208

ashie, 2018-10-16:
上記PRでmasterの現時点のHEADに対応済みなので、クローズします。
ちゃんとした対応は検討中。

closed 2018-10-16 by ashie