webdino/gecko-embedded #67: R-Car M3でStyloのビルドを成功させる

cosmo0920, 2018-07-08:
#66 の後でないとこのIssueは動けない。
meta-clangでlibclang.soをインストールできるようになればYoctoですべて完結するため、Styloのビルドの道が開けるはず。

cosmo0920, 2018-07-08:
また、表題にはR-Car M3と付けたが、Yocto 2.4を採用したBSPを扱ってのStyloのビルドにも高い確度でこのIssueの対応は必須になる見込み。

ashie, 2018-11-19:
有効化できた。
魚拓:

[image]

いまのレシピだとincludeパスの指定が適当すぎるので、その辺をもう少し整理する必要がある。

cosmo0920, 2018-11-20:
clang/llvm 3.9縛りが消えた気がするのでclang/llvm 4.0でビルド確認中。

cosmo0920, 2018-11-20:
clang/llvm 4.0でもビルドできるようになってました。

ashie, 2018-11-20:
修正をpushしたのでクローズします。

https://github.com/webdino/meta-browser/commit/a78ae6465516aa2be6d6b790d3bdb25d7bc7b648

> clang/llvm 4.0でもビルドできるようになってました。

この辺り何かやるのであれば、別issue立ててそちらでお願いします。

closed 2018-11-20 by ashie