webdino/gecko-embedded #69: meta-browser: 60.5.0への更新

ashie, 2019-02-21:
作業メモ用

* 60.4.0までは[OSSystems/meta-browser](https://github.com/OSSystems/meta-browser)にフィードバック済み
* 60.5.0については、clang-formatによるソース修正がESRにまで降りてきているため、バックポートパッチが当たらなくなっている
  * [update-to-firefox-60.5.0esr-wip](https://github.com/webdino/meta-browser/tree/update-to-firefox-60.5.0esr-wip)ブランチを切ってパッチを対応中
  * X11版については既にビルドが通るようになっている
  * musl、Wayland、OpenMAX対応のパッチの修正はこれから。以下のパッチ **以外** は全部当たらない
    * file://musl/musl-mutex.patch
    * file://musl/musl_webrtc_glibcism.patch
    * file://musl/fix-bug-1261392.patch
    * file://wayland/bug1462622-Dont-use-GLXVsyncSource-on-non-X11-displays.patch
    * file://wayland/egl/0001-Mark-GLFeature-framebuffer_multisample-as-unsupporte.patch
  * この機会にできれば古いパッチ（他のBugでさらに修正がかかってるやつ）も更新しておきたいが、余力が無ければやらない
  * ESRにまでこのタイミングで降りてくるとは思ってなかったが、バックポートすること考えたらコードベース合わせないとつらいのか...

ashie, 2019-05-28:
#70 を優先しているので、こちらの優先度は低め（今のところ、やらない可能性が高い）

dynamis, 2019-05-28:
そうですね。セキュリティパッチを当てていくことも出来れば望ましいですが、具体的な採用メーカーからのサポート要望が来ない限り Gecko 68 に進めることでセキュリティも aarch64 対応も解消していく方が良いですよね。

note:
Firefox Release Calendar https://wiki.mozilla.org/Release_Management/Calendar

December 11, 2018 https://www.mozilla.org/en-US/firefox/60.4.0/releasenotes/
January 29, 2019 https://www.mozilla.org/en-US/firefox/60.5.0/releasenotes/
...
May 21, 2019 https://www.mozilla.org/en-US/firefox/60.7.0/releasenotes/

ashie, 2019-07-02:
最新版meta-rust（Rust-1.34.2）+60.4以前のレシピでStyloを有効化するとビルドに失敗する。
（60.5以降だとどうなるのかは知らない）

```
11:52.09 error: missing documentation for macro
11:52.09    --> servo/components/style_traits/values.rs:139:1
11:52.09     |
11:52.09 139 | macro_rules! serialize_function {
11:52.09     | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
11:52.09     |
11:52.09 note: lint level defined here
11:52.09    --> servo/components/style_traits/lib.rs:12:22
11:52.09     |
11:52.09 12  | #![deny(unsafe_code, missing_docs)]
11:52.09     |                      ^^^^^^^^^^^^
11:52.09 
11:52.09 error: missing documentation for macro
11:52.09    --> servo/components/style_traits/values.rs:408:1
11:52.09     |
11:52.09 408 | macro_rules! define_css_keyword_enum {
11:52.09     | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
11:52.09 
11:52.09 error: aborting due to 2 previous errors
```

ashie, 2019-07-02:
> 60.5以降だとどうなるのかは知らない

60.7.2のコードを見てみたら直ってた。

ashie, 2019-07-03:
> 60.7.2のコードを見てみたら直ってた。

もうちょっとよく見てみたら、直ったんじゃなくて[間違って入ったコード](https://phabricator.services.mozilla.com/D31388)らしい（ひどい）。
60.8.0では元に戻るようだ。

dynamis, 2020-03-19:
Gecko 60 系列の security update 需要・重要度は無さそうなので一旦 wontfix としてクローズします。

closed 2020-03-19 by dynamis