webdino/gecko-embedded #70: 次期ESR（68）対応

ashie, 2019-02-21:
そろそろ次期ESR（68）を見据えて、出来ることはやっておきたい。

* 独自パッチはほとんど本体にフィードバック済みなので次期ESRではだいぶ減らせるはず
* フィードバックできるのにまだしてないパッチもあると思われる（RUST_TARGETの指定とか）ので、それらについては68がβになる前（[Firefox Release Calendar](https://wiki.mozilla.org/Release_Management/Calendar)によると2019-05-13）にフィードバックしておきたい
* 試しに66.0b9がビルドできるかどうか[firefox-66-wip](https://github.com/webdino/meta-browser/tree/firefox-66-wip)で試し始めてる
  * 案の定RUST関連で引っかかってる
  * β版使ってる意味は特に無い。ESR60の時と同じようにgecko-devに切り替えた方が良いだろう。

ashie, 2019-03-28:
ブランチを以下に移行してアルファチャネルで試し始めている

* https://github.com/webdino/meta-browser/tree/firefox-68-wip

これまでに自分がはまったところの対策は入れてあるけど、まだビルドは通ってない。

Styloのビルドにはclang-7が必要っぽい。3.9〜6でそれぞれ試したがダメだった。`mach bootstrap`で入るclangも7だし。
また、clang-7を使っても、まだ何かオプションが通ってないようでcstdintが見つからない。
とりあえずビルド時に以下のように手動でパスを通してみたところ、そこは突破した。

```diff
diff --git a/servo/components/style/build_gecko.rs b/servo/components/style/build_gecko.rs
index 4866909899c9..485817bd83eb 100644
--- a/servo/components/style/build_gecko.rs
+++ b/servo/components/style/build_gecko.rs
@@ -172,6 +172,9 @@ mod bindings {
                 builder = builder.clang_arg(item.as_str().expect("Expect string in list"));
             }

+           builder = builder.clang_arg("-I/build/master/build/tmp/work/i586-poky-linux/firefox/68esr-r0/recipe-sysroot/usr/include/c++/8.3.0");
+           builder = builder.clang_arg("-I/build/master/build/tmp/work/i586-poky-linux/firefox/68esr-r0/recipe-sysroot/usr/include/c++/8.3.0/i586-poky-linux");
+
             builder
         }
         fn include<T: Into<String>>(self, file: T) -> Builder {
```

いまは以下で悩んでる。なんじゃろ？

```
 0:01.89 config/external/nspr/pr
 0:01.89 i586-poky-linux-as: invalid option -- 'P'
 0:01.89 /build/master/build/tmp/work/i586-poky-linux/firefox/68esr-r0/git/config/rules.mk:763: recipe for target 'os_Linux_x86.o' failed
 0:01.89 make[4]: *** [os_Linux_x86.o] Error 1
 0:01.89 /build/master/build/tmp/work/i586-poky-linux/firefox/68esr-r0/git/config/recurse.mk:74: recipe for target 'config/external/nspr/pr/ta
rget' failed
 0:01.89 make[3]: *** [config/external/nspr/pr/target] Error 2
 0:01.89 make[3]: *** Waiting for unfinished jobs....
```

cosmo0920, 2019-03-28:
meta-clangのclangは既に8系に移行済みなようです。
https://github.com/kraj/meta-clang

できれば今度こそmeta-clangを使いたいですねぇ。。。

ashie, 2019-03-28:
68で[AArch64向けにIonMonkeyが有効化される](https://bugzilla.mozilla.org/show_bug.cgi?id=1536220)ようなので、R-Car Gen3でJavaScriptのベンチマーク性能が低い問題が解消される & WASMが使えるようになるんじゃないかと期待してるのだが、相変わらず一筋縄ではいかない。

ashie, 2019-04-01:
一点書き忘れてた。
今の自分の環境だとcbindgenが`mach bootstrap`で入れた~/.mozbuild/cbindgen/cbindgenに向いている。
同じように別途インストールしてないと、configureすら通らないと思う。

ashie, 2019-04-03:
メモ: 

```
 0:01.89 /build/master/build/tmp/work/i586-poky-linux/firefox/68esr-r0/git/config/rules.mk:763: recipe for target 'os_Linux_x86.o' failed
```

ソースはここ

* nsprpub/pr/src/md/unix/os_Linux_x86.s

moz.buildはここ

* config/external/nspr/pr/moz.build

```
 0:01.89 i586-poky-linux-as: invalid option -- 'P'
```

これやってるところはどこだか分かってない。

nsprpub/pr/include/md/_linux.hとconfig/external/nspr/pr/moz.buildを修正してarch依存のコードをジェネリックなコードに置き換えると先に進むようではあるので、上記箇所を見つければnspr回りはなんとかなるだろう。

その後はskiaでエラー

```
 0:01.69 /build/master/build/tmp/work/i586-poky-linux/firefox/68esr-r0/git/gfx/skia/skia/third_party/skcms/src/Transform_inl.h: In function 'v
oid baseline::exec_ops(const Op*, const void**, const char*, char*, int)':
 0:01.69 /build/master/build/tmp/work/i586-poky-linux/firefox/68esr-r0/git/gfx/skia/skia/third_party/skcms/src/Transform_inl.h:640:13: interna
l compiler error: in convert_move, at expr.c:218
 0:01.69  static void exec_ops(const Op* ops, const void** args,
 0:01.69              ^~~~~~~~
 0:01.70 Please submit a full bug report,
 0:01.70 with preprocessed source if appropriate.
 0:01.70 See <https://gcc.gnu.org/bugs/> for instructions.
 0:01.70 /build/master/build/tmp/work/i586-poky-linux/firefox/68esr-r0/git/config/rules.mk:805: recipe for target 'skcms.o' failed
 0:01.70 make[4]: *** [skcms.o] Error 1
```

ashie, 2019-04-04:
> ```
>  0:01.89 i586-poky-linux-as: invalid option -- 'P'
> ```
> 
> これやってるところはどこだか分かってない。

それ以前に、どう見ても使おうとしているアセンブラの想定が違っている。

configureのログを見ると

```
 0:02.78 checking for nasm... /build/master/build/tmp/work/i586-poky-linux/firefox/68esr-r0/recipe-sysroot-native/usr/bin/nasm
 0:02.78 checking nasm version... 2.14.02
 0:02.80 checking for linker... bfd
 0:02.80 checking for the assembler... /build/master/build/tmp/work/i586-poky-linux/firefox/68esr-r0/recipe-sysroot-native/usr/bin/i586-poky-l
inux/i586-poky-linux-as
```

対して、普通に`./mach bootstrap`で環境整えて`./mach build`でセルフビルドする場合は

```
 0:01.79 checking for llvm_profdata... /home/aho/.mozbuild/clang/bin/llvm-profdata
 0:01.80 checking for nasm... /usr/bin/nasm
 0:01.80 checking nasm version... 2.13.02
 0:01.86 checking for linker... gold
 0:01.86 checking for the assembler... /usr/bin/clang
```

そもそもmozilla-centralの方はclangに全面移行してるが、今のレシピは60のときのやつをベースにあまり変えないので使用してるので、使用するデフォルトツールチェーンも以前のまま。それが問題か。

ashie, 2019-04-04:
C++コンパイラもこの通り

Yocto:

```
 0:02.63 checking for the target C++ compiler... /build/master/build/tmp/work/i586-poky-linux/firefox/68esr-r0/recipe-sysroot-native/usr/bin/i586-poky-linux/i586-poky-linux-g++
 0:02.64 checking whether the target C++ compiler can be used... yes
 0:02.64 checking the target C++ compiler version... 8.3.0
```

セルフ:

```
 0:01.49 checking for the target C++ compiler... /usr/bin/clang++
 0:01.53 checking whether the target C++ compiler can be used... yes
 0:01.53 checking the target C++ compiler version... 6.0.0
```

ashie, 2019-04-08:
[meta-clang](https://github.com/kraj/meta-clang)を使用し、以下の設定をlocal.confに追加

```
HOSTTOOLS += "autoconf2.13"
IMAGE_INSTALL_append = " firefox "
TOOLCHAIN_pn-firefox = "clang"
AS_pn-firefox="${TARGET_PREFIX}clang"

# Wayland版パッチは不要なものは削除したが、残りの移植が済んでない
# パッチを削除すればとりあえずビルドは通るかも
PACKAGECONFIG_remove_pn-firefox = " wayland "
```

これで、少なくとも`qemux86`向けにはビルドが通ることを確認できた。

ただし、以下のツールは今のところ`./mach bootstrap`で導入したものを使用:

```
 0:03.66 checking for cbindgen... /home/aho/.mozbuild/cbindgen/cbindgen
 0:03.66 checking cbindgen version... 0.8.3
 0:03.66 checking for rustfmt... /home/aho/.cargo/bin/rustfmt
...
 0:03.74 checking for nodejs... /home/aho/.mozbuild/node/bin/node (8.11.3)
```

また、パッケージ作成後のverifyで以下のエラーが出る。

```
ERROR: firefox-68esr-r0 do_package_qa: QA Issue: /usr/lib/firefox/fix_linux_stack.py contained in package firefox requires /usr/bin/python, but no providers found in RDEPENDS_firefox? [file-rdeps]
```

ashie, 2019-04-08:
> ```
> # Wayland版パッチは不要なものは削除したが、残りの移植が済んでない
> # パッチを削除すればとりあえずビルドは通るかも
> ```

やってみたが、問題なく通った。
残りパッチは少ないので、コンフリクトもすぐ解決できると思うが。

ashie, 2019-04-09:
> # Wayland版パッチは不要なものは削除したが、残りの移植が済んでない

修正済み。 コミットメッセージ等はもうちょっと意図がわかるようにしておきたいが。
不要パッチはファイル自体はまだ残っていたので削除した。

メモ:

bug1374136-Enable-sharing-SharedSurface_EGLImage.patch はもともとはRZ/G1でWebGLがクラッシュする問題のワークアラウンドのために入れていたが、 0001-GLLibraryLoader-Use-given-symbol-lookup-function-fir.patch の修正で直ったので、必要なくなっている。毎フレーム毎にEGLImageを再確保する無駄を省けるので多少の性能改善はあるかもしれないが、以前測定したときには有意な差は無かった。独自パッチは減らしたいという観点で、このタイミングで削除した。

削除したその他のパッチはバックポートパッチなので68では必要無い。

ashie, 2019-04-09:
メモ:

* またmeta-browserに依存が増えそうなので、OSSystems/meta-browser にissueを立てて相談したい
* `AS`の設定はmeta-clangにあった方が良さそうなので、そちらにissueを立てて聞いてみる
* cbindgenはmeta-rustにフィードバック
* rustfmtやnodejsはビルドに必要無さそうなら無視、必要なら適当なレイヤ探してを活用するかフィードバック
* `RUST_TARGET`と`RUST_STDLIB`のパッチは整理して本体に入れたい。

ashie, 2019-04-09:
AArch64向けにビルドを走らせてみたらこうなった

```
|   clang-8: error: the clang compiler does not support
|   '-mtune=cortex-a57.cortex-a53'
```

cosmo0920, 2019-04-09:
-mtuneを設定しているのがどこかわからないですけど、 `-march=armv8-a` に差し替えると動きますかねぇ。
https://bugs.chromium.org/p/chromium/issues/detail?id=554632

ashie, 2019-04-09:
Yoctoに`TUNE_FEATURES`という環境変数があるけど、これがgcc用になってるとか？

cosmo0920, 2019-04-09:
```
~/ebisu-bsp$ ag cortex-a57 meta-renesas/
meta-renesas/meta-rcar-gen3/conf/machine/include/tune-cortexa57.inc
4:TUNEVALID[cortexa57] = "Enable Cortex-A57 specific processor optimizations"
11:TUNE_CCARGS .= "${@bb.utils.contains("TUNE_FEATURES", "cortexa57", " -mtune=cortex-a57", "", d)}"

meta-renesas/meta-rcar-gen3/conf/machine/include/tune-cortexa57-cortexa53.inc
4:TUNEVALID[cortexa57-cortexa53] = "Enable big.LITTLE Cortex-A57.Cortex-A53 specific processor optimizations"
11:TUNE_CCARGS .= "${@bb.utils.contains("TUNE_FEATURES", "cortexa57-cortexa53", " -mtune=cortex-a57.cortex-a53", "", d)}"
```

meta-renesas内部みたいですヨ。

cosmo0920, 2019-04-09:
$ ag cortex-a57 meta-renesas/
meta-renesas/meta-rcar-gen3/conf/machine/include/tune-cortexa57.inc
4:TUNEVALID[cortexa57] = "Enable Cortex-A57 specific processor optimizations"
11:TUNE_CCARGS .= "${@bb.utils.contains("TUNE_FEATURES", "cortexa57", " -mtune=cortex-a57", "", d)}"

meta-renesas/meta-rcar-gen3/conf/machine/include/tune-cortexa57-cortexa53.inc
4:TUNEVALID[cortexa57-cortexa53] = "Enable big.LITTLE Cortex-A57.Cortex-A53 specific processor optimizations"
11:TUNE_CCARGS .= "${@bb.utils.contains("TUNE_FEATURES", "cortexa57-cortexa53", " -mtune=cortex-a57.cortex-a53", "", d)}"

cosmo0920, 2019-04-09:
`DEFAULTTUNE = "cortexa57"`を設定すればbig.LITTLE構成としては使えなくなりますが、ビルドは通せるんじゃないでしょうか。

cosmo0920, 2019-04-09:
> Yoctoに`TUNE_FEATURES`という環境変数があるけど、これがgcc用になってるとか？

https://en.wikichip.org/wiki/arm_holdings/microarchitectures/cortex-a57#Compiler_support

> -mtune=cortex-a57.cortex-a53

はgcc用のオプションのようです。

ashie, 2019-04-09:
なるほど、ありがとうございます :smiley:

ashie, 2019-04-09:
> rustfmtやnodejsはビルドに必要無さそうなら無視、必要なら適当なYoctoレイヤで対応

試してみたらこうなった。 Node.jsはやっぱり必要か。

```
0:04.27 checking for rustfmt... not found
...
 0:04.34 checking for nodejs... no
 0:04.34 ERROR: could not find Node.js executable later than 8.11; ensure `node` or `nodejs` is in PATH or set NODEJS in environment to point to an executable.
```

cosmo0920, 2019-04-09:
meta-openembeddedのthudブランチにNodejsの8.12が居るみたいなのでここらへんから取ってくると良さそうです。
https://github.com/openembedded/meta-openembedded/blob/thud/meta-oe/recipes-devtools/nodejs/nodejs_8.12.0.bb

ashie, 2019-04-09:
> AArch64向けにビルドを走らせてみたらこうなった
> 
> ```
> |   clang-8: error: the clang compiler does not support
> |   '-mtune=cortex-a57.cortex-a53'
> ```

以下でいけそう

```
TUNE_FEATURES_remove_toolchain-clang="cortexa57-cortexa53"
```

ashie, 2019-04-10:
ホスト側のコンパイラとしてgcc-6.1以降を求められる。
これまで推奨環境としていたUbuntu-16.04では辛そう。
18.04なら大丈夫。

```
|  0:02.52 checking for the host C compiler... /build/rcar-gen3/build/tmp/hosttools/gcc
|  0:02.64 checking whether the host C compiler can be used... no
|  0:02.64 DEBUG: <truncated - see config.log for full output>
|  0:02.64 DEBUG: | #elif __NetBSD__
|  0:02.64 DEBUG: | %KERNEL "NetBSD"
|  0:02.64 DEBUG: | #elif __APPLE__
|  0:02.64 DEBUG: | %KERNEL "Darwin"
|  0:02.64 DEBUG: | #endif
|  0:02.64 DEBUG: | #if __ANDROID__
|  0:02.64 DEBUG: | %OS "Android"
|  0:02.64 DEBUG: | #endif
|  0:02.64 DEBUG: | #if __BYTE_ORDER__ == __ORDER_LITTLE_ENDIAN__
|  0:02.64 DEBUG: | %ENDIANNESS "little"
|  0:02.64 DEBUG: | #elif __BYTE_ORDER__ == __ORDER_BIG_ENDIAN__
|  0:02.64 DEBUG: | %ENDIANNESS "big"
|  0:02.64 DEBUG: | #endif
|  0:02.64 DEBUG: Executing: `/build/rcar-gen3/build/tmp/hosttools/gcc -std=gnu99 -E /tmp/conftest._UENkA.c`
|  0:02.64 DEBUG: COMPILER = gcc
|  0:02.64 DEBUG: VERSION = 5.4.0
|  0:02.64 DEBUG: STDC_VERSION = 199901L
|  0:02.64 DEBUG: CPU = x86_64
|  0:02.64 DEBUG: KERNEL = Linux
|  0:02.64 DEBUG: ENDIANNESS = little
|  0:02.64 ERROR: Only GCC 6.1 or newer is supported (found version 5.4.0).
|  0:02.69 *** Fix above errors and then restart with\
|  0:02.69                "./mach build"
```

cosmo0920, 2019-04-10:
一応Ubuntu 16.04でもubuntu-toolchain-rのPPAを追加するとgcc 6.1.0以降もインストールできますね。
https://launchpad.net/~ubuntu-toolchain-r/+archive/ubuntu/test

ashie, 2019-04-12:
R-Car M3で動いたので、SunSpiderとOctane-2.0のスコアを貼っておく。

SunSpider: 1088.3ms
Ocatane 2.0: 3131

[image]

[image]

[他の環境での結果はこちら](https://github.com/webdino/gecko-embedded/wiki/Benchmark)

60ではIonMonkeyがAArch64未対応だったため惨憺たる有様だったが、今回は他と同程度のオーダーはでているようだ。Octane-2.0のスコアはまだArm 32bitより悪いが、ベンチマークだけで性能を語るのは軽率だし、実用上は問題ないレベルだと思う。

ashie, 2019-04-12:
メモ: 

今回は必要なさそうだけど、WebRTCの対応プラットフォームを増やすにはgn-configsに設定を追加する必要がある。

* [60ESRのときのメモ](https://github.com/webdino/gecko-embedded/issues/60#issuecomment-365825848)
* [ソース内のドキュメント](https://dxr.mozilla.org/mozilla-central/source/build/docs/gn.rst)
* [gn](https://gn.googlesource.com/gn/)
  * [linux-amd64バイナリ](https://chrome-infra-packages.appspot.com/dl/gn/gn/linux-amd64/+/latest)

ashie, 2019-04-12:
その他、R-Car M3で動かしてみてのメモ

* 今のところ常にGDK_BACKEND=waylandが必要
* OpenGL Compositior問題なく使える
* WebGL1は動くが有効化しているとabout:supportでクラッシュする
* WebRenderは有効化するとクラッシュする
* e10sは今回は問題なく動きそう
  * なのでAPZも動く
* about:supportのAzureCanvasAcceleratedの欄が無くなってる。なんでだろう？
  * なのでGPU効いてるのかどうかわからない。がGUIMark3 Bitmapはコンスタントに60fps出る。
* レシピで追加してるvendor.jsで以下のエラー（Debianのパッチが当たってないので`lockPref`を認識しない）
```
/usr/lib/firefox/defaults/pref/vendor.js:8: prefs parse error: unknown keyword
/usr/lib/firefox/defaults/pref/vendor.js:9: prefs parse error: unknown keyword
/usr/lib/firefox/defaults/pref/vendor.js:10: prefs parse error: unknown keyword
```
* OpenMAXは相変わらず動かない

ashie, 2019-04-17:
```
 0:04.34 ERROR: could not find Node.js executable later than 8.11; ensure `node` or `nodejs` is in PATH or set NODEJS in environment to point to an executable.
```

meta-openembeddedに含まれるNode.js:

* sumo(2.5): 8.9.4
* thud(2.6): 8.12.0

ashie, 2019-05-15:
どうせビルド環境の大半は Linux x86_64 だし、ツールチェーンは`./mach bootstrap`でセットアップできるのに、ここで労力を割くのはなんとも馬鹿馬鹿しい。ツールチェーンのビルドで結構時間がかかるのもかったるい。

最初に以下を実行して、依存を極力減らす方向性も模索してみる。

```
$ ./mach bootstrap --no-system-changes --no-interactive --application-choice=browser
```

ashie, 2019-05-15:
現在のmozilla-centralではrust-1.34が要求されるが、meta-rustはまだ1.33なのでビルドできなくなっている。
そのうちアップデートされるだろうからしばらく様子見。

こちらもrustupの再導入を検討してみるか...（60対応の初期の頃はrustupを使ってた）

ashie, 2019-05-23:
> 最初に以下を実行して、依存を極力減らす方向性も模索してみる。
> 
> ```
> $ ./mach bootstrap --no-system-changes --no-interactive --application-choice=browser
> ```

firefox-60.0b3のtarballでやったらこうなった

```
Installing Stylo and NodeJS packages requires a checkout of mozilla-central.
Once you have such a checkout, please re-run `./mach bootstrap` from the
checkout directory.
```

ashie, 2019-05-30:
>     * [ ]  cbindgenはmeta-rustに入れる？

レシピは[cargo-bitbake](https://crates.io/crates/cargo-bitbake)で簡単に作れることが[分かった](https://github.com/OSSystems/meta-browser/issues/265#issuecomment-497193408)

ashie, 2019-06-05:
今のところ見えてるビルド上の課題

* meta-rustに最新版Rustを投入しないといけない
* Node.js: rocko向けにレシピをバックポートするか、ホスト側のを使う手順を用意するか
* `AS`の設定: 設定入れる箇所によって挙動が変わってビルドできないことがあるのをちゃんとやるにはどうすれば良いのか調べられてない
  * 本体は暗黙的にclangが必要（GASとかではビルド失敗する）
  * ICUは逆にclangでビルドできない（GASかYASM）

あとは、最新版のβはまだビルドしてないので、他に問題があるかどうか確認できてない。
meta-rustのバージョンを上げるのは、誰もやらないようであれば6月後半辺りに対応する予定。
それまではペンディング。

ashie, 2019-06-05:
ランタイムの課題として現時点で認識しているのは以下

* about:supportがクラッシュする: [Bug1548911かな?](https://bugzilla.mozilla.org/show_bug.cgi?id=1548911)

ashie, 2019-07-11:
現状を共有しておきます

* 先日リリースされた68.0esrは以下ブランチに投入済み
 *  https://github.com/webdino/meta-browser/tree/firefox-68-wip
* 上記でビルドしたものがR-Car M3で起動する程度は確認できている
* 参考までに @ashie のビルド環境は以下の通り
  * ビルドOSはUbuntu 18.04
  * Node.jsは[mozilla-central](https://hg.mozilla.org/mozilla-central/)で`./mach bootstrap`してインストールされるものを使用
  * meta-browser: 上記ブランチ(faaeef64487c970ef77884c250723c93015f7585)
  * meta-rust: masterブランチ(9487b089ea4779c2b494b17b9254219226efa539)
  * meta-clang: masterブランチ(1feea26ec0843b7e621973c0feafa25b8ca1b2de)
 * ビルド設定は後述
* Wayland版は環境変数`GDK_BACKEND=wayland`をつけて起動する必要あり
* about:supportはクラッシュする（69以降では直っていると思われる）

bblayers.conf: 以下を追加
```
BBLAYERS += " ${TOPDIR}/../meta-browser "
BBLAYERS += " ${TOPDIR}/../meta-rust "
BBLAYERS += " ${TOPDIR}/../meta-clang "
```

local.conf: 以下を追加
```
IMAGE_INSTALL_append = " firefox "
IMAGE_INSTALL_append = " ttf-sazanami-gothic ttf-sazanami-mincho "
PACKAGECONFIG_append_pn-firefox = " gpu "
PACKAGECONFIG_append_pn-firefox = " webgl "
#HOSTTOOLS += "autoconf2.13"
TOOLCHAIN_pn-firefox = "clang"
AS_pn-firefox="${TARGET_PREFIX}clang"
TUNE_FEATURES_remove_toolchain-clang="cortexa57-cortexa53"
```

OpenMAXについては追加で以下が必要と思われる（未確認）

* local.confに`PACKAGECONFIG_append_pn-firefox = " openmax "`
* レシピのopenmax.jsに`pref("media.omx.enabled", true);`あるいはprefs.jsに`user_pref("media.omx.enabled", true);`
* #58 のパッチ

ashie, 2019-07-11:
meta-clang絡みの設定の整理がついたらコミットをまとめてupstreamに還元したい。

ashie, 2019-07-15:
> meta-clang絡みの設定の整理がついたらコミットをまとめてupstreamに還元したい。

https://github.com/OSSystems/meta-browser/pull/284 にプルリク済み

ashie, 2019-07-17:
アップストリームにマージされた

> TOOLCHAIN_pn-firefox = "clang"
> AS_pn-firefox="${TARGET_PREFIX}clang"

この辺りはレシピの方に入れたので、local.confでは不要になった。
（正しい方法かどうかよくわからないけど）

#72 を緩和するパッチとかはいい加減すぎるのでアップストリームには入れてない。
Wayland版でのabout:supportのクラッシュもまだ特に何もしていない。
その辺は必要に応じて別issueにて対応ということで、このissueは閉じます。

closed 2019-07-17 by ashie

ashie, 2019-08-05:
>     * Waylandではabout:supportがクラッシュする: [Bug1548911](https://bugzilla.mozilla.org/show_bug.cgi?id=1548911)かな?

バックトレースとってみたら全然違った。
[Bug 1209612](https://bugzilla.mozilla.org/show_bug.cgi?id=1548911)と似たような問題で、`eglQueryString(nullptr, LOCAL_EGL_EXTENSIONS)`が`NULL`を返して、それが最終的に`strlen()`に渡ってクラッシュしてる。

```
Thread 1 "firefox" received signal SIGSEGV, Segmentation fault.
strlen () at ../sysdeps/aarch64/strlen.S:94
94              ldp     data1, data2, [srcin]
(gdb) where
Thread 1 (Thread 0xffffbf6f0430 (LWP 3214)):
#0  strlen () at ../sysdeps/aarch64/strlen.S:94
#1  0x0000ffffb7dbce90 in nsTSubstring<char>::Append(char const*, unsigned int, std::nothrow_t const&) () from /usr/lib/firefox/libxul.so
#2  0x0000ffffb7dbcd38 in nsTSubstring<char>::Append(char const*, unsigned int) () from /usr/lib/firefox/libxul.so
#3  0x0000ffffb94c2a20 in mozilla::WebGLExtensionMOZDebug::GetParameter(JSContext*, unsigned int, JS::MutableHandle<JS::Value>, mozilla::Erro\
rResult&) const () from /usr/lib/firefox/libxul.so
#4  0x0000ffffb91858a0 in mozilla::dom::MOZ_debug_Binding::getParameter(JSContext*, JS::Handle<JSObject*>, mozilla::WebGLExtensionMOZDebug*, \
JSJitMethodCallArgs const&) () from /usr/lib/firefox/libxul.so
#5  0x0000ffffb944d5b4 in bool mozilla::dom::binding_detail::GenericMethod<mozilla::dom::binding_detail::NormalThisPolicy, mozilla::dom::bind\
ing_detail::ThrowExceptions>(JSContext*, unsigned int, JS::Value*) () from /usr/lib/firefox/libxul.so
#6  0x0000ffffbaa90278 in js::InternalCallOrConstruct(JSContext*, JS::CallArgs const&, js::MaybeConstruct) () from /usr/lib/firefox/libxul.so
#7  0x0000ffffbaa8c4bc in Interpret(JSContext*, js::RunState&) () from /usr/lib/firefox/libxul.so
#8  0x0000ffffbaa85730 in js::RunScript(JSContext*, js::RunState&) () from /usr/lib/firefox/libxul.so
#9  0x0000ffffbaa90408 in js::InternalCallOrConstruct(JSContext*, JS::CallArgs const&, js::MaybeConstruct) () from /usr/lib/firefox/libxul.so
#10 0x0000ffffbae15924 in js::jit::DoCallFallback(JSContext*, js::jit::BaselineFrame*, js::jit::ICCall_Fallback*, unsigned int, JS::Value*, J\
S::MutableHandle<JS::Value>) () from /usr/lib/firefox/libxul.so
#11 0x000029ab90b131d0 in ?? ()
#12 0x0000ffffffffca48 in ?? ()
Backtrace stopped: previous frame inner to this frame (corrupt stack?)
```

RZ/G1の時は上記Bugと同じで`eglQueryString()`自体でクラッシュするので[その箇所をAndroid版と同様にまるごと殺している](https://github.com/webdino/meta-browser/blob/firefox-60esr/recipes-browser/firefox/firefox/wayland/egl/0002-Disable-query-EGL_EXTENSIONS.patch)（R-Car Gen3の場合は上記の通りクラッシュはしないが`NULL`が返る）。

そのパッチは今でも当たっているが、68では[同様のコードが増えて](https://dxr.mozilla.org/mozilla-central/rev/30a8df41ff6db0323d045bdc56cb5f0c95e92b9a/gfx/gl/GLContextProviderEGL.cpp#529)、そっちでクラッシュしてるようだ。

ashie, 2019-08-05:
>     * WebRenderは有効化するとクラッシュする

リリース版では動作している（性能評価はまったく出来ていないが）。

ashie, 2019-12-09:
>     * about:supportのAzureCanvasAcceleratedの欄が無くなってる。なんでだろう？
>       
>       * なのでGPU効いてるのかどうかわからない。がGUIMark3 Bitmapはコンスタントに60fps出る。

今更だがメモ。
経緯は以下辺りのBugを見ると分かる。

* https://bugzilla.mozilla.org/show_bug.cgi?id=1468801
* https://bugzilla.mozilla.org/show_bug.cgi?id=1530471

SkiaGLは68では削除されている。

  * 特殊なベンチマークを除けばソフトウェアで十分性能が出ている
  * メンテナンスが大変
  * 本当に性能を求めるコンテンツではWebGLを使うべき

ということらしい。
確かに、60ではGUIMark3とかではほとんど差が見られなくなっていた。

ashie, 2019-12-19:
今更だけど、リリースノートにARM64のJITをサポートしたと書かれているのは69からだったようだ。

* https://www.mozilla.org/en-US/firefox/69.0/releasenotes/

68でも間違いなくIonMonkeyは有効化されているのだが、ベンチマークスコア的には確かにもうちょっと頑張って欲しい感じだった。
69の方が大きく性能向上している可能性がありそうだ。

https://bugzilla.mozilla.org/show_bug.cgi?id=1551339

> This is preliminary work to allowing encoding of JSCVTFP, the instruction that exists
> on new AArch64 devices that greatly speeds up websites that use　floating-point math.

ashie, 2019-12-20:
69もわずかな変更でビルドできたので試してみたが、SunSpiderのスコアは特に変わらず

dynamis, 2019-12-20:
どちらかというと 68 のリリースノートに書き忘れてただけなんじゃないかと思ってたりします。リリースノート書いてるメンバーはこれまでも頻繁にミス書いてて適当だし、実際 mozilla68 で有効化されているはずなので:

Enable ARM64 Ion in-browser https://bugzilla.mozilla.org/show_bug.cgi?id=1536220

69 以降は積み残しの整理を続けてるだけなのではないかと。

プロジェクトページなども特に 69 に postpone するとかいったやり取りは見当たらないし:
https://wiki.mozilla.org/Mobile/ARM64