webdino/gecko-embedded #71: プロジェクト Web サイトを作る

dynamis, 2019-03-27:
昨年ようやく本プロジェクトで移植した Gecko が RZ/G1 向けに正式リリースしているが、ルネサスのサイト上ではあまり説明や記載がない。

VLP の説明などで HTML が含まれていることや https://www.renesas.com/jp/ja/products/rzg-linux-platform/verified-linux-package.html RZ/G2 リリースで HTML5 サポートに言及している程度 https://www.renesas.com/jp/ja/about/press-center/news/2019/news20190221.html

記事などで紹介するのにも、採用検討者に見てもらうのにも、このリポジトリ URL だけではなく Web サイトを用意して認知を広げやすいように簡単なページを用意する。

2016 年から確保してる塩漬けドメインを使う想定: https://gecko-embedded.org/
Slack も確保済みのものに self invite リンクを用意: https://gecko-embedded.slack.com/

- シンプルな markdown を編集して netlify でデプロイするパターンを想定
- github pages 互換にするか気にせず huxt や hugo などにするかは要検討
- ページ末尾には github 上で markdown を編集するページへのリンクを付ける
- プロジェクト概要説明スライドなども 4 月に更新するものを掲載する
- ビルド手順書などは github wiki に残す想定だが、web サイト配下に移動させて見やすくするのもありかも知れない？その辺は日々更新する人次第感はある
- 英語情報は e-linux サイトに勝手に追加する？要検討

ashie, 2019-03-28:
ビルド手順に関してはどう整理するのか悩ましいのですが、ジェネリックな話はレシピ本体に入れていっているので

https://github.com/OSSystems/meta-browser/tree/master/recipes-browser/firefox

ここに環境毎の設定の勘所とかの話も入れていくのが良いのだろうか？とも思っています。

dynamis, 2019-04-08:
なるほど、レシピ本体の Readme とかにも書いて Upstream に入れていくのも含めてですね。

基本的には独自 Web サイト的なところの情報源は最小限、各リポジトリの Readme など既存の場所に書いていくのが良く、リポジトリも一杯あるのでそこへの誘導、初めてさん向けの記述等だけ Web / wiki とかにしていけると良いのかなと思っています。作成できるタイミングにまた一通り見ながら考えてみるのでまたアドバイスくださいませ。

Slack については少し前に各リポジトリからのコミット通知が来る設定のチャンネルを作った状態ですが、こちらから参加できるリンク (有効期限無し) 用意しました:
https://join.slack.com/t/gecko-embedded/shared_invite/enQtNjAzNTI2Njc2NjI5LTczNDIwYjM5Njk5N2IzZDhmNDE0YmQ4ZDgyYWM5MGI0ZjVmYjM0OWRlOTIwYjlmNWM1ZmE5NWEyYTczZTY0MTc

dynamis, 2019-05-31:
そのうちレシピ本体への記述追加や Web サイトに情報を移動して見やすくしたいとは思っているが、まだちょっと Web の準備まで手が回っていないので、取り急ぎは RZ/G2E 向けのページも既存情報を元に Wiki に書き始めてみた:
https://github.com/webdino/meta-browser/wiki/Build-RZ-G2E

dynamis, 2019-10-23:
組込向けには React/Angular よりも軽量なフレームワークが望ましく、それを使い慣れるという意味でも軽量フレームワーク Inferno, Preact, Svelte 辺りのいずれかベースの HeadlessCMS で構築する方向で検討中。