webdino/gecko-embedded #76: RZ/G2 の BSP で USB Web カメラが認識されない

dynamis, 2019-05-31:
WebRTC をテストしようと思って USB WebCam を Ebisu や RZ/G2E ボードに接続しても high-speed USB device とだけ認識されてブラウザからカメラとして認識されないため使えない。

USB カメラではなく他の端子でカメラを繋げば大丈夫なのか等は未確認。

畑ケさんのログ情報:

```
[  981.751986] usb 3-1.4: new high-speed USB device number 7 using xhci-hcd
```

```
root@ebisu:~# lsmod
Module                  Size  Used by
vspm_if                40960  0
vsp2                   61440  0
vspm                   90112  2 vsp2,vspm_if
uvcs_drv               45056  0 # UVCモジュール自体は存在
pvrsrvkm             1232896  4
mmngrbuf               16384  0
mmngr                  20480  0
ipv6                  385024  26
```

YoshihiroOota, 2019-06-04:
uvcs_drv はおそらく Unified Video CODEC Server のドライバで、ルネサス製(?)と思われます。
現状、UVC モジュール自体が VLP に含まれておらず、含める形でカーネルを再ビルドする必要があります。
今後、UVC を標準として入れるかどうかは、BSP 担当者に確認中です。

取りあえずUVC を入れてリビルドしたカーネルを使用した場合、https://mozilla.github.io/webrtc-landing/gum_test.html にてカメラ入力の動画が表示できることを確認できました。
USB Webカメラを挿したときのログは、下記となります。
```
[   47.039379] usb 1-1.1: new high-speed USB device number 3 using ehci-platform
[   47.174403] uvcvideo: Found UVC 1.00 device MicrosoftR LifeCam HD-3000 (045e:0779)
```

dynamis, 2020-03-19:
Gecko ビルド時のみレシピを変える、デフォルト BSP 側はそのままということで進んでいるので制限事項・修正予定なし (wontfix) としてクローズします。

closed 2020-03-19 by dynamis