webdino/gecko-embedded #79: OS デフォルトのカーソルアイコンが取得できていない

dynamis, 2019-06-19:
リンクなどにマウスオーバーしたときにカーソルが手のアイコンなどになるはずがならない。
多分 Gdk のアイコンファイルが足りてないのかアクセスできないのか。

ターミナルに残るメッセージ: `Gdk-Message: Unable to load hand2 from the cursor theme`

Gecko 60 on RZ/G2E で確認。それ以外の発生条件は未確認だが多分広範に発生している。

ashie, 2019-08-02:
adwaita-icon-theme-cursors のパッケージを追加すると解消する。

RZ/G1のBSPがjethroベースだったときにはGTKを更新する必要があった（3.16以前はWayland対応が貧弱だった）ため以下の追加レイヤを使っていて

* https://github.com/webdino/meta-gecko-embedded/

そのついでに以下の追加レシピでこの問題を解決していた（完全に忘れてた）

* https://github.com/webdino/meta-gecko-embedded/blob/master/meta-poky-jethro/recipes-graphics/images/core-image-weston.bbappend

rockoに更新されてからはGTKを無理に上げる必要が無くなり、上記の追加レシピも使わなくなったため、アイコンがインストールされなくなっていた。

とりあえずブートイメージのビルド手順を修正すれば解決する話だけど、gtk+3の依存にadwaita-icon-theme-symbolicの依存だけ入っていてadwaita-icon-theme-cursorsの依存が入っていないので、pokyの側で修正した方が良いのかもしれない。

anyakichi, 2021-01-08:
本件についてはパッケージの依存関係的に解決しなくても、local.conf に
```
IMAGE_INSTALL_append = " adwaita-icon-theme-cursors "
```
を書いておけばよいということでもあり、ひとまずビルド手順にこれを加えることで解決したことにする。

RZ/G2 向けのビルド手順については更新済み。

https://github.com/webdino/meta-browser/wiki/Build-RZ-G2#gecko-68-%E3%82%92%E5%90%AB%E3%82%81%E3%81%9F%E3%83%93%E3%83%AB%E3%83%89

closed 2021-01-08 by anyakichi