webdino/gecko-embedded #80: Native Messaging 対応の検討

dynamis, 2019-06-19:
ブラウザでのネイティブ連携の代表的手段の一つ、Native Messaging について対応可能かどうか、可能であればどのような手順ですればよいか、パフォーマンス面の課題は無いかといった確認をしておきたい

https://developer.mozilla.org/ja/docs/Mozilla/Add-ons/WebExtensions/Native_messaging

勿論これは Gecko エンジンの HTML5 対応の機能ではなく Firefox ブラウザの Extension サポート機能の一部という特殊な位置づけなので、動作しないならしないで構わないがまずは現状を把握しておくところから

ashie, 2019-06-19:
試してないですけど、普通に動くような気がします。
ちょうど最近Native Messagingを使う拡張を作ったのですけど

  * https://addons.mozilla.org/ja/firefox/addon/ime-mode-we/

これのコマンド呼び出し部分を適当なものに差し替えれば、動くかどうか程度はさくっと確認できるんじゃないかと（別にこれじゃなくてもそこのサンプルで十分ですが、なんかタイムリーだったもので）。

dynamis, 2019-06-20:
おぉ、Firefox 対応 AMO 署名付き Native Messaging アドオンありがたい！ (今すぐ試したいというわけではないのだけど何となく issue にしておくかと思って書いたらタイミング良かった)

試すときは (あまり大きなアドオンでの利用例ではなく) MDN のドキュメント見るか Google のサンプルを書き換えてと思ってたのですが ime-mode-we もかなりシンプルかつ実用的でとてもいいですね。

url memo:
- https://github.com/clear-code/ime-mode-we
- https://developer.chrome.com/extensions/nativeMessaging
- https://developer.chrome.com/extensions/examples/api/nativeMessaging/app.zip

ashie, 2019-06-27:
>   * https://addons.mozilla.org/ja/firefox/addon/ime-mode-we/
> 
> これのコマンド呼び出し部分を適当なものに差し替えれば、動くかどうか程度はさくっと確認できるんじゃないかと

試しにメッセージをファイルに書き出すコマンドに差し替えてみたところ、動作していることを確認できた。
例えば/usr/local/bin/ibus-set-input-modeを以下に差し替える。

```
#!/bin/sh
echo $1 > "${HOME}/ime-mode.txt"
```

dynamis, 2019-06-28:
素早く検証までありがとうございます。使えることも検証済みとしてクローズします。

勿論ホストアプリケーションとの間にはコンテンツ側から拡張機能の JS に渡して更に JSON (UTF-8) に変換などしたうえで受け取り側は JSON.parse 処理が必要になるし、応答を得るときも同様にホスト側で JSON にした上でデータ長を付与して拡張機能側に送るということになるので、そのオーバーヘッドが許容できる場合に限る。

TCP/IP 層を経由する WebSocket とどっちがオーバーヘッドが大きくなるかは未確認だが、オーバーヘッドを許容できないケースは普通 (Gecko を自分でビルドできない理由がない限り) #85 に書いた通り独自 API を追加実装するべき。

closed 2019-06-28 by dynamis