webdino/gecko-embedded #81: RZ/G2: WebGL2 サポート

dynamis, 2019-06-21:
RZ/G2 は OpenGL ES3 をサポートしているので WebGL2 のサポートが出来ても良さそうだが、現状 WebGL2 が有効化されていない問題

- WebGL 確認ページ: https://webglreport.com/?v=2
- about:support でも確認可能

原因については未調査。

dynamis, 2019-09-12:
実際の GL コンテンツを使った動作確認用ページ
- WebGL 2 Samples Pack http://webglsamples.org/WebGL2Samples/
  - 説明は github https://github.com/WebGLSamples/WebGL2Samples

anyakichi, 2019-10-04:
RZ/G2 で WebGL2 を使用できないのには、2 つ問題がある。

1. glxtest が X11 前提で実装されている
2. dlopen() でロードできるライブラリの上限に達している
 
## glxtest

Firefox 起動後に WebGL2 が使えるかどうかを判断するために、glxtest (https://github.com/mozilla/gecko-dev/blob/esr68/toolkit/xre/glxtest.cpp) を使用している。glxtest では、#ifdef などなしに直接 XOpenDisplay() 等を呼んでおり、明らかに Wayland 対応していない。

XWayland を有効にするか、glxtest の Wayland 対応を行うかが必要になる。少なくとも XWayland を導入することで glxtest のエラーが起こらなくなることは確認できている。

https://bugzilla.mozilla.org/show_bug.cgi?id=1570441

## dlopen()

dlopen() で TLS (Thread Local Storage) を要求するライブラリをロードすると、glibc の内部的にそのライブラリ用に DTV (Dynamic Thread Vector) を消費するようである。Firefox は多数のライブラリを dlopen() によりロードするので、glxtest において libGL.so.1 をロードする際に dlopen() が予め確保された DTV 数の上限に達しエラーとなる現象が見られた。

上限は DTV_SURPLUS (14) によって決定されているので、これを 32 などに増加させると dlopen() のエラーが回避できる。少なくともこの glibc の修正と XWayland のインストールにより、RZ/G2E で WebGL2 が動作することは確認済み。

なお、glibc-2.22 では dlopen() の内部処理が変わっており、DTV_SURPLUS を上限として使用しなくなっている。未確認だが、glibc-2.22 以降を使うことでもこの問題は回避できるものと思われる。

anyakichi, 2019-10-24:
XWayland の有効化については、local.conf などに

    DISTRO_FEATURES_append = " x11"

と書いてビルドした上で、/etc/xdg/weston/weston.ini の core セクションに

    [core]
    xwayland=true

のように書くことで行うことができる。

glibc のパッチを含めたビルドは meta-browser の gem-firefox-68.0.0esr ブランチを使用することで行うことができる。
https://github.com/webdino/meta-browser/tree/gem-firefox-68.0.0esr

以下の動作について確認済み。

* https://webglreport.com/?v=2 でサポートされていることになること。
* http://webglsamples.org/WebGL2Samples/ のサンプルが一通り動作すること。

anyakichi, 2019-10-31:
ビルド手順および Docker によるビルド環境について、WebGL2 が有効になる形でアップデートを行った。

* https://github.com/webdino/meta-browser/wiki/Build-RZ-G2
* https://github.com/anyakichi/docker-gecko-embedded-builder

ashie, 2020-01-09:
glxtestは https://phabricator.services.mozilla.com/D57474 でWayland/EGLに対応したようですね。
（名前がglxtestのままなのがまぎらわしいけど）。

closed 2022-09-05 by dynamis