webdino/gecko-embedded #86: オフラインビルド

ashie, 2019-06-28:
Cargoがつい最近までオフラインビルドに対応していなかったため、オンラインでしかビルドできなかった。
Cargoについては https://github.com/rust-lang/cargo/commit/309eb874fdd6636317927645b6a504389fac8a66#diff-9a700b7a7dcf88f0eb41583a366c9d61 でstableチャネルでもオフラインに対応したので、設定次第ではオフライン対応できるだろうか?

ただしstage0で使うプレビルドのcargoもオフライン対応させないと、完全なオフライン化はできないかもしれない。

ashie, 2019-06-28:
参考: 最近のlibrsvgもRustに置き換わっているが、[librsvgのCOMPILIG.md](https://github.com/GNOME/librsvg/blob/master/COMPILING.md#building-with-no-network-access)にオフラインビルドについての言及がある。

ashie, 2019-06-28:
Yoctoの方は`BB_NO_NETWORK=1`でオフライン化できる。
meta-rustでもこれを見て`offline`オプションを切り替えられるようになっていると良いかも。

ashie, 2019-06-28:
cargoのダウンロード先は [ meta-rust/classes/cargo_common.bbclass の `export CARGO_HOME = "${WORKDIR}/cargo_home"
`](https://github.com/meta-rust/meta-rust/blob/9487b089ea4779c2b494b17b9254219226efa539/classes/cargo_common.bbclass#L16) でセットされている。

YoshihiroOota, 2019-10-23:
すみません、今更ですが、もしかして依頼の仕方が間違っていた気がしてきました。
(1) 普通にオンラインでビルドする。
(2) build ディレクトリの下の cache や tmp を削除する。
(3) BB_NO_NETWORK=1 に設定し、更に物理的にネットワークから切断する。
(4) GStreamer のオフラインビルド用の変更を行い、再度ビルドする。
これでビルドが通ったため、その後の oss_packages を Gecko のオフラインビルド環境として提供できるのではないかと VLP 担当者より連絡がありました。
これまでの ahie さんの調査状況を拝見しますとそう簡単にはいかないように思え、ここに何か差異がありそうなのですが…

ashie, 2019-10-23:
sstate-cacheの方も削除されていますか？あるいは残されていますか？

YoshihiroOota, 2019-10-23:
はい、sstate-cache も削除しております。

YoshihiroOota, 2019-10-24:
以前に ashieさんの方で試されたオフラインビルドも、上記と同じやり方でしたでしょうか。

ashie, 2019-10-24:
> (4) GStreamer のオフラインビルド用の変更を行い、再度ビルドする。

というのはよくわかりませんが、

* オンラインでビルド後
* tmp、cache、sstate-cacheを全て削除、downlaodsは残す
* `BB_NO_NETWORK=1 `を設定、物理的にもネットワークを遮断

というところは同じかと思います。
以前に試したときにはこれでビルドできていなかったと思いますが、ちょっと今はまだ試せてません

YoshihiroOota, 2019-10-25:
情報ありがとうございます、手順は同じかと思います。不思議…。
> (4) GStreamer のオフラインビルド用の変更を行い、再度ビルドする。

GStreamer がオフラインビルドで失敗することはご存知と思いますが、それの対策としてビルド前に各 .gitmodules の url の内容を手動で変更しています。本件には関係なさそうですが...

dynamis, 2022-09-05:
当面必要性がないと判断し一旦 close

closed 2022-09-05 by dynamis