webdino/gecko-embedded #87: MSE による動画ストリーミング再生ができない

dynamis, 2019-07-03:
Video タグで単体の動画を src に設定したり webm ファイル単体を開くなどするケースでは正常に最後まで動画再生が可能だが、MSE でストリーミングとしてチャンクファイル (分割された動画ファイル) を流し込んでいる場合には最初の動画チャンクファイルの終端までは再生されるがそこで映像が停止して次の動画ストリーミングの再生がされない (プレイヤー上は再生が続いていることになっている)。

ひとまず Gecko 60 build on RZ/G2E で確認したが恐らく他のボード、ビルドでも再現する

- H.264 非対応のビルドで webm (VPx) でのテストをする場合
  - http://dash-mse-test.appspot.com/dash-player.html?url=http://yt-dash-mse-test.commondatastorage.googleapis.com/media/feelings_vp9-20130806-manifest.mpd
  - https://shaka-player-demo.appspot.com/demo/#asset=https://storage.googleapis.com/shaka-demo-assets/angel-one/dash.mpd
  - http://v1-6-2.shaka-player-demo.appspot.com/
  - https://www.youtube.com/tv
- H.264 対応ビルドでテストする場合
  - https://jsbin.com/gejugat/edit?output
  - https://videojs.com/
  - https://players.akamai.com/hls/
    - webm の mpd ファイルを指定すれば VP9 の動画でのテストも可能

dynamis, 2019-11-25:
神宮さんより Firefox 68 on RZ/G2E にて pulseaudio 有効化していたら問題なく動作するという報告をもらい、Firefox on RZ/G2M でも再試験したところ確かに YouTube その他の MSE プレイヤーで動作することが確認できた。前回テスト時に pulseaudio が有効化されていなかったか、何かそのビルドによる問題だった様子。

YouTube 以外に試験しやすいページ: https://bitmovin.com/demos/hls-fmp4

note: hls.js demo では MesiaSoure.isTypeSupported() などでサポートしているコーデックの検査をしてくるがその時点で `video/mp4; codecs="avc1.42E01E, mp4a.40.2` などと音声に AAC (mp4a.40.2) が含まれるなどと明記して確認すると非対応とされることに注意。動画コーデック側までならパスして実際動画の再生が可能。

コーデックパラメータ情報: https://wiki.whatwg.org/wiki/Video_type_parameters

yoshikuni-kamimiya, 2019-11-25:
Closeしたところすみません。コメントさせてください。
私の方でFirefox 60 on RZ/G2EでYoutube再生したところ、pulseaudioの有無関係なく再生不可でしたので、貴社の前回テスト時NGだったのはFirefox60だった影響かと思います。

また、動作確認の結果、Youtube再生等々可能だったため、「MSEによる動画ストリーミングは可能」という認識で問題ないでしょうか。

＊以下補足。
「YouTube 以外に試験しやすいページ」についてですが、Firefox68ではpulseaudio関係なく動画再生できました。Firefox60ではpulseaudio関係なく動画再生NGでした。(少し再生されて停止するの繰り返し)

最後にnoteについて確認させてください。浅井さんに添付いただいた試験しやすいページは、MediaSoure.isTypeSupported() (https://bitmovin.com/demos/hls-fmp4のソースにはこの記述はない？)で音声コーデックの検査がされないので、上記補足のように、pulseaudio関係なく動画再生が可能だった。という認識で正しいでしょうか。(youtubeなどではちゃんと音声コーデック検査をされていた？)

dynamis, 2019-11-25:
確かに以前は Fx68 の前で Fx60 で試験しており、今回は Fx68 でのみ試験していたようです。
Fx68 では問題ないが Fx60 では問題がある状態として reopen します。

> 浅井さんに添付いただいた試験しやすいページは、MediaSoure.isTypeSupported() (https://bitmovin.com/demos/hls-fmp4のソースにはこの記述はない？)で音声コーデックの検査がされないので、上記補足のように、pulseaudio関係なく動画再生が可能だった。という認識で正しいでしょうか。(youtubeなどではちゃんと音声コーデック検査をされていた？)

MediaSoure.isTypeSupported() の挙動に pulseaudio の有効無効が関係ないという確証があるわけではありませんが、以下の範囲については pulseaudio の有効無効と関係なかったようです:

```
window.MediaSoure.isTypeSupported('video/mp4') === true
window.MediaSoure.isTypeSupported('video/mp4; codecs="avc1.42E01E"') === true
window.MediaSoure.isTypeSupported('video/mp4; codecs="avc1.42E01E, mp4a.40.2"') === false
```

youtube や bitmovin などは isTypeSupported を使っているかも知れませんが、このような false になる引数では使用していないものと思われます。

hls.js は最後の AAC 音声を含むコーデックをサポートしているか確認するコードがあるため再生を試みる前に停止します (このチェック部分のコードだけをスキップしたら動画は再生可能でした。但し音声出力までは確認できておりません)。

dynamis, 2020-03-18:
Gecko 68 on RZ/G2E では動作しないケースだがその時に

> ts:1584515929.009239  level:0x00010000        func:omxrMcApiProxy_UseEGLImage(1212)   tid:1061      mes:This function is not implemented

というようなログが出ている。ゼロコピー化 (#83) の話の時に出ていたのと同じ？

dynamis, 2020-03-19:
func:omxrMcApiProxy_UseEGLImage のメッセージがログに残るのは RZ/G2E であるからとか MSE (HLS or DASH) で再生しているからではなく MP4 動画を再生しているときのメッセージだった。

|                   | RZ/G2E                                       | RZ/G2M                     |
| ----------------- | -------------------------------------------- | -------------------------- |
| 単体 MP4          | 再生可、ログメッセージあり                   | 再生可、ログメッセージあり |
| 単体 WebM/VP9/Ogg | 再生可、ログメッセージなし                   | 再生可、ログメッセージなし |
| MSE MP4           | 冒頭のみ再生、途中で停止、ログメッセージあり | 再生可、ログメッセージあり |
| MSE VP9           | 冒頭のみ再生、途中で停止、ログメッセージなし | 再生可、ログメッセージなし |

試験に利用したページはこちら:
- MP4, WebM (VPx) etc 単体動画再生 https://tekeye.uk/html/html5-video-test-page
- MP4 動画の MSE (HLS) ストリーミング再生 https://bitmovin.com/demos/stream-test
- VP9 動画の MSE (DASH) ストリーミング再生 https://bitmovin.com/mpeg-dash-vp9-vod-live/

起動シェル側へのログだけでなく JS Console も当然ながらエラーナシのためもう少しちゃんとデバッグしての調査が必要。

再生開始から動画再生領域の描画更新停止 (Web ページ的には動画の再生自体は続いている) までの時間は一定ではない。VP9 のサンプル https://bitmovin.com/mpeg-dash-vp9-vod-live/  で 18 秒くらいで止まるとき、26 秒程度などがある。数回ずつ試した程度なので偶然かも知れないが開発ツールを左右に表示してコンテンツ領域の画面幅を変えると継続時間に影響している用に思える。開発ツールで画面半分くらいにしたときに長くなったが更に幅狭くするとまた短くなった。

OS のメモリ自体は余っている状態だが、ビデオのデコード・バッファに使用するメモリ領域などに影響するのかも知れない。取りあえず再現条件は 68 on G2E で確定なのでその調査をしましょう。