webdino/gecko-embedded #89: WebRender の検証とデフォルト有効化の検討

dynamis, 2019-09-04:
Gecko 68 ビルドで WebRender を有効化しても問題なく動作していそうだが、十分な検証が出来ていない。いろいろなコンテンツやベンチマークでのテストを行って問題なくパフォーマンス向上が得られるようであればデフォルト有効化 (gfx.webrender.enabled=true, gfx.webrender.all=true) を検討する。

一般的な組込向けのコンテンツでの影響についてはまだ把握できていないが、WebRender が有効に働く、レイアウト処理が重要な chalkboard bench では 8 倍程度速くなっている:
https://github.com/webdino/gecko-embedded/wiki/Benchmark#chalkboard-html5-benchmark

dynamis, 2019-10-08:
note: Platform Graphic Team 的には Q3 2020 に Wayland Support と位置づけてみていた様子 (いくらでも延期できる位置づけ):
https://github.com/orgs/FirefoxGraphics/projects/1#card-19879690

dynamis, 2019-10-23:
note: 負荷テスト的にベンチを動かしたりしている範囲でクラッシュが見られるのは OrangeMark だがこれは WebRender 無効でも同じくクラッシュする。今のところ WebRender オンの時だけの問題は見つかっていない。

dynamis, 2020-01-14:
gfx.webrender.all=true にして試験していてそれに起因する問題は特に観測しておらず (問題があるなら観測するためにも)、パフォーマンス的には明らかに良いのでデフォルトオンに設定を入れる。

https://github.com/webdino/meta-browser/blob/gem-firefox-68.0.0esr/dynamic-layers/rust-layer/recipes-browser/firefox/firefox/prefs/webgl.js
https://github.com/webdino/meta-browser/blob/gem-firefox-68.0.0esr/dynamic-layers/rust-layer/recipes-browser/firefox/firefox_68.0esr.bb#L95

等と同様に webrender.js を
```
pref("gfx.webrender.all", true);
```

として作成、追加してデフォルト有効化する。

cosmo0920, 2020-01-14:
> 今のところ WebRender オンの時だけの問題は見つかっていない。

MotionMark1.0:のトランプのマークがぼやけるという問題を畑ケは把握していますが、そのようなコンテンツがぼやける問題は通常のコンテンツでは起きていないのでしょうか？

[image]

dynamis, 2020-01-14:
> WebRenderを有効化すると、MotionMark1.0と1.1のトランプのマークがぼやけるという問題を畑ケは把握しています

それは気付きませんでした。なるほど、まだダメなケースがあるのですね (後ほど再現性確認してからまたコメント追記します)...

ちなみに、小さな領域ではなく大きな矩形領域がノイズのようになる現象はいつかのベンチマークコンテンツで発生していますがそちらは WebRender オフでも発生していたように思います。

ベンチ系ではないコンテンツで問題を見かけたことはないのですが、RefTest および目視でのテストを早め、多めにしていかないと不味いですね。

dynamis, 2020-01-29:
調査途中ですがひとまず進捗共有します。

MotionMark のうち Suits suite  で shape を使っているテストで再現することを確認 (後半のテストだから目視でずっと見て無くて気付かなかった...) developer ページで該当テストだけの実施が可能:

https://browserbench.org/MotionMark1.1/developer.html
- Suits suite のうち次のテスト
  - [Suits: shape only](https://browserbench.org/MotionMark1.1/developer.html?test-interval=30&display=minimal&tiles=big&controller=ramp&frame-rate=50&kalman-process-error=1&kalman-measurement-error=4&time-measurement=performance&suite-name=Suitssuite&test-name=Suitsshapeonly) 
  - [Suits: clip, shape, rotation](https://browserbench.org/MotionMark1.1/developer.html?test-interval=30&display=minimal&tiles=big&controller=ramp&frame-rate=50&kalman-process-error=1&kalman-measurement-error=4&time-measurement=performance&suite-name=Suitssuite&test-name=Suitsclipshaperotation)
  - [Suits: clip, shape, gradient](https://browserbench.org/MotionMark1.1/developer.html?test-interval=30&display=minimal&tiles=big&controller=ramp&frame-rate=50&kalman-process-error=1&kalman-measurement-error=4&time-measurement=performance&suite-name=Suitssuite&test-name=Suitsclipshapegradient)

SVG Shape のテストを探してみるとこちらのようなサンプルで全く描画されない (ぼやけるのでは無く描画されない) ケースを確認。詳細まだ確認できていないが Shape.star の引数で css: true とすると描画されず css:false であれば正常に描画されるので SVG x CSS となるケースに何か問題がありそう:

https://codepen.io/dynamis/pen/mdyYENv (元のサンプルは https://codepen.io/daybrush/full/ReYxLy via [shape-svg](https://github.com/daybrush/shape-svg))

dynamis, 2020-01-30:
Next: WPT テストを WebRender 有効で実行、失敗するものは WebRender オフにして再試験とすることで、まずは WebRender 有効化に伴う副作用・制限事項の洗い出しを行う。

デフォルト有効にするかどうか (問題を解決または制限事項扱いするかどうか) はその結果や他の issue との優先順位を踏まえて判断する。

dynamis, 2020-03-05:
そもそもの情報として Bugzilla のリンク貼ってなかったので:

- [[meta] wr-linux](https://bugzilla.mozilla.org/show_bug.cgi?id=1491303) - [bug tree](https://bugzilla.mozilla.org/showdependencytree.cgi?id=1491303)
  - [[meta] wr-linux-gnome-x11-mvp](https://bugzilla.mozilla.org/show_bug.cgi?id=1614523)
    - まだ Gnome X11 での MVP リリースすらされておらず、色々問題残っててデフォルト有効化は厳しいかも

先の MotionMark での bug がどの bug に該当するのか、あるいは未報告なのはぱっと見で分からなかった。未確認。とはいえ WPT 実行して実際問題が再現するケースは見つけておき、普通の Linux Desktop での再現性も確認の上で報告するなどしていきたい。

dynamis, 2020-03-18:
WebRender 有効化時に画面の描画が一部ブロックノイズになるページ:
- CSS Animation コンテンツ
  - https://dynamis.github.io/flux-slider/demo.html
  - https://codepen.io/suez/full/AXQaEg
  - CSS Animation でも小さな領域のものは (ズーム表示しても) 問題ない。大きさ依存ありそう
- Canvas
  - http://craftymind.com/factory/html5video/CanvasVideo.html
- SVG 関連
  - https://bitmovin.com/demos/hls-fmp4
    - ビデオじゃなくてページ上部のバーの背景に SVG を backgroud-image で設定しているものが化ける (`background-image: url(/demos/assets/images/header-bg.svg)`) 読み込んでいる SVG ファイル単体で読み込むと問題はないが背景画像に指定している時にノイズになる
- その他
  - https://css-tricks.com/lodge/svg/06-using-svg-svg-background-image/
    - 動画を onload 後に拡大している (Canvas に描いてる？) ときに化ける。動画再生後は問題ない。

何かデコードしてバッファに持つ領域のサイズが一定を超えるとノイズになってる印象。

dynamis, 2020-03-19:
Geckoo 78 での改善に期待としてデフォルト無効のままとすることで確定

closed 2020-03-19 by dynamis