webdino/gecko-embedded #94: Gecko 78 に向けての調査、検討

dynamis, 2020-02-03:
Gecko 68 から 78 までの差分、メリット、リスクなどを確認する

取りあえず公開情報のリリースノートのメモはこちらに:

https://www.mozilla.org/en-US/firefox/69.0/releasenotes/
- The Block Autoplay feature is enhanced to give users the option to block any video that automatically starts playing, not just those that automatically play with sound.
  - 自動再生したい場合に要注意

https://www.mozilla.org/en-US/firefox/70.0/releasenotes/
- A faster Javascript Baseline Interpreter to handle the modern web’s
large codebases and improve page load performance by as much as 8
percent.
  - ページ読み込み (遷移) 時間の大幅短縮

https://www.mozilla.org/en-US/firefox/71.0/releasenotes/
- Native MP3 decoding on Windows, Linux, and macOS
- New kiosk mode functionality, which allows maximum screen space for customer-facing displays
  - オフィシャルなキオスクモード対応

https://www.mozilla.org/en-US/firefox/72.0/releasenotes/
- Picture-in-picture video is now also available in Firefox for Mac and Linux: Select the blue icon from the right edge of a video to pop open a floating window so you can keep watching while working in other tabs or apps. Learn how the feature works.

https://www.mozilla.org/en-US/firefox/73.0/releasenotes/
- Added support for setting a default zoom level applicable for all web content, regardless of site. This option is available in about:preferences under "Language and Appearance" and can be scaled up or down from 100% as needed.

以下はリリースノートまだなし:

https://www.mozilla.org/en-US/firefox/74.0/releasenotes/
https://www.mozilla.org/en-US/firefox/75.0/releasenotes/
https://www.mozilla.org/en-US/firefox/76.0/releasenotes/
https://www.mozilla.org/en-US/firefox/77.0/releasenotes/
https://www.mozilla.org/en-US/firefox/78.0/releasenotes/

dynamis, 2020-03-10:
少しメモを追記:

### Project Fission
Meltdown や Spectre などに対するセキュリティ対策を強化する必要もあり iframe などのプロセス分離を行う Project Fission (フィジョン) が進行中で、Fx69 から試験的に有効化が可能になっている。正式リリースタイミングは未定 (78 に入るか不明)。

https://wiki.mozilla.org/Project_Fission
https://firefox-source-docs.mozilla.org/dom/Fission.html
https://bugzilla.mozilla.org/show_bug.cgi?id=fission

プロセスが増えることによりメモリ使用量が増えないように MemShrink も頑張っている:
https://wiki.mozilla.org/Project_Fission/Memory

JS の使用メモリは 51%, Resident Unique は 43% の削減を実現している

公開の解説動画としてはこちらが詳しい:
https://onlinexperiences.com/Launch/Event.htm?ShowKey=44908&DisplayItem=E334923

論理的には iframe などで重たい JS コンテンツ (ソーシャル埋め込みなど) を読み込んでいるサイトなどでは顕著なパフォーマンス向上が見られるハズ。

### WebGPU
Firefox 73 から試験的に対応。dom.webgpu.enabled にて有効化。
但し 78 までに正式サポートはしなさそう。てかそもそも SoC 側が Vulkan

webgpu-mvp https://bugzilla.mozilla.org/show_bug.cgi?id=1602129

https://fosdem.org/2020/schedule/event/rust_webgpu/attachments/slides/3983/export/events/attachments/rust_webgpu/slides/3983/BuildingWebGPUwithRust_Fosdem2020.pdf

### パフォーマンス向上
某 Android 環境での Firefox 52 と 68 について YouTube 起動時間を比較計測することがあったのでその結果をメモ。計測方法はリモートデバッグでキャッシュ無効、リロードしてページロード完了までの時間 (DOMContentloaded などは 52 で得られないので完了のみでの比較)。

主要因がなんなのかなどは一切見ていないし直接関係はないが、まぁ日々性能は上がってるよねというサンプルデータの一つと言うことで。

78 がリリースされたら適当な arm 環境での比較をしてみるのもよい。但し、Android Firefox は既存の Fennec 実装が 68 止まりでそれ以降は Fenix になって設計変更による性能向上が入ってくるため arm linux ビルド・環境を適当に見繕って試験した方が良い。

|                                       | Fx68 DOM | Fx68 load | Fx68 Complete |      | Fx52 DOM | Fx52 load | Fx52 Complete |      | diff   |
| ------------------------------------- | -------- | --------- | ------------- | ---- | -------- | --------- | ------------- | ---- | ------ |
| [m.youtube.com](http://m.youtube.com) | 2.04     | 3.31      | 5.13          |      | n/a      | n/a       | 6.67          |      |        |
|                                       | 1.99     | 3.42      | 5.69          |      | n/a      | n/a       | 7.32          |      |        |
|                                       | 1.7      | 3.74      | 6             |      | n/a      | n/a       | 6.17          |      |        |
|                                       | 2.7      | 4.29      | 5.78          |      | n/a      | n/a       | 6.04          |      |        |
|                                       | 1.79     | 2.98      | 4.66          |      | n/a      | n/a       | 6.36          |      |        |
|                                       | 1.98     | 3.41      | 5.05          |      | n/a      | n/a       | 6.53          |      |        |
|                                       | 2.01     | 2.65      | 5.88          |      | n/a      | n/a       | 6.55          |      |        |
| average                               | 2.03     | 3.40      | 5.46          |      |          |           | 6.52          |      | 19.51% |
|                                       |          |           |               |      |          |           |               |      |        |
| m.youtube.com/user/xxxxxxxx           | 2.81     | 3.71      | 5.01          |      | n/a      | n/a       | 7.08          |      |        |
|                                       | 2.33     | 3.51      | 4.82          |      | n/a      | n/a       | 6.14          |      |        |
|                                       | 2.07     | 3.29      | 4.36          |      | n/a      | n/a       | 6.04          |      |        |
|                                       | 2.1      | 3.57      | 5.92          |      | n/a      | n/a       | 5.73          |      |        |
|                                       | 2.17     | 3.59      | 5.29          |      | n/a      | n/a       | 5.72          |      |        |
|                                       | 2.25     | 3.46      | 4.57          |      | n/a      | n/a       | 6.49          |      |        |
|                                       | 2.41     | 3.73      | 5.05          |      | n/a      | n/a       | 6.04          |      |        |
| average                               | 2.31     | 3.55      | 5.00          |      |          |           | 6.18          |      | 23.47% |
|                                       |          |           |               |      |          |           |               |      |        |
| m.youtube.com/watch?v=xxxxxxx         | 1.96     | 4.05      | 4.74          |      | n/a      | n/a       | 8.7           |      |        |
|                                       | 1.88     | 4.11      | 4.67          |      | n/a      | n/a       | 6.62          |      |        |
|                                       | 2.21     | 4.23      | 4.8           |      | n/a      | n/a       | 6.4           |      |        |
|                                       | 2.2      | 4.3       | 5.21          |      | n/a      | n/a       | 5.36          |      |        |
|                                       | 1.86     | 4.01      | 4.6           |      | n/a      | n/a       | 6.24          |      |        |
|                                       | 1.79     | 3.75      | 4.32          |      | n/a      | n/a       | 6             |      |        |
|                                       | 1.87     | 4.13      | 4.63          |      | n/a      | n/a       | 6.25          |      |        |
| average                               | 1.97     | 4.08      | 4.71          |      |          |           | 6.51          |      | 38.22% |

ashie, 2020-03-10:
当プロジェクト的には WaylandのDMABufサポートが来るのが大きいです。 https://mastransky.wordpress.com/

dynamis, 2020-03-11:
>> 当プロジェクト的には WaylandのDMABufサポートが来るのが大きいです。
>> https://mastransky.wordpress.com/2020/03/03/webgl-and-fgx-acceleration-on-wayland/
> 
>  we finally have all pieces together to build fully accelerated browser on Linux which is equal to its Windows siblings.

すてき。遂に Linux の大きな弱点が解消される！

[Wayland] Use wayland dmabuf as WebGL backend https://bugzilla.mozilla.org/show_bug.cgi?id=1586696

memo: widget.wayland-dmabuf-webgl.enabled を true に設定して試す。

dynamis, 2020-03-19:
Gecko 78 に対する期待または調査事項リスト

- [ ] Gecko 68 と比べてどの程度 WebGL のパフォーマンスが向上するか (DMAbuf サポート結果への期待)
- [ ] Gecko 68 と比べてどの程度メモリ使用量が削減されるか (Fission に伴う削減への期待)
- [ ] e10s 有効でも WebGL のパフォーマンスが出るようになるか (Gecko 68 では e10s 無効である必要があった: #91)
- [ ] WebRender を有効化して問題無くならないか (Gecko 68 では断念: #89)
- [ ] MSE の再生に支障がないか (see also: #87)
- [ ] Weston での不安定性に改善が見られないか

dynamis, 2020-06-22:
結局 WebRender の Linux 正式サポートは 78 に入らなかったというか Win10 だけだった:
https://wiki.mozilla.org/Platform/GFX/WebRender_Where

dynamis, 2020-10-30:
- [ ] Firefox 標準の kiosk モードが RZ/G で問題ないか確認も必要

Fx 78 では Fx71 で導入された本家の kiosk モードを利用、現状 #72 への workaround として [GEM 側で独自に実装しているモノ](https://github.com/webdino/meta-browser/blob/firefox-60esr/recipes-browser/firefox/firefox/wayland/kiosk/0001-Make-the-window-fullscreen-by-default.patch) は廃止にできないか。Fx78 の official kiosk モードで問題無いか確認、検討を行う

https://support.mozilla.org/ja/kb/firefox-enterprise-kiosk-mode
https://bugzilla.mozilla.org/show_bug.cgi?id=158968

dynamis, 2020-10-30:
- [ ] e10s 周り次第でタッチイベント対応 #98 が Fx78 でも継続して必要になるかどうか要確認

dynamis, 2021-07-19:
Gecko 78 対応 VLP リリース済なのでクローズ

closed 2021-07-19 by dynamis