webdino/gecko-embedded #98: Gecko 68 で e10s/APZ 無効時にタッチイベントが発行されない

dynamis, 2020-08-18:
TL;DR: RZ/G では e10s/APZ 無効で起動している場合にもタッチイベントをハンドリングできるようにする必要がある。

詳細:

現状の Gecko 68 ビルドはデフォルト e10s/APZ 有効でその場合にタッチイベントをマルチタッチ込みでサポートしているが、WebGL の FSP 上限が 60fps にならない問題 (#91) が発生する場合の回避策として e10s の無効化が必要な場合がある。

その場合には e10s と共に APZ が無効化される結果タッチイベント処理に対応しなくなってしまう (マウスイベントとして処理される)。これは [Firefox 標準の制限事項扱いされている挙動](https://developer.mozilla.org/ja/docs/Web/API/Touch_events#Browser_compatibility) であるがそれでは実用上マルチタッチ操作ができなくなって困るため修正が必要。

多分適切なところで `RegisterTouchWindow()` する感じだろうか。

参考:
- ポインターイベント (マウス・タッチ) の発生確認テスト: https://patrickhlauke.github.io/touch/tests/event-listener.html
  - タッチイベントが発行されてないとタッチ操作時も mouseXXX イベントが発生
- タッチイベント発生時のポインタ位置可視化 (マルチタッチ確認にも): https://patrickhlauke.github.io/touch/touchlist-objects/
- Gecko 60 時点での強制タッチイベント対応パッチ
https://github.com/webdino/meta-browser/blob/firefox-60-wip/recipes-mozilla/firefox/firefox/fixes/0001-Add-a-preference-to-force-enable-touch-events-withou.patch

anyakichi, 2020-08-25:
Gecko 60 のパッチがほぼそのまま当たる状態だったので、調整して 68 にも追加した。

https://github.com/webdino/meta-browser/commit/3e17e9e6865535459a495ef518e0422c85ed359d

anyakichi, 2020-10-15:
Gecko 68 については動作確認済み。

Gecko 78 についての状況確認の備忘録のため、もうしばらくクローズはしないでおく。

anyakichi, 2020-12-17:
Gecko 78 についても同様だったため、パッチを追加した。動作確認済み。

closed 2020-12-17 by anyakichi