webdino/gecko-embedded #99: HID デバイスがない場合に起動しない

dynamis, 2020-08-19:
マウス・キーボードなどの HID デバイスが存在しない状態で firefox/webview のビルドが Segmentation fault で起動しない問題が発生している。

再現確認済みの環境は RZ/G2E v1.0.4 + Gecko 68

dynamis, 2020-08-19:
Oota wrote:
>  G2E (VLP64 v1.0.4 + Gecko 68) 環境にて、何もUSBに接続していない状態 (USB-serialのみ)でFirefoxが起動しない現象が発生しました。
> マウス、キーボード、タッチパネルの入力用USBケーブルのどれかを接続した場合に起動することを確認したため、上記の報告とさせていただきました。
(それ以外の機材、たとえばUSBカメラ等では試しておりません)

log by Inajima:
```
 (firefox:727): Gdk-CRITICAL **: gdk_seat_get_keyboard: assertion 'GDK_IS_SEAT (seat)' failed
Redirecting call[  106.197487] audit: type=1701 audit(1593490273.343:6): auid=4294967295 uid=0 gid=0 ses=4294967295 pid=727 comm="firefox" exe="/usr/lib64/firefox/firefox" sig=11 res=1
 to abort() to mozalloc_abort
Segmentation fault (core dumped)
```

log by Oota:
```
root@ek874:~# firefox
(firefox:493): Gdk-CRITICAL **: gdk_seat_get_keyboard: assertion 'GDK_IS_SEAT (seat)' failed
(firefox:493): Gdk-CRITICAL **: gdk_seat_get_keyboard: assertion 'GDK_IS_SEAT (seat)' failed
(firefox:493): Gdk-CRITICAL **: gdk_seat_get_keyboard: assertion 'GDK_IS_SEAT (seat)' failed
(firefox:493): Gdk-CRITICAL **: gdk_seat_get_keyboard: assertion 'GDK_IS_SEAT (seat)' failed
(firefox:493): Gdk-CRITICAL **: gdk_seat_get_keyboard: assertion 'GDK_IS_SEAT (seat)' failed
(firefox:493): Gdk-CRITICAL **: gdk_seat_get_keyboard: assertion 'GDK_IS_SEAT (seat)' failed
(firefox:493): Gdk-CRITICAL **: gdk_seat_get_keyboard: assertion 'GDK_IS_SEAT (seat)' failed
(firefox:493): Gdk-CRITICAL **: gdk_seat_get_keyboard: assertion 'GDK_IS_SEAT (seat)' failed
(firefox:493): Gdk-CRITICAL **: gdk_seat_get_keyboard: assertion 'GDK_IS_SEAT (seat)' failed
Attempting load of libEGL.so
(firefox:493): Gdk-CRITICAL **: gdk_seat_get_keyboard: assertion 'GDK_IS_SEAT (seat)' failed
console.error: "Could not write session state file " (new Error("_initWorker called too early! Please read the session file from disk first.", "resource:///modules/sessionstore/SessionFile.jsm", 337)) "_initWorker/<@resource:///modules/sessionstore/SessionFile.jsm:337:15\n_initWorker@resource:///modules/sessionstore/SessionFile.jsm:330:12\n_postToWorker@resource:///modules/sessionstore/SessionFile.jsm:354:11\nwrite@resource:///modules/sessionstore/SessionFile.jsm:394:19\nwrite@resource:///modules/sessionstore/SessionFile.jsm:66:12\n_writeState@resource:///modules/sessionstore/SessionSaver.jsm:357:12\n_saveState@resource:///modules/sessionstore/SessionSaver.jsm:295:12\n_saveStateAsync@resource:///modules/sessionstore/SessionSaver.jsm:341:5\nsaveStateAsyncWhenIdle@resource:///modules/sessionstore/SessionSaver.jsm:192:9\n"
Fontconfig warning: Directory/file mtime in the future. New fonts may not be detected.
(firefox:493): Gtk-WARNING **: Error loading theme icon 'dialog-warning' for stock: Icon 'dialog-warning' not present in theme gnome
1591600113946   addons.productaddons    WARN    Failed downloading XML, status: 0, reason: error
^C
```

Oota wrote:

> G2E v1.0.2 + Gecko 60で何も繋がずに firefox を実行した場合、同様の GDK のエラーが複数回発生しますが起動自体は成功しました。
> Gecko 68 になってからの挙動と思われます。

anyakichi, 2020-08-19:
Firefox 78 でも同じようなものが出ていたので、--enable-debug で作ったバイナリを gdb でバックトレースを見てみた結果（途中で OOM Killer が来たりとデバッグそのものがかなり苦しい）。

```
root@hihope-rzg2h:~# gdb firefox
GNU gdb (GDB) 8.0
Copyright (C) 2017 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "aarch64-poky-linux".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from firefox...Reading symbols from /usr/lib64/firefox/.debug/firefox...done.
done.
(gdb) run
Starting program: /usr/bin/firefox 
warning: the debug information found in "/lib/ld-2.19.so" does not match "/lib/ld-linux-aarch64.so.1" (CRC mismatch).

warning: the debug information found in "/lib/.debug/ld-2.19.so" does not match "/lib/ld-linux-aarch64.so.1" (CRC mismatch).

[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".
warning: File "/usr/lib64/libstdc++.so.6.0.24-gdb.py" auto-loading has been declined by your `auto-load safe-path' set to "$debugdir:$datadir/auto-load".
To enable execution of this file add
	add-auto-load-safe-path /usr/lib64/libstdc++.so.6.0.24-gdb.py
line to your configuration file "/home/root/.gdbinit".
To completely disable this security protection add
	set auto-load safe-path /
line to your configuration file "/home/root/.gdbinit".
For more information about this security protection see the
"Auto-loading safe path" section in the GDB manual.  E.g., run from the shell:
	info "(gdb)Auto-loading safe path"
[New Thread 0xfffff78ff1c0 (LWP 741)]
[Thread 0xfffff78ff1c0 (LWP 741) exited]

(firefox:736): Gdk-CRITICAL **: gdk_seat_get_keyboard: assertion 'GDK_IS_SEAT (seat)' failed
[New Thread 0xffffea5a41c0 (LWP 751)]
[New Thread 0xffffea5641c0 (LWP 752)]
[New Thread 0xffffe60ff1c0 (LWP 753)]
[New Thread 0xffffe60bf1c0 (LWP 754)]
[New Thread 0xffffe607f1c0 (LWP 755)]
[New Thread 0xfffff78ff1c0 (LWP 756)]
[Thread 0xfffff78ff1c0 (LWP 756) exited]
[New Thread 0xffffea5241c0 (LWP 757)]
[New Thread 0xffffe5cff1c0 (LWP 758)]
[New Thread 0xffffe5b011c0 (LWP 759)]
[New Thread 0xffffe59031c0 (LWP 760)]
[New Thread 0xffffe57051c0 (LWP 761)]
[New Thread 0xffffe55071c0 (LWP 762)]
[New Thread 0xffffe53091c0 (LWP 763)]
[New Thread 0xffffe510b1c0 (LWP 764)]
[New Thread 0xffffe4f0d1c0 (LWP 765)]
[New Thread 0xffffe603f1c0 (LWP 766)]
[New Thread 0xffffe45ff1c0 (LWP 767)]
[New Thread 0xffffe45df1c0 (LWP 768)]
[New Thread 0xffffe459f1c0 (LWP 770)]
[New Thread 0xffffe40ff1c0 (LWP 771)]
[Thread 0xffffe603f1c0 (LWP 766) exited]
[New Thread 0xffffe455f1c0 (LWP 772)]
[New Thread 0xffffe603f1c0 (LWP 773)]
[New Thread 0xffffe3aee1c0 (LWP 774)]
[New Thread 0xffffe3aae1c0 (LWP 775)]
[New Thread 0xffffe3a6e1c0 (LWP 776)]
[New Thread 0xffffe3a2e1c0 (LWP 777)]
[... 93 lines cut ...]
[ 1200.917376] [    620]     0   620     6308      442    86016        0             0 weston-keyboard
[ 1200.926424] [    629]     0   629     7785     1985    94208        0             0 weston-desktop-
[ 1200.935497] [    631]     0   631      845       86    45056        0             0 sh
[ 1200.943422] [    734]     0   734   856927   837945  6811648        0             0 gdb
[ 1200.951426] [    736]     0   736   615429    13345   790528        0             0 firefox
[ 1200.959793] Out of memory: Kill process 734 (gdb) score 869 or sacrifice child
[ 1200.967172] Killed process 736 (firefox) total-vm:2461716kB, anon-rss:53224kB, file-rss:0kB, shmem-rss:156kB
[ 1200.984811] oom_reaper: reaped process 736 (firefox), now anon-rss:0kB, file-rss:0kB, shmem-rss:4kB
#0  0x0000aaaaaaac8d58 in mozalloc_abort(char const*) (msg=<optimized out>)
    at /build/build/tmp/work/aarch64-poky-linux/firefox/78.0esr-r0/firefox-78.0/memory/mozalloc/mozalloc_abort.cpp:33
#1  0x0000aaaaaaac8dbc in abort() ()
    at /build/build/tmp/work/aarch64-poky-linux/firefox/78.0esr-r0/firefox-78.0/memory/mozalloc/mozalloc_abort.cpp:82
#2  0x0000fffff62cee4c in wl_abort (fmt=<optimized out>)
    at /usr/src/debug/wayland/1.13.0-r0/wayland-1.13.0/src/wayland-util.c:419
#3  0x0000fffff62cb18c in wl_proxy_marshal_array_constructor_versioned (proxy=proxy@entry=0xffffe37e0c40, opcode=Python Exception <class 'gdb.MemoryError'> Cannot access memory at address 0xffffffffc4e4: 
#4  0x0000fffff62cb1f8 in wl_proxy_marshal_array_constructor (proxy=proxy@entry=0xffffe37e0c40, opcode=<error reading variable: Cannot access memory at address 0xffffffffc440>, args=args@entry=0xffffffffc510, interface=interface@entry=0xfffff62e0928 <wl_data_device_interface>)
    at /usr/src/debug/wayland/1.13.0-r0/wayland-1.13.0/src/wayland-client.c:599
#5  0x0000fffff62cb374 in wl_proxy_marshal_constructor (proxy=0xffffe37e0c40, opcode=<error reading variable: Cannot access memory at address 0xffffffffc440>, interface=0xfffff62e0928 <wl_data_device_interface>)
    at /usr/src/debug/wayland/1.13.0-r0/wayland-1.13.0/src/wayland-client.c:733
#6  0x0000ffffee713b4c in wl_data_device_manager_get_data_device(wl_data_device_manager*, wl_seat*) (wl_data_device_manager=<error reading variable: Cannot access memory at address 0x28>, seat=0xfffff7ff4ef0)
    at /build/build/tmp/work/aarch64-poky-linux/firefox/78.0esr-r0/recipe-sysroot/usr/include/wayland-client-protocol.h:2700
---Type <return> to continue, or q <return> to quit--- 
#7  0x0000ffffee713b4c in nsRetrievalContextWayland::nsRetrievalContextWayland() (this=0xffffe37e0b00)
    at /build/build/tmp/work/aarch64-poky-linux/firefox/78.0esr-r0/firefox-78.0/widget/gtk/nsClipboardWayland.cpp:669
```

anyakichi, 2020-08-19:
現象そのものはこれと同じもので、wl_data_device_manager を別途用意してやらないとダメなのだと思う。

https://github.com/swaywm/wlroots/issues/103

anyakichi, 2020-08-19:
もしくは dummy input デバイス的なものを wayland に見せられればそれで回避できるのかもしれない（？）

anyakichi, 2020-08-21:
これくらいのものを事前に起動しておけば回避できる。

```
#include <linux/uinput.h>

#include <fcntl.h>
#include <stdlib.h>
#include <string.h>
#include <unistd.h>

int main(void)
{
   struct uinput_setup usetup;

   int fd = open("/dev/uinput", O_WRONLY | O_NONBLOCK);

   /*
    * The ioctls below will enable the device that is about to be
    * created, to pass key events, in this case the space key.
    */
   ioctl(fd, UI_SET_EVBIT, EV_KEY);
   ioctl(fd, UI_SET_KEYBIT, KEY_SPACE);

   memset(&usetup, 0, sizeof(usetup));
   usetup.id.bustype = BUS_USB;
   usetup.id.vendor = 0x1234; /* sample vendor */
   usetup.id.product = 0x5678; /* sample product */
   strcpy(usetup.name, "NULL device");

   ioctl(fd, UI_DEV_SETUP, &usetup);
   ioctl(fd, UI_DEV_CREATE);

   for (;;) {
      sleep(1);
   }

   ioctl(fd, UI_DEV_DESTROY);
   close(fd);

   return 0;
}
```

参考: https://www.kernel.org/doc/html/latest/input/uinput.html

anyakichi, 2021-01-08:
上記を少々手直ししたプログラムに自動起動用の systemd の unit ファイルを追加したレシピを作成済み。FW に組み込むと自動的にサービスが有効になりダミーデバイスが作られるようになっている。

現時点では公開された場所にレシピが置かれていないが、対応自体は完了した。

closed 2021-01-08 by anyakichi

dynamis, 2022-09-27:
> 現時点では公開された場所にレシピが置かれていないが、対応自体は完了した。

非公開だがリンクを張っておく https://sh171.soum.co.jp/rzg-gecko/proprietary/-/blob/master/patches/0001-Add-dummy-input-package.patch